avrdudes/avrdude #1288: More compiler warnings from github action build

mcuee, 2023-01-15:
1) ARM Linux build (I can reproduce under Raspberry Pi 400 as well)

```
[ 19%] Building C object src/CMakeFiles/libavrdude.dir/config.c.o
/__w/avrdude/avrdude/src/config.c: In function 'cfg_escape':
/__w/avrdude/avrdude/src/config.c:811:28: warning: comparison is always true due to limited range of data type [-Wtype-limits]
  811 |       if(*s == 0x7f || (*s >= 0 && *s < 32)) {
      |    

```

mcuee, 2023-01-15:
macOS github action -- this may not be an issue with avrdude so we may have to live with it.

```
[ 87%] Building C object src/CMakeFiles/libavrdude.dir/__/lexer.c.o
/Users/runner/work/avrdude/avrdude/build/lexer.c:21[67](https://github.com/avrdudes/avrdude/actions/runs/3919001520/jobs/6699731454#step:5:68):38: warning: comparison of integers of different signs: 'unsigned long' and 'int' [-Wsign-compare]
        if (((yy_n_chars) + number_to_move) > YY_CURRENT_BUFFER_LVALUE->yy_buf_size) {
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 warning generated.

```

mcuee, 2023-01-15:
MinGW build -- we have to live with this as per @stefanrueger 

```
[ 92%] Building C object src/CMakeFiles/avrdude.dir/term.c.obj
In function 'tokenize',
    inlined from 'process_line' at D:/a/avrdude/avrdude/src/term.c:1294:10,
    inlined from 'terminal_mode_noninteractive' at D:/a/avrdude/avrdude/src/term.c:1420:14:
D:/a/avrdude/avrdude/src/term.c:1202:17: warning: pointer 'buf' may be used after 'realloc' [-Wuse-after-free]
 1202 |       ptrdiff_t k = buf - bufp;
      |                 ^
D:/a/avrdude/avrdude/src/term.c:1186:18: note: call to 'realloc' here
 1186 |       buf_tmp  = realloc(buf, bufsize);
      |                  ^~~~~~~~~~~~~~~~~~~~~

```

stefanrueger, 2023-01-15:
Neither is a bug and neither can be easily changed. In the first and third case the compiler overreaches itself (they are kinda false warnings and the code should work as intended and expected), the warning in the middle case is probably owing to code generated by flex, and unlikely wrong, which will heal itself at some point when flex outputs code that would silence that warning. So, I'd suggest not to fix any of that. (The third case could be fixed by a longish rewrite, but there are more interesting things to fix).

closed 2023-01-15 by stefanrueger