avrdudes/avrdude #1289: The programmer option "arduino" does not work.

mikrocoder, 2023-01-15:
Hi,

On the page https://github.com/avrdudes/avrdude at the bottom there is an ex.
`avrdude -c arduino -P COM1 -b 115200 -p atmega328p -D -U flash:w:objs/blink.hex:i`
with new programmer "arduino". Unfortunately this does not work. I use an original Arduino Mega2560.

works for me
```avrdude -v -c wiring -P COM3 -b 115200 -p m2560 -D -U flash:w:C:\avrToolchain\avrdude\blink25x75.hex:i```

does not work for me
```avrdude -v -c arduino -P COM3 -b 115200 -p m2560 -D -U flash:w:C:\avrToolchain\avrdude\blink25x75.hex:i```

However, when I look into the .conf I am surprised because the settings are the same. Only the description is different with protocol v1 vs. v2. "arduino" must use v2 too, right?
```
programmer
    id                     = "wiring";
    desc                   = "Wiring for bootloader using STK500 v2 protocol";
    type                   = "wiring";
    prog_modes             = PM_SPM;
    connection_type        = serial;
;

programmer
    id                     = "arduino";
    desc                   = "Arduino for bootloader using STK500 v1 protocol";
    type                   = "arduino";
    prog_modes             = PM_SPM;
    connection_type        = serial;
; 
```

Why a programmer named "arduino" at all if Arduino or the Arduino IDE uses "wiring"?

MCUdude, 2023-01-15:
`arduino` is IMO an unfortunate name for a programmer, as different Arduinos use different bootloaders, and `-c arduino` may not work.

the Arduino MEGA (2560) uses an STK500v2 compatible bootloader originally developed for the [Wiring board](http://wiring.org.co), hence the programmer name `wiring`.

Other AVR-based Arduinos (except the ATmega32U4 ones) either use ATmegaBOOT or Optiboot, both using the stk500v1 protocol, which is more lightweight than its successor, the stk500v2 protocol.

mikrocoder, 2023-01-15:
Okay understood, then the name is really unfortunate choice. Maybe the protocol version should be included in the name. I would use yes for the names stk500v1 and stk500v2.  😀

mikrocoder, 2023-01-15:
One idea for this. You know which controller uses which protocol. Then the manual distinction between v1 and v2 could be omitted and this is decided internally based on the specification of the controller. But I don't know what effort this is.

mcuee, 2023-01-16:
`-c arduino` = based on stk500v1 plus some extra bits, for bootloaders like optiboot.
`-c wiring` = based on stk500v2 plus some extra bits, for wiring bootloader.

Close this as it is not really an avrdude issue.

closed 2023-01-16 by mcuee

mikrocoder, 2023-01-16:
That's a strange answer. stk500v1 has nothing directly to do with Arduino. You should really think again about the name.

stefanrueger, 2023-01-18:
| You should really think again about the name.

Like ships and houses, one doesn't rename `-c` programmers: `-c arduino` was established in Feb 2009 presumably at a time when the Arduino IDE only served few MCUs and when arduino boards came with stk500v1-protocol bootloaders (the main difference between `-c stk500v1` and `-c arduino` is that the latter plugs RTS/DTR in an attempt to automatically reset the board). It will have made eminent sense then to name that programmer arduino. Some 14 years later and *thousands* and *thousands* (and *thousands*) of internet docs, tips & tricks etc mentioning `avrdude -c arduino` explicitly, and dozens of IDEs/Cores that use that *implicitly* it isn't really a good idea to start renaming the programmer.

If you want to know which avrdude programmers can (in theory) program the `ATmega2560` you can type in bash
``` console
$ avrdude -c \? -p ATmega2560 2>&1 | more
```

If you want to know which of these serve bootloaders on an ATmega2560, try

``` console
$ avrdude -c \? -p ATmega2560 2>&1 | grep bootloader
  adafruit_gemma     = Adafruit Trinket Gemma bootloader disguised as USBtiny via bootloader
  arduino            = Arduino for bootloader using STK500 v1 protocol via bootloader
  arduino_gemma      = Arduino Gemma bootloader disguised as USBtiny via bootloader
  avr109             = Atmel for bootloader using AppNote AVR109 via bootloader
  avr911             = Atmel for bootloader using AppNote AVR911 AVROSP via bootloader
  butterfly          = Atmel for bootloader (Butterfly Development Board) via bootloader
  butterfly_mk       = Mikrokopter.de Butterfly via bootloader
  flip1              = FLIP for bootloader using USB DFU protocol version 1 (doc7618) via bootloader
  flip2              = FLIP for bootloader using USB DFU protocol version 2 (AVR4023) via bootloader
  micronucleus       = Micronucleus via bootloader
  mkbutterfly        = Mikrokopter.de Butterfly via bootloader
  teensy             = Teensy via bootloader
  urclock            = Urclock programmer for urboot bootloaders using urprotocol via bootloader
  wiring             = Wiring for bootloader using STK500 v2 protocol via bootloader
  xbee               = XBee for Series 2 Over-The-Air (XBeeBoot) bootloader using STK500 v1 protocol via bootloader
```
This gives you the gamut of possibilities. It's then up to you to figure out which bootloader is actually on your board. 

Note that not all of these bootloaders may actually have been compiled or cater for the ATmega2560, but that's outside the remit of avrdude.

mikrocoder, 2023-01-19:
That's a reasonable answer. Thanks.