avrdudes/avrdude #1293: MSYS2: Linker error when readline is present.

mariusgreuel, 2023-01-20:
On MSYS2/MinGW64, it appears that libreadline is linked against ncurses. Currently, main does not link correctly.

```console
[100%] Linking C executable avrdude.exe
D:/Work/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: D:/Work/msys64/mingw64/lib/libreadline.a(display.o):(.text+0xed0): undefined reference to `tputs'
D:/Work/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: D:/Work/msys64/mingw64/lib/libreadline.a(display.o):(.text+0xf39): undefined reference to `tputs'
...
```

mcuee, 2023-01-21:
Previous discussions:
* https://github.com/avrdudes/avrdude/issues/1155
Previous PR to fix this issue:
* https://github.com/avrdudes/avrdude/pull/1270

Ultimate long term solution:
* https://github.com/avrdudes/avrdude/issues/1271

mcuee, 2023-02-01:
PR #1294 is good and it fixes the issue.

PR #1294 (2023-03-04): CMake: Remove LIB_LIBHID from configuration is favor of hard-coding it
Fixes #1006, fixes #1290, fixes #1293

For some reason CMake does not find `hid.lib` in Windows when run outside of a Developer Prompt for VS. This caused the PickIt2 programmer to be removed.

Rather than fixing the CMake issue, I decided to hard-code `hid.lib` for WIN32 target. This library has been available for 10+ years, so this should not be an issue.

Also, I threw in `ncurses` when `readline` is present.
closed 2023-03-04 by stefanrueger