avrdudes/avrdude #1295: libusb_init() is now deprecated

mcuee, 2023-01-25:
With the merging of https://github.com/libusb/libusb/pull/1026 recently, libusb_init() is now deprecated.

This will cause a warning when building avrdude. It is better we fix this in avrdude.

Discussions here:
* https://github.com/libusb/libusb/issues/1236

dl8dtl, 2023-01-25:
In my opinion, it is a very bad decision to change the API of a library that has been in use for more than a decade.
Quite possible that the original implementation of libusb doesn't do anything there – other (re)implementations *are* doing something, like the FreeBSD one.
If we (or anyone else) now removes the call to `libusb_init()`, that application will break on such implementations.

tormodvolden, 2023-01-26:
> If we (or anyone else) now removes the call to `libusb_init()`, that application will break on such implementations.

libusb_init_context does the same as the old libusb_init, just allows setting options for the new context.

dl8dtl, 2023-01-26:
Then we have to verify whether the respective implementation supports `libusb_init_context()`, and `#ifdef` appropriately.

At the very least, as long as the API version is called "1.0", the old (even if deprecated) implementation should remain, rather than threatening to eventually remove it.

tormodvolden, 2023-01-26:
For sure, libusb_init() will stay available in 1.0.x but possibly have disappeared in 1.1.x.

But we'd like to deprecate it for any new and maintained application code.

mcuee, 2023-01-29:
The fix from OpenOCD project seems to be simple enough.
https://review.openocd.org/c/openocd/+/7467/2/src/jtag/drivers/libusb_helper.c

```
#if !defined(LIBUSB_API_VERSION) || (LIBUSB_API_VERSION < 0x0100010A)
#define libusb_init_context(a, b, c) libusb_init(a)
#endif
```

mcuee, 2023-02-01:
Let's wait for the outcome of the discussion in https://github.com/libusb/libusb/issues/1236 before we carry out any modifications. I have created a pull request https://github.com/libusb/libusb/pull/1241 to address the issue from libusb project. If it gets merged then we can close this issue without carrying out any changes in avrdude.

mcuee, 2023-03-12:
https://github.com/libusb/libusb/commit/fcf0c710ef5911ae37fbbf1b39d48a89f6f14e8a 

The above has been commited. So this issue can be closed. There is no action needed from avrdude side.

closed 2023-03-12 by mcuee