avrdudes/avrdude #1296: Avrdude terminal history broken on macOS

MCUdude, 2023-01-29:
I just stumbled over this. When building with make, the Avrdude terminal history is broken on macOS.
I've narrowed it down to the following commit by @mcuee: 6595f534258f137c213f7de4c29fde77d67e69a9

```
$ ./avrdude -cusbasp -patmega1284p -t

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9705 (probably m1284p)
avrdude> read eeprom

Reading | ################################################## | 100% 0.07 s 

0000  48 65 6c 6c 6f 20 57 6f  72 6c 64 00 ff ff ff ff  |Hello World.....|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> ^[[A^[[A^[[A^[[A^C
```

mcuee, 2023-01-29:
Hmm, that is stange. The change is supposed to change nothing but the prompt only. Let me check later.

mcuee, 2023-01-30:
Strange, git main is broken, even with CMake build.

7.1 release is okay.

```
mcuee@mcuees-Mac-mini avrdude_main % ./build.sh
...
mcuee@mcuees-Mac-mini avrdude_main % otool -L ./build_darwin/src/avrdude             
./build_darwin/src/avrdude:
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1319.0.0)
	/opt/homebrew/opt/libusb-compat/lib/libusb-0.1.4.dylib (compatibility version 9.0.0, current version 9.4.0)
	/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/opt/homebrew/opt/hidapi/lib/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.13.1)
	/opt/homebrew/opt/libftdi/lib/libftdi1.2.dylib (compatibility version 2.0.0, current version 2.5.0)
	/usr/lib/libedit.3.dylib (compatibility version 2.0.0, current version 3.0.0)

mcuee@mcuees-Mac-mini avrdude_main % ./build_darwin/src/avrdude -c usbasp -p m168p -t

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e940b (probably m168p)
avrdude> read flash 0 0x20

Reading | ################################################## | 100% 0.02 s 

0000  0c 94 61 00 0c 94 73 00  0c 94 73 00 0c 94 73 00  | .a. .s. .s. .s.|
0010  0c 94 73 00 0c 94 73 00  0c 94 73 00 0c 94 73 00  | .s. .s. .s. .s.|

avrdude> ^[[A^[[A^[[A^[[B^[[B^[[B^[[B
```

mcuee, 2023-01-30:
@MCUdude 

Indeed you are right, revert that PR #1280 will indeed fix the issue. I think `configure.ac` is pretty broken. 

```
mcuee@mcuees-Mac-mini avrdude_main % git diff
diff --git a/src/configure.ac b/src/configure.ac
index 4f35c1c..6470f88 100644
--- a/src/configure.ac
+++ b/src/configure.ac
@@ -83,7 +83,7 @@ AC_CHECK_LIB([ws2_32], [WSAStartup])
 
 AC_CHECK_LIB([termcap], [tputs])
 AC_CHECK_LIB([ncurses], [tputs])
-AC_CHECK_LIB([readline], [readline], [have_readline=yes])
+AC_CHECK_LIB([readline], [readline])
 AH_TEMPLATE([HAVE_LIBELF],
             [Define if ELF support is enabled via libelf])
 AC_CHECK_LIB([elf], [elf_getshdrstrndx], [have_libelf=yes])
```

mcuee, 2023-01-30:
@MCUdude 

I have created PR #1297 to revert the change of PR #1280.

mcuee, 2023-01-31:
@MCUdude 

I have merged #1297 to fix this issue.

closed 2023-01-31 by mcuee