avrdudes/avrdude #1298: Linux build instructions fail on Ubuntu 18.04

jclxx, 2023-02-01:
_Many many thanks for avrdude, it's really incredibly useful_

Summary
--------------

**The out-of-the-box build instructions for Linux fail on Ubuntu 18.04 LTS (server), because of cmake errors.  It succeeds in 20.04.**

Although 18.04 is approaching end-of-maintenance it remains under security maintenance until 2028.

How to reproduce
---

The instructions for building under Linux [here](https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-Linux) say that to build (Linux/Debian variety), you do:

```
sudo apt-get install build-essential git cmake flex bison libelf-dev libusb-dev libhidapi-dev libftdi1-dev libreadline-dev
git clone https://github.com/avrdudes/avrdude.git
cd avrdude
./build.sh
```

Method

* Create freshAWS Lightsail instances (smallest kind), with Ubuntu 20.04 LTS or 18.04 LTS. 
* After first boot: apt-get update, apt-get install, reboot.
* Then the instructions above (apt-get deps, git clone, cd avrdude, ./build.sh)
* Git pull got  commit dd7b359d6b5dfcf02374fc697c80dd6d54cd7a81 of 2023-01-31

Reproducing failure in 18.04
---

This fails early in the build script

    18.04.1-LTS$ ./build.sh 
    CMake Error: The source directory "/home/ubuntu/avrdude/build_linux" does not exist.
    Specify --help for usage, or press the help button on the CMake GUI.
    CMake failed.

I am unfamiliar with cmake.  My guess is that there's some flag needed the first time to populate `build_linux`, but this should be in the build.sh script.  (Or I suppose added to the build instructions.)

Versions

```
18.04.1-LTS$ uname -a
Linux ip-172-26-6-124 4.15.0-1021-aws #21-Ubuntu SMP Tue Aug 28 10:23:07 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
18.04.1-LTS$ cmake --version
cmake version 3.10.2
```

Reproducing success in 20.04
---

```
20.04-LTS$ ./build.sh 
-- The C compiler identification is GNU 9.4.0
...
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/avrdude/build_linux
[  1%] [BISON][Parser] Building parser with bison 3.5.1
...
[100%] Built target conf

Build succeeded.

Run

sudo cmake --build build_linux --target install

to install.
```

Versions

```
20.04-LTS$ uname -a
Linux ip-172-26-13-18 5.4.0-1018-aws #18-Ubuntu SMP Wed Jun 24 01:15:00 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
20.04-LTS$ cmake --version
cmake version 3.16.3
```
---

mariusgreuel, 2023-02-01:
My guess is you have an unsupported version of CMake (we require 3.14+), perhaps one that does not support the `-B` option used in `build.sh`.

Try updating CMake. Or try using the instructions for the old version of CMake you may have. See https://cmake.org/cmake/help/latest/manual/cmake.1.html

mariusgreuel, 2023-02-01:
Just read you have 3.10. Then use [these instructions](https://cmake.org/cmake/help/v3.10/manual/cmake.1.html)

mcuee, 2023-02-01:
@jclxx 

In this case, I will close the issue as "invalid". Any objections?

jclxx, 2023-02-01:
Just checking some things now ... give me a little while.

mcuee, 2023-02-02:
No problem. take your time.

jclxx, 2023-02-02:
Portable build and installation scripts are extremely hard, so I'm extremely sympathetic to the problem.  It appears that it is exactly as you say, a version problem with cmake.  However, the build instructions don't say what the supported version is, so it's difficult to track down unless one is familiar with cmake, which I am not.

**My view is there's still a (tiny) bug which is that it would be much, much, better if the build script gave an intelligible error.  Otherwise the user, who is debugging an avrdude problem, is now debugging a cmake issue instead of simply finding out how to get a later cmake.**

The version check is inside cmake (CMakeLists.txt says `cmake_minimum_required(VERSION 3.14)`, but the cmake fails before we get to the version check.  Leaving the user clueless.

I see that under some conditions cmake will say "CMake 3.14 or higher is required.  You are running version 3.10.2", we just need to add something to trigger this inside build.sh.

Workaround
---

This is a minimal-installation workaround, tested on AWS with Ubuntu 18.04.1 LTS server as before.

```
# install all prerequisites except cmake
sudo apt-get install build-essential git flex bison libelf-dev libusb-dev libhidapi-dev libftdi1-dev libreadline-dev
# get cmake current release and unpack
wget https://github.com/Kitware/CMake/releases/download/v3.25.2/cmake-3.25.2-linux-x86_64.tar.gz
tar xvfpz cmake-3.25.2-linux-x86_64.tar.gz

# get avrdude source
git clone https://github.com/avrdudes/avrdude.git
cd avrdude/

# run build script with -x to see command that fails
sh -x build.sh
# run that command with path to latest cmake
../cmake-3.25.2-linux-x86_64/bin/cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_linux
# lots of output
../cmake-3.25.2-linux-x86_64/bin/cmake --build build_linux
# lots of output

# which produces executable successfully
$ build_linux/src/avrdude
avrdude version 7.1-20230131 (dd7b359), URL: <https://github.com/avrdudes/avrdude>

```

jclxx, 2023-02-02:
Portable build and installation scripts are extremely hard, so I'm extremely sympathetic to the problem.  It is exactly as you say, a version problem with cmake.  However, the build instructions don't say what the supported version is, so it's difficult to track down unless one is familiar with cmake, which I am not.

**My view is there's still a (very small) bug which is that it would be much, much, better if the build script gave an intelligible error.   The bug is that the current cmake version check doesn't work, because cmake fails before the check is done.**

I see no need to try to make avrdude compile with the cmake 3.10.2 you get from apt-get install on Ubuntu 18.04.

Otherwise the user, who is debugging an avrdude problem, is now debugging a cmake issue instead of simply finding out how to get a later cmake. The version check is inside cmake (CMakeLists.txt says `cmake_minimum_required(VERSION 3.14)`, but the cmake fails before we get to the version check.  Leaving the user clueless.  I see that under some conditions cmake will say "CMake 3.14 or higher is required.  You are running version 3.10.2", we just need to add something to trigger this inside build.sh.

Workaround
---

This is a minimal-installation workaround, tested on AWS with Ubuntu 18.04.1 LTS server as before.

```
# install all prerequisites except cmake
sudo apt-get install build-essential git flex bison libelf-dev libusb-dev libhidapi-dev libftdi1-dev libreadline-dev
# get cmake current release and unpack
wget https://github.com/Kitware/CMake/releases/download/v3.25.2/cmake-3.25.2-linux-x86_64.tar.gz
tar xvfpz cmake-3.25.2-linux-x86_64.tar.gz

# get avrdude source
git clone https://github.com/avrdudes/avrdude.git
cd avrdude/

# run build script with -x to see command that fails
sh -x build.sh
# run that command with path to latest cmake
../cmake-3.25.2-linux-x86_64/bin/cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_linux
# lots of output
../cmake-3.25.2-linux-x86_64/bin/cmake --build build_linux
# lots of output

# which produces executable successfully
$ build_linux/src/avrdude
avrdude version 7.1-20230131 (dd7b359), URL: <https://github.com/avrdudes/avrdude>

```

jclxx, 2023-02-02:
@mcuee Should I put in another issue for Cmake Version Number Not Checked and we close this one?  Or should we leave it here?

mcuee, 2023-02-02:
@jclxx 

We can just keep this issue open. I have changed the label to "enhancement" to detect CMake version.

mcuee, 2023-02-02:
> The version check is inside cmake (CMakeLists.txt says `cmake_minimum_required(VERSION 3.14)`, but the cmake fails before we get to the version check. Leaving the user clueless. I see that under [...]

I am not so sure if the fix should be inside `CMakeLists.txt` or `build.sh`. I tend to think it should be inside  `CMakeLists.txt`.

jclxx, 2023-02-02:
> we just need to add something to trigger this inside build.sh.

Just for clarification, I meant adding/changing an invocation of cmake inside build.sh which causes cmake to fail with version-too-small error.

The following patch to build.sh works, by making a temporary directory with a minimal CMakeLists.txt containing just the version requirement, then running cmake on that; with flags for quiet and test for exit.  

For certain there must be a more elegant way, but it achieves the goals of a) giving a clear warning message and exit status, b) getting cmake to do the test, c) not relying on any complex cmake, in particular the flags which only worked in the later versions.

Kindly consider merging this in.

Many thanks.

Check error on old cmake (3.10.2 on stock Ubuntu 18.04.1 LTS server)
---

```
$ ./build.sh
Checking CMake version
CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  CMake 3.14 or higher is required.  You are running version 3.10.2

$ echo $?
1
```

Check success on newer cmake (3.16.3 on stock Ubuntu 20.04 LTS server)
---

```
$ cmake --version
cmake version 3.16.3
...
$ ./build.sh
Checking CMake version
-- The C compiler identification is GNU 9.4.0
...
Build succeeded.
...

$ echo $?
0
```

Patch for build.sh
---

```
*** build.sh.DIST	2023-02-02 10:28:19.293785663 +0000
--- build.sh	2023-02-02 10:28:19.293785663 +0000
***************
*** 74,79 ****
--- 74,87 ----
  	;;
  esac
  
+ # force cmake version check
+ echo "Checking CMake version"
+ T=/tmp/versioncheck.$$
+ mkdir $T
+ fgrep cmake_minimum_required CMakeLists.txt > $T/CMakeLists.txt
+ (cd $T && cmake -Wno-dev .) > /dev/null || exit 1
+ rm -Rf $T
+ 
  cmake ${build_flags} ${extra_enable} -D CMAKE_BUILD_TYPE=${build_type} -B build_${ostype} ||\
      { echo "CMake failed."; exit 1; }
  cmake --build build_${ostype} ||\
```

mcuee, 2023-02-02:
@jclxx 

Please create a pull request so that others can review. Then the admins can merge if the review is good.

jclxx, 2023-02-02:
@mcuee I made pull request it's showing up as 1300.  I'm not sure if I did the request right, it's a long time since I did one. hope it's okay.

jclxx, 2023-02-04:
This is fixed by the update to wiki commit 8f9b826cff6c02691a0b778a255f3f3ce7a59965

mcuee, 2023-03-05:
Close this one as alternative #1301 has been merged.

closed 2023-03-05 by mcuee