avrdudes/avrdude #1302: Programming via JTAG with MPLab Snap

SamsiFPV, 2023-02-07:
The MPLab Snap can only program via the ISP interface with avrdude. With another open source application, bloom, it is possible to program via JTAG.

Is it possible to implement this in avrdude? Would somebody be so kind and implement this for me, or at least point me in the right direction on how to implement this?

SamsiFPV, 2023-02-07:
Correction, the MPLab Snap cannot program via ISP in avrdude. Can this be implemented too? Or is something with the hardware/software of the Snap preventing this?

mcuee, 2023-02-07:
Hmm, which avrdude version are you using?

Microchip SNAP supports both ISP and JTAG under latest avrdude 7.1 release.

SamsiFPV, 2023-02-07:
Oh. My bad then. I am using the latest available release from the Fedora RPM packages, which is 6.4
Thanks for the help

closed 2023-02-07 by SamsiFPV