avrdudes/avrdude #1303: Segfault when choosing snap programmer

SamsiFPV, 2023-02-07:
I built avrdude myself and built from main and the 7.1 release.

I get a segmentation fault when doing this command: "avrdude -c snap".

gdb says this:
```Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7e6a2da in hid_get_serial_number_string (dev=dev@entry=0x2, string=string@entry=0x7fffffff9490 L"\x479458", maxlen=maxlen@entry=256) at /usr/src/debug/hidapi-0.12.0-2.fc37.x86_64/libusb/hid.c:1455
1455            return hid_get_indexed_string(dev, dev->serial_index, string, maxlen);
```

I am building on fedora 37 and the libhid version is 0.12.0

I assume libhid is not new enough, but I haven't found any requirements in the avrdude docs

SamsiFPV, 2023-02-07:
It's not a very important problem, I can work with the nightly builds.

mcuee, 2023-02-07:
Try upgrade your hidapi installation to the latest 0.13.1 to see if the issue goes away.

SamsiFPV, 2023-02-07:
That works, thanks!

closed 2023-02-07 by SamsiFPV