avrdudes/avrdude #1304: Programmer type?

orbanp, 2023-02-07:
Hello Everyone,

I am new here, so please advise if this is the wrong place to ask the following question.

I am trying to program an AT89S52 chip.
I am using avrdude 6.3 and an ebay programmer which is a version of the usbasp programmer.
The OS is Kubuntu 20.04 LTS.
I am getting the following error message:

$ avrdude -p usbasp -c 5082 -U flash:r:"test.bin":r
avrdude: error at /etc/avrdude.conf:336: programmer type must be written as "id_type"
avrdude: error reading system wide configuration file "/etc/avrdude.conf"

I updated the configuration file that has an entry for the AT89S52 chip, the programmer description is the same that came with the installation.
The suggestion that the programmer type should be written in the configuration file with double quotes did not make any difference.

Previously I installed avrdude 7.0 and got the same error. 
avrdude 6.3 came from the Software Center of the OS.

Any help would be appreciated!
Thanks, Peter

MCUdude, 2023-02-07:
Hi!

First of all, Avrdude doesn't support programming of AT89 chips out of the box, but with a custom avrdude.conf file it might work.

First, can you share the part of your avrdude.conf where the AT89S52 chip is defined?

orbanp, 2023-02-07:
Thanks for the reply!

The error message complains about the programmer type!

I did try the original installation as well, without the updated config file, with another programmer, and with another device that were in the original config file, and got the same error message about the programmer type.

Enclosing the config file section about the AT89S52.

Thanks, Peter
```
#------------------------------------------------------------
# AT89S52
#------------------------------------------------------------
part
    id               = "8052";
    desc             = "AT89S52";
    signature        = 0x1E 0x52 0x06;
    chip_erase_delay = 500000;
    pgm_enable       = "1 0 1 0  1 1 0 0    0 1 0 1  0 0 1 1",
                       "x x x x  x x x x    x x x x  x x x x";

    chip_erase       = "1 0 1 0  1 1 0 0    1 0 0 x  x x x x",
                       "x x x x  x x x x    x x x x  x x x x";

    timeout      = 200;
    stabdelay      = 100;
    cmdexedelay      = 25;
    synchloops      = 32;
    bytedelay      = 0;
    pollindex      = 3;
    pollvalue      = 0x53;
    predelay      = 1;
    postdelay      = 1;
    pollmethod      = 0;

    memory "flash"
        size            = 8192;
        paged           = no;
        min_write_delay = 4000;
        max_write_delay = 9000;
        readback_p1     = 0xff;
        readback_p2     = 0xff;
        read            = "  0   0   1   0    0   0   0   0",
                          "  x   x   x a12  a11 a10  a9  a8",
                          " a7  a6  a5  a4   a3  a2  a1  a0",
                          "  o   o   o   o    o   o   o   o";

        write           = "  0   1   0   0    0   0   0   0",
                          "  x   x   x a12  a11 a10  a9  a8",
                          " a7  a6  a5  a4   a3  a2  a1  a0",
                          "  i   i   i   i    i   i   i   i";
   mode      = 0x21;
   delay      = 12;
      ;

    memory "signature"
        size            = 3;
        read            = "0  0  1  0   1  0  0  0   x  x  x  0   0  0 a1 a0",
                          "0  0  0  0   0  0  0  0   o  o  o  o   o  o  o  o";
      ;
  ; 

```

MCUdude, 2023-02-07:
It might be something else that's wrong with your avrdude.conf. I'm using the latest 7.1 version, and I'm able to get Avrdude to run even though there are a few warnings. Try building Avrddue yourself, and add the AT89S52 target to the avrdude.conf file that was created by the build.

```
$ ./avrdude -cusbasp -p8052
avrdude warning: a1 would normally be expected to be a9 [/Users/hans/Downloads/avrdude/src/avrdude.conf:16243]
avrdude warning: a0 would normally be expected to be a8 [/Users/hans/Downloads/avrdude/src/avrdude.conf:16243]
avrdude warning: mcuid -1 for AT89S52 is out of range [0..2039], use a free number >= 372 [/Users/hans/Downloads/avrdude/src/avrdude.conf:16245]

avrdude error: program enable: target does not answer (0x01)
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude done.  Thank you.

```

mcuee, 2023-02-08:
Since this is not supported by avrdude, I will close this issue.

The command line mentioned in the first post is totally wrong. It should be `avrdude -cusbasp -p8052 ...`.

llinjupt, 2023-03-04:
> Since this is not supported by avrdude, I will close this issue.
> 
> The command line mentioned in the first post is totally wrong. It should be `avrdude -cusbasp -p8052 ...`.

Even though avrdude doesn't support  AT89S series, for coherence, the 7.1 should behaviour the same as 7.0 which works well.  

```
$ ./avrdude.exe -P COM180 -c avrisp -p 8052 -U flash:r:flash.bin:r

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e5206 (probably 89s52)
avrdude.exe: reading flash memory:

Reading | ################################################## | 100% 3.51s

avrdude.exe: writing output file "flash.bin"

avrdude.exe done.  Thank you.
```

It seems that 7.1 couldn't analyse the old avrdude.conf well, and throwed out the error:

```
$ ./avrdude.exe -P COM180 -c avrisp -p 8052 -U flash:r:flash.bin:r
avrdude warning: a1 would normally be expected to be a9 [F:\avrdude7.1\win64\avrdude.conf:49]
avrdude warning: a0 would normally be expected to be a8 [F:\avrdude7.1\win64\avrdude.conf:49]
avrdude warning: mcuid -1 for AT89S52 is out of range [0..2039], use a free number >= 372 [F:\avrdude7.1\win64\avrdude.conf:51]

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0xffffff (probably .xmega) (retrying)
avrdude: device signature = 0xffffff (probably .xmega) (retrying)
avrdude: device signature = 0xffffff (probably .xmega)
avrdude error: Yikes!  Invalid device signature.
avrdude error: expected signature for AT89S52 is 1E 52 06
        Double check connections and try again, or use -F to override
        this check.

avrdude done.  Thank you.
```

Anyway, for convenience I strongly recommand to keep the coherence with 7.0.

llinjupt, 2023-03-04:
BTW, I have tried to update the avrdude.conf as below to fit 7.1:

```
part
    id               = "89s52";
    desc             = "AT89S52";
    stk500_devcode   = 0xE1;
    mcuid            = 372;
    signature        = 0x1e 0x52 0x06;
    chip_erase_delay = 500000;
    reset            = dedicated;
    pgm_enable       = "1010.1100--0101.0011--xxxx.xxxx--xxxx.xxxx";
    chip_erase       = "1010.1100--100x.xxxx--xxxx.xxxx--xxxx.xxxx";

    memory "flash"
        size            = 8192 ;             # bytes
        read            = "0010.0000--xxxa.aaaa--aaaa.aaaa--oooo.oooo";
        write           = "0100.0000--xxxa.aaaa--aaaa.aaaa--iiii.iiii";
        ;
    memory "signature"
        size            = 3;
        read            = "0010.1000--0000.00aa--0000.0000--oooo.oooo";
        ;
  ;

part parent "89s52"
    id = "89s51";
    desc = "AT89S51";
    signature = 0x1e 0x51 0x06;
    mcuid     = 373;
    memory "flash"
        size            = 4096 ;             # bytes
        read            = "0010.0000--xxxx.aaaa--aaaa.aaaa--oooo.oooo";
        write           = "0100.0000--xxxx.aaaa--aaaa.aaaa--iiii.iiii";
      ;
;
```

And I got another error. Since I know nothing about the a1 bit, any advice?

```
$ ./avrdude.exe -P COM180 -c avrisp -p AT89S52 -U flash:r:flash.bin:r
avrdude warning: AT89S52's signature read misses a necessary address bit a1 [F:\avrdude7.1\win64\avrdude.conf:20]

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e1e1e
avrdude error: expected signature for AT89S52 is 1E 52 06
        double check chip or use -F to override this check

avrdude done.  Thank you.
```

mcuee, 2023-03-04:
> Anyway, for convenience I strongly recommand to keep the coherence with 7.0.

No, the warnings will be there. And we will not support AT89S52.

mcuee, 2023-03-04:
That being said, maybe @MCUdude or @stefanrueger will be able to see if they can help you here or not.

mcuee, 2023-03-04:
Hmm, it seems to be that PlatformIO seems to support AT89S52 using avrdude. So maybe there are valid use case here. I will leave this to other admins to decide.
https://docs.platformio.org/en/latest/boards/intel_mcs51/AT89S52.html

stefanrueger, 2023-03-04:
Try the following entries
```
#------------------------------------------------------------
# AT89S52
#------------------------------------------------------------

part
    desc                   = "AT89S52";
    id                     = "8052";
    mcuid                  = 373;
    chip_erase_delay       = 500000;
    signature              = 0x1e 0x52 0x06;
    timeout                = 200;
    stabdelay              = 100;
    cmdexedelay            = 25;
    synchloops             = 32;
    pollindex              = 3;
    pollvalue              = 0x53;
    predelay               = 1;
    postdelay              = 1;
    chip_erase             = "1010.1100--100x.xxxx--xxxx.xxxx--xxxx.xxxx";
    pgm_enable             = "1010.1100--0101.0011--xxxx.xxxx--xxxx.xxxx";

    memory "flash"
        size               = 8192;
        min_write_delay    = 4000;
        max_write_delay    = 9000;
        readback           = 0xff 0xff;
        mode               = 0x21;
        delay              = 12;
        read               = "0010.0000--xxxa.aaaa--aaaa.aaaa--oooo.oooo";
        write              = "0100.0000--xxxa.aaaa--aaaa.aaaa--iiii.iiii";
    ;

    memory "signature"
        size               = 3;
        read               = "0 0 1 0 1 0 0 0  x x x 0 0 0 a1 a0  0 0 0 0 0 0 0 0  o o o o o o o o";
    ;
;

#------------------------------------------------------------
# AT89S51
#------------------------------------------------------------

part parent "AT89S52"
    desc                   = "AT89S51";
    id                     = "89s51";
    mcuid                  = 372;
    signature              = 0x1e 0x51 0x06;

    memory "flash"
        size               = 4096;
        read               = "0010.0000--xxxx.aaaa--aaaa.aaaa--oooo.oooo";
        write              = "0100.0000--xxxx.aaaa--aaaa.aaaa--iiii.iiii";
    ;
;

```

One problem with this part is the odd documentation for the read-signature command (see below); it claims to utilise A7..A12, but that cannot be because signatures have 3 bytes, so only A0 and A1 should be relevant. I suspect the documentation is wrong, and that A0 and A1 need to be set. I *looks like* this part has A0 and A1 not in their standard position (end of Byte 3) but instead at end of Byte 2.  *All other parts* known to AVRDUDE have the address bits in byte2 and byte3 in a systematic way, which has allowed a new shorter syntax for the command. So the read signature command for AT90S51/52 *must* use the old-style notation.

Fun fact: It used to be that contributors could not cope with the old-style notation to the effect that some 25% of the avrdude.conf entries had bits in the wrong position. This is why the warning message has been introduced. It can be ignored in this case (assuming A0/A1 have to be at the unsystematic position for these parts).

[image]

stefanrueger, 2023-03-04:
@orbanp: The avrdude command mixed up the  `-p`  *part* and `-c` *programmer* flags. It should be `-c usbasp -p 5082`, not the other way round.

> I am getting the following error message:
> ```
> $ avrdude -p usbasp -c 5082 -U flash:r:"test.bin":r
> avrdude: error at /etc/avrdude.conf:336: programmer type must be written as "id_type"
> avrdude: error reading system wide configuration file "/etc/avrdude.conf"
> ```

Please try again and report back.

llinjupt, 2023-03-04:
@stefanrueger, THX. I have tried your new avrdude.conf, but unluckly it didn't work:

'''
$ ./avrdude.exe -P COM180 -c avrisp -p AT89S52 -U flash:r:flash.bin:r
avrdude warning: a1 would normally be expected to be a9 [F:\BiTForest\avrdude7.1\win64\avrdude.conf:36]
avrdude warning: a0 would normally be expected to be a8 [F:\BiTForest\avrdude7.1\win64\avrdude.conf:36]

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0xffffff (probably .xmega) (retrying)
avrdude: device signature = 0xffffff (probably .xmega) (retrying)
avrdude: device signature = 0xffffff (probably .xmega)
avrdude error: Yikes!  Invalid device signature.
avrdude error: expected signature for AT89S52 is 1E 52 06
        Double check connections and try again, or use -F to override
        this check.

avrdude done.  Thank you.
'''

BTW, in the same environment, V7.0 with this old avrdude.conf works well:

'''
part
    id               = "89s52";
    desc             = "AT89S52";
    stk500_devcode   = 0xE1;
    signature        = 0x1e 0x52 0x06;
    chip_erase_delay = 500000;
    reset            = dedicated;
    pgm_enable       = "1 0 1 0  1 1 0 0    0 1 0 1  0 0 1 1",
                       "x x x x  x x x x    x x x x  x x x x";

    chip_erase       = "1 0 1 0  1 1 0 0    1 0 0 x  x x x x",
                       "x x x x  x x x x    x x x x  x x x x";

    memory "flash"
        size            = 8192 ;             # bytes
        read            = "0 0 1 0 0 0 0 0  x x x a12 a11 a10 a9 a8",
                          "a7 a6 a5 a4 a3 a2 a1 a0 o o o o o o o o";
        write           = "0 1 0 0 0 0 0 0 x x x a12 a11 a10 a9 a8",
                          "a7 a6 a5 a4 a3 a2 a1 a0 i i i i i i i i";
        ;
    memory "signature"
        size            = 3;
        read            = "0 0 1 0 1 0 0 0  0 0 0 0 0 0 a1 a0",
                          "0 0 0 0 0 0 0 0  o o o o o o  o  o";
        ;
  ;
part parent "89s52"
    id = "89s51";
    desc = "AT89S51";
    signature = 0x1e 0x51 0x06;
    memory "flash"
        size            = 4096 ;             # bytes
        read            = "0 0 1 0 0 0 0 0  x x x x a11 a10 a9 a8",
                          "a7 a6 a5 a4 a3 a2 a1 a0 o o o o o o o o";
        write           = "0 1 0 0 0 0 0 0 x x x x a11 a10 a9 a8",
                          "a7 a6 a5 a4 a3 a2 a1 a0 i i i i i i i i";
      ;
;
'''

stefanrueger, 2023-03-04:
@llinjupt Thanks for sharing a working 7.0 entry. My guess was based on @orbanp's entry.

It might have been the missing `stk500_devcode` entry. Try in v7.1 the following (based on your entry)
```
#------------------------------------------------------------
# AT89S52
#------------------------------------------------------------

part
    desc                   = "AT89S52";
    id                     = "89s52";
    prog_modes             = PM_ISP | PM_HVPP;
    mcuid                  = 373;
    stk500_devcode         = 0xe1;
    chip_erase_delay       = 500000;
    signature              = 0x1e 0x52 0x06;
    chip_erase             = "1010.1100--100x.xxxx--xxxx.xxxx--xxxx.xxxx";
    pgm_enable             = "1010.1100--0101.0011--xxxx.xxxx--xxxx.xxxx";

    memory "flash"
        size               = 8192;
        read               = "0010.0000--xxxa.aaaa--aaaa.aaaa--oooo.oooo";
        write              = "0100.0000--xxxa.aaaa--aaaa.aaaa--iiii.iiii";
    ;

    memory "signature"
        size               = 3;
        read               = "0 0 1 0 1 0 0 0  0 0 0 0 0 0 a1 a0  0 0 0 0 0 0 0 0  o o o o o o o o";
    ;
;

#------------------------------------------------------------
# AT89S51
#------------------------------------------------------------

part parent "89s52"
    desc                   = "AT89S51";
    id                     = "89s51";
    mcuid                  = 372;
    signature              = 0x1e 0x51 0x06;

    memory "flash"
        size               = 4096;
        read               = "0010.0000--xxxx.aaaa--aaaa.aaaa--oooo.oooo";
        write              = "0100.0000--xxxx.aaaa--aaaa.aaaa--iiii.iiii";
    ;
;
```

llinjupt, 2023-03-04:
@stefanrueger  Good news it works and bad news with warninings:

```
$ ./avrdude.exe -P COM180 -c avrisp -p AT89S52 -U flash:r:flash.bin:r
avrdude warning: a1 would normally be expected to be a9 [F:\BiTForest\avrdude7.1\win64\avrdude.conf:24]
avrdude warning: a0 would normally be expected to be a8 [F:\BiTForest\vrdude7.1\win64\avrdude.conf:24]

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e5206 (probably 89s52)
avrdude: reading flash memory ...

Reading | ################################################## | 100% 3.54s

avrdude: writing output file flash.bin

avrdude done.  Thank you.
```

stefanrueger, 2023-03-04:
This is the first time the project has come across an AVR where the address bits do not conform to their usual positions. When we create a PR to support AT89S51/52, we can try to suppress the warning for these (odd) parts. 

Before I do a PR, please also try to **write flash** and verify. If you have the AT89S51, please also try that part.

BTW, in order to migrate a part entry from an older avrdude version, you can normally put that into avrdude.conf, add new parameters needed for the new version (here, `prog_modes` and `mcuid`, though neither would have been needed I guess) and then run the new avrdude with
```
$ avrdude -pAT89S5[12]/s
```
This outputs migrated, canonical entries suitable for the new version.

I also noted that the flash r/w commands have one address bit too many: it's word addresses, not byte addresses for flash. This should be better:
```
#------------------------------------------------------------
# AT89S52
#------------------------------------------------------------

part
    desc                   = "AT89S52";
    id                     = "89s52";
    prog_modes             = PM_ISP | PM_HVPP;
    mcuid                  = 373;
    stk500_devcode         = 0xe1;
    chip_erase_delay       = 500000;
    signature              = 0x1e 0x52 0x06;
    chip_erase             = "1010.1100--100x.xxxx--xxxx.xxxx--xxxx.xxxx";
    pgm_enable             = "1010.1100--0101.0011--xxxx.xxxx--xxxx.xxxx";

    memory "flash"
        size               = 8192;
        read               = "0010.0000--xxxx.aaaa--aaaa.aaaa--oooo.oooo";
        write              = "0100.0000--xxxx.aaaa--aaaa.aaaa--iiii.iiii";
    ;

    memory "signature"
        size               = 3;
        read               = "0 0 1 0 1 0 0 0  0 0 0 0 0 0 a1 a0  0 0 0 0 0 0 0 0  o o o o o o o o";
    ;
;

#------------------------------------------------------------
# AT89S51
#------------------------------------------------------------

part parent "89s52"
    desc                   = "AT89S51";
    id                     = "89s51";
    mcuid                  = 372;
    signature              = 0x1e 0x51 0x06;

    memory "flash"
        size               = 4096;
        read               = "0010.0000--xxxx.xaaa--aaaa.aaaa--oooo.oooo";
        write              = "0100.0000--xxxx.xaaa--aaaa.aaaa--iiii.iiii";
    ;
;
```

stefanrueger, 2023-03-04:
@llinjupt ... and I suspect that the AT89S51 needs a different `stk500_devcode         = 0xe0;` (not 0xe1):
https://github.com/avrdudes/avrdude/blob/4019edf60d34e2dcfdafe6d00c33f58db8615781/src/avrdude.conf.in#L344-L345

llinjupt, 2023-03-05:
> @llinjupt ... and I suspect that the AT89S51 needs a different `stk500_devcode = 0xe0;` (not the inherited 0xe1):
> 
> https://github.com/avrdudes/avrdude/blob/4019edf60d34e2dcfdafe6d00c33f58db8615781/src/avrdude.conf.in#L344-L345

@stefanrueger , Yes, you are right that AT89S51 has its own devcode. And I have checked how it is used. Actually, in avrisp, it's only  used to detemine the rst polarity, and that's why the old avrdude.conf is accidentally ok for AT89S51:

`rst_active_high = (param.devicecode >= 0xe0);`

Reading back the flash and fill it up to the full falsh size:
`truncate.exe -s 8192 flash.bin`

All writing and reading opts are ok for AT89S52 with your lastest avrdude.conf. There is no AT89S51 on hand, so I have ordered some chips already. Maybe two or three days the feedback will be ready.

Another subtle problem is that I have noticed the lock bits in the datasheet as well as the picture you pasted above. For reading there are 3 bits LB3/2/1 and for writing there are 2 bits B1/B2. Isn't it a bit confusing? Would you like to config it too for both 51/52 in avrdude.conf? And I could verify at the same time.

stefanrueger, 2023-03-05:
> Wrting failed 

Thanks for testing. I suspect for writing to work the flash memory should specify the correct `page_size` > 1.

Maybe try, 2, 4, 8, 16, ...

It is a pity there does not seem to be an .atdf file (or .xml file as they were called in the early days) containing the necessary write delay etc parameters that need to go into the .conf file. The data sheet is very terse, too. 

In order for HVPP programming to work, other parameters would need to be specified, too, in avrdude.conf. Maybe try setting the other parameters like in @orbanp 's entry? Or copying parameters from "similar" other parts in avrdude.conf for this entry? You could try a few settings.  

I suspect AVRDUDE will be able to read the lock bits but won't be able to write them: The .conf file can set pretty much any SPI command (the old notation has a very versatile syntax!), but apparently the footnote in the data sheet says that one needs *several* writes of *different* SPI-commands to effect certain lock levels... That *procedure* isn't in the code base (and is probably not worth our while implementing!)

For now I wouldn't worry too much about the lock bits, but AVRDUDE has a vvv flexible syntax for these, too. Let's

stefanrueger, 2023-03-05:
> check the datasheet there is really an A12 

Ooops that's right. And the data sheet says "byte mode" for flash r/w. So, page size 1 would appear correct for flash. And one would need a12 in the flash read/write commands (as it's a byte address not a word address).

So, this makes me wonder. Has your .conf file also worked for writing in AVRDUDE v7.0 or any earlier AVRDUDE? If so let us know.

stefanrueger, 2023-03-05:
Here another one to try with some sensible other values set
```
#------------------------------------------------------------
# AT89S51
#------------------------------------------------------------

part
    desc                   = "AT89S51";
    id                     = "89S51";
    prog_modes             = PM_ISP | PM_HVPP;
    mcuid                  = 372;
    stk500_devcode         = 0xe0;
    chip_erase_delay       = 250000;
    signature              = 0x1e 0x51 0x06;
    timeout                = 200;
    stabdelay              = 100;
    cmdexedelay            = 25;
    synchloops             = 32;
    pollindex              = 4;
    pollvalue              = 0x69;
    predelay               = 1;
    postdelay              = 1;
    chiperasepulsewidth    = 15;
    programfusepulsewidth  = 2;
    programlockpolltimeout = 1;
    chip_erase             = "1010.1100--100x.xxxx--xxxx.xxxx--xxxx.xxxx";
    pgm_enable             = "1010.1100--0101.0011--xxxx.xxxx--xxxx.xxxx";

    memory "flash"
        size               = 4096;
        min_write_delay    = 4500;
        max_write_delay    = 4500;
        readback           = 0xff 0xff;
        mode               = 0x02;
        delay              = 10;
        blocksize          = 256;
        read               = "0010.0000--xxxa.aaaa--aaaa.aaaa--oooo.oooo";
        write              = "0100.0000--xxxa.aaaa--aaaa.aaaa--iiii.iiii";
        # Page mode is available in the device, but not implemented here.
    ;

    memory "signature"
        size               = 3;
        read               = "0 0 1 0 1 0 0 0  x x x x x x a1 a0  x x x x x x x 0  o o o o o o o o";
    ;
    # Lock bits are available in the device but not implemented here
;

#------------------------------------------------------------
# AT89S52
#------------------------------------------------------------

part parent "89S51"
    desc                   = "AT89S52";
    id                     = "89S52";
    mcuid                  = 373;
    stk500_devcode         = 0xe1;
    signature              = 0x1e 0x52 0x06;

    memory "flash"
        size               = 8192;
        # Page mode is available in the device but not implemented here
    ;
  # Lock bits are available in the device but not implemented here
;
```

llinjupt, 2023-03-06:
@stefanrueger 
> That procedure isn't in the code base (and is probably not worth our while implementing!)

I see. If the code base don't suppport, it's reasonable and acceptable.

> Has your .conf file also worked for writing in AVRDUDE v7.0 or any earlier AVRDUDE? If so let us know.

Yes. I have tried 6.3 and it worked well.

With your newest avrdude.conf, both reading and writing are passed for 89S52. And as soon as 89S51 is arrived I will give feedback. 

```
$ ./avrdude.exe -c avrisp -P COM245 -p AT89S52 -U flash:w:AT89S52.bin:r
avrdude warning: a1 would normally be expected to be a9 [F:\BiTForest\ATMEL\TEST_BINs\avrdude.conf:42]
avrdude warning: a0 would normally be expected to be a8 [F:\BiTForest\ATMEL\TEST_BINs\avrdude.conf:42]

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e5206 (probably 89S52)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file AT89S52.bin for flash
         with 8192 bytes in 1 section within [0, 0x1fff]
avrdude: writing 8192 bytes flash ...

Writing | ################################################## | 100% 24.57s

avrdude: 8192 bytes of flash written
avrdude: verifying flash memory against AT89S52.bin

Reading | ################################################## | 100% 8.19s

avrdude: 8192 bytes of flash verified

avrdude done.  Thank you.
```

llinjupt, 2023-03-06:
> And the data sheet says "byte mode" for flash r/w. So, page size 1 would appear correct for flash. And one would need a12 in the flash read/write commands (as it's a byte address not a word address).

Yes, the address space is exactly the same as the flash size. And I have noticed AT89S51/2 all support page mode read/write. Is it possible to support it? If so, it would be great for speed consideration.

orbanp, 2023-03-06:
Hello Everyone,

I appreciate the continuing help and discussions!

To reiterate, I have avrdude version 3.6 on my machine, ubuntu 20.04 with KDE frontend.

I have two hardware interfaces for programming. One of them is a pair of usbasp dongles. The usbasp interfces do work, I did upgrade the firmware in them,  programming one with the other. It was a couple of years ago when I obtained them on ebay, not sure about the firmware version.

The other interface is a simplified version of the STK200 dongle, a cable from the parallel port of the PC, with four simple wires for the reset, in and out data, and clock signals. 
The cable works. I could read/write (program) the AT89S52 flash with it using an old program under W10, the IspPgm30a, a program that has been long abandoned and has some issues.

Looking at the signals with a scope, using the usbasp interface, that hardware does not handle the reset signal. The AT89S52 has an active high reset, and it needs to be set to high to initiate ISP programming and the usbisp interface does not do that.

"My STK200" dongle shows some life when trying to program. 
It initiates contact with the processor, however the signature and data returned is invalid.
I defined a new programmer in the config file, same features as that of the STK200, with the reset signal inverted.
Using that hardware description the reset signal is raised high and I can see the clock signal and the input data signal changing, however the data is not what I would expect when forcing the flash read operation.
There is a correct program in the processor, it runs as it is supposed to, but the data received is a simple counter, the values increasing to 00x to FFx.
The processor is running at 12MHz, the ISP programming max speed could be 1/16-the of that clock speed, that is 750kHz.
The scope shows that the ISP clock speed is 500kHz.

Again, here the processor description in the config file that I have been using with avrdude 6.3:
```
#------------------------------------------------------------
# AT89S52
#------------------------------------------------------------
part
    id               = "8052";
    desc             = "AT89S52";
    signature        = 0x1E 0x52 0x06;
    chip_erase_delay = 500000;
    pgm_enable       = "1 0 1 0  1 1 0 0    0 1 0 1  0 0 1 1",
                       "x x x x  x x x x    x x x x  x x x x";

    chip_erase       = "1 0 1 0  1 1 0 0    1 0 0 x  x x x x",
                       "x x x x  x x x x    x x x x  x x x x";

    timeout      = 200;
    stabdelay      = 100;
    cmdexedelay      = 25;
    synchloops      = 32;
    bytedelay      = 0;
    pollindex      = 3;
    pollvalue      = 0x53;
    predelay      = 1;
    postdelay      = 1;
    pollmethod      = 0;

    memory "flash"
        size            = 8192;
        paged           = no;
        min_write_delay = 4000;
        max_write_delay = 9000;
        readback_p1     = 0xff;
        readback_p2     = 0xff;
        read            = "  0   0   1   0    0   0   0   0",
                          "  x   x   x a12  a11 a10  a9  a8",
                          " a7  a6  a5  a4   a3  a2  a1  a0",
                          "  o   o   o   o    o   o   o   o";

        write           = "  0   1   0   0    0   0   0   0",
                          "  x   x   x a12  a11 a10  a9  a8",
                          " a7  a6  a5  a4   a3  a2  a1  a0",
                          "  i   i   i   i    i   i   i   i";
   mode      = 0x21;
   delay      = 12;
      ;

    memory "signature"
        size            = 3;
        read            = "0  0  1  0   1  0  0  0   x  x  x  0   0  0 a1 a0",
                          "0  0  0  0   0  0  0  0   o  o  o  o   o  o  o  o";
      ;
  ; 
```
Thanks, Peter

stefanrueger, 2023-03-06:
@orbanp I suspect adding the correct `stk500_devcode` will make a difference (0xe0 for 89s50 and 0xe1 for 89s51); also try out PR #1310.

@llinjupt Thanks for confirming r/w now works for AT89S52. I've created PR #1310 so you can try that out (also later with the AT89S51). Note I added lock read; please check (though you probably won't see anything more interesting than 0xff, I suspect!)

>  AT89S51/2 all support page mode read/write. Is it possible to support it?

No, page mode is nonstandard so unless we receive a working PR we won't be able to support that. Same with lock write.
I guess(!) you *could* add memories lockmode0, lockmode1, lockmode2 and lockmode3 each with a different lock write command as detailed in the [data sheet](https://ww1.microchip.com/downloads/en/DeviceDoc/doc1919.pdf) and then write the lock modes in sequence `-U lockmode0:w:0xff:m -U lockmode1:w:0xff:m -U lockmode2:w:0xff:m ...` as many times as the lock bit(s) you want to set as alluded to in Note  1 on p27 (but no warranties my interpretation of the data sheet bricks your device :)

llinjupt, 2023-03-08:
@stefanrueger 

With your patch, AT89S51/2 are OK to read and write.

```
$ ./avrdude.exe -P COM245 -c avrisp -p at89s51 -U flash:w:AT89S51.bin:r

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e5106 (probably 89S51)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file AT89S51.bin for flash
         with 4096 bytes in 1 section within [0, 0xfff]
avrdude: writing 4096 bytes flash ...

Writing | ################################################## | 100% 12.29s

avrdude: 4096 bytes of flash written
avrdude: verifying flash memory against AT89S51.bin

Reading | ################################################## | 100% 4.10s

avrdude: 4096 bytes of flash verified

avrdude done.  Thank you.
```

> as many times as the lock bit(s) you want to set as alluded to in Note 1 on p27 (but no warranties if my interpretation of the data sheet bricks your device :)

Yes, your interpretation is correct. I have write and read back lock bits and the value is just like the datasheet said. BTW, these lock bits' fucntion is very different from morden MCUs and I don't think there are many people who are interested in it. I just put the write cmd here for reference. 

```
    memory "lock"
        size               = 1;
        read               = "0010.0100--xxxx.xxxx--xxxx.xxxx--xxxo.ooxx";
        write              = "1010.1100--1110.00ii--xxxx.xxxx--xxxx.xxxx";
        # Nonstandard lock write is available but not implemented
    ;

```

orbanp, 2023-03-09:
Hello Everyone,

I am glad to hear that at least for some of you the AT89S52 programming works with the avrisp.
It still does not work for me with the usbasp programmer!

I installed avrdude 7.1 with the LINUXSPI and PARPORT options added. 
I can read/write the ATmega8A on the usbasp dongle, I updated the firmware to the latest version.

Added to the config file stefanrueger's latest S51/S52 part descriptions.
(Also added the STK200 type programmers, that is why some of the warnings.)
When trying to read the AT89S52 I am getting the following errors:
```
$ sudo avrdude -p 89s52 -c usbasp -U flash:r:flash.hex -B 125kHz -F
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/usr/local/etc/avrdude.conf:457]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/usr/local/etc/avrdude.conf:459]
avrdude warning: a1 would normally be expected to be a9 [/usr/local/etc/avrdude.conf:2441]
avrdude warning: a0 would normally be expected to be a8 [/usr/local/etc/avrdude.conf:2441]

avrdude: set SCK frequency to 93750 Hz
avrdude error: program enable: target does not answer (0x01)
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000
avrdude error: Yikes!  Invalid device signature.
avrdude warning: expected signature for AT89S52 is 1E 52 06

avrdude done.  Thank you.
```
Interesting that first the device fails to initialize, but after a while it does, but the programmer still fails to get the correct signature.
Also, -F does not seem to have any effect.

Any clue what is wrong when trying to use the usbasp for programming the S52?

Thanks, Peter

llinjupt, 2023-03-10:
@orbanp, 
1. Check if there is stk500_devcode for AT89S51/2 in avrdude.conf. Trying PR #1310 is a better choice.
2. AFAIK, usbasp has a special firmware to support AT8xxx，did you update the right FW? 
Could you check if the rst is pulled high with a scope while do programming. 
3. If still not work, with -vvvv and paste the message.

orbanp, 2023-03-10:
Thanks for the help!

Yes, there is the stk500_devcode for AT89S51/2 in avrdude.conf:

`stk500_devcode         = 0xe1;`

I am not sure what do you mean by "PR [#1310](https://github.com/avrdudes/avrdude/pull/1310) is a better choice"?
That link brings me back to this same discussion!

For the usbasp dongles I used "usbasp.atmega8.2011-05-28.hex" file from Thomas Fischl's web site. The usbasp dongles I have have ATmega8A processors.

Checking with a scope the reset line is not pulled high when trying to program with the usbasp, as you suggested.

This is the reply with the -vvvv option, hopefully it gives some clue as to what is amiss:
```
$ sudo avrdude -p 89s52 -c usbasp -U flash:r:flash.hex -F -vvvv -B 125kHz

avrdude: Version 7.1-20230304 (4019edf)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /usr/local/etc/avrdude.conf
avrdude yywarning() [config.c:242] warning: mosi is deprecated, will be removed in v8.0, use sdo [/usr/local/etc/avrdude.conf:457]
avrdude yywarning() [config.c:242] warning: miso is deprecated, will be removed in v8.0, use sdi [/usr/local/etc/avrdude.conf:459]
avrdude yywarning() [config.c:242] warning: a1 would normally be expected to be a9 [/usr/local/etc/avrdude.conf:2441]
avrdude yywarning() [config.c:242] warning: a0 would normally be expected to be a8 [/usr/local/etc/avrdude.conf:2441]
         User configuration file is /root/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         Setting bit clk period        : 8.0
avrdude: usbasp_open("usb")
avrdude: seen device from vendor >www.fischl.de<
avrdude: seen product >USBasp<
         AVR Part                      : AT89S52
         Chip Erase delay              : 250000 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 4
         PollValue                     : 0x69
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash                   2    10   256    0 no       8192    1      0  4500  4500 0xff 0xff
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: usbasp_initialize()
avrdude: usbasp_transmit("USBASP_FUNC_GETCAPABILITIES", 0x00, 0x00, 0x00, 0x00)
         <= [01] [00] [00] [00] 
avrdude: usbasp_spi_set_sck_period(8e-06)
avrdude: try to set SCK period to 8e-06 s (= 125000 Hz)
avrdude: connected USBasp is not cabable of 3 MHz SCK
avrdude: set SCK frequency to 93750 Hz
avrdude: usbasp_transmit("USBASP_FUNC_SETISPSCK", 0x08, 0x00, 0x00, 0x00)
         <= [00] 
avrdude: usbasp_transmit("USBASP_FUNC_CONNECT", 0x00, 0x00, 0x00, 0x00)
avrdude: usbasp_program_enable()
avrdude: usbasp_transmit("USBASP_FUNC_ENABLEPROG", 0x00, 0x00, 0x00, 0x00)
         <= [01] 
avrdude usbasp_spi_program_enable() [usbasp.c:726] error: program enable: target does not answer (0x01)
avrdude main() [main.c:1246] error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000
avrdude main() [main.c:1349] error: Yikes!  Invalid device signature.
avrdude main() [main.c:1364] warning: expected signature for AT89S52 is 1E 52 06
avrdude: usbasp_close()
avrdude: usbasp_transmit("USBASP_FUNC_DISCONNECT", 0x00, 0x00, 0x00, 0x00)

avrdude done.  Thank you.
```
Thanks, Peter

llinjupt, 2023-03-11:
> For updating the firmware in the the usbasp dongles I used "usbasp.atmega8.2011-05-28.hex" file from Thomas Fischl's web site. The usbasp dongles I have have ATmega8A processors.

No. AFAIK, these firmwares don't support AT89 series.  Google or search in https://www.avrfreaks.net/forum may help you find  one working firmware hacked for AT89 series. If the souce is avaliable, search AT89/89S51/2 as keywords to see if if is supported. AT89 series need a high-level reset signal which  is not the same as other parts.

mcuee, 2023-03-11:
@orbanp 

I do not have the AT89S51/52 chip to test but you are not using PR #1310. You are using git main (https://github.com/avrdudes/avrdude/commit/4019edf60d34e2dcfdafe6d00c33f58db8615781)
PR #1310 has one more commit https://github.com/stefanrueger/avrdude/commit/6c8f7e10fd4e6d3ee9184e9e7375efcf64e17277 than git main.

You can build PR #1310 using the following command.
```
git clone https://github.com/stefanrueger/avrdude.git avrdude_pr1310
cd avrdude_pr1310
git checkout at89s5x
./build.sh
```

mcuee, 2023-03-11:
@llinjupt 

Just wondering which usbasp firmware are you using to program the AT89S51/52?

mcuee, 2023-03-11:
> No. AFAIK, these firmwares don't support AT89 series. Google or search in https://www.avrfreaks.net/forum may help you find one working firmware hacked for AT89 series. If the souce is avaliable, [...]

Indeed you are right. Good found this one, which has the firmware.
https://labprojectsbd.com/2020/04/19/make-your-own-8051-mcu-programmer-with-usbasp/

mcuee, 2023-03-11:
@stefanrueger 

I am thinking that we need to have a new programmer type called `usbasp_8051` to differentiate the above two types of FW (Chinese FW which supports 8051, original USBasp firmware or improvement firmware which does not support 8051).

As far as I know, the Chinese FW is not open as I can not find the source code.

BTW, I also can not find the USB HID FW source codes for the USBISP (ATmega88A) but that is anyway not supported by avrdude.

llinjupt, 2023-03-11:
> @llinjupt
> Indeed you are right. Thanks for the info. Google found this one, which has the firmware.
> https://labprojectsbd.com/2020/04/19/make-your-own-8051-mcu-programmer-with-usbasp/

I don't remember when I started fiddling with USBASP just like the Chinese clone one. And it didnt' seem verfy powerful and useful to me due to no labels of pins,  no open and things like this.  So I gave up and never touch it. I recommand avrisp and avrispmkii which is faster, open, easy to be hacking.

BTW I am working on a project named iLINK Infite (means my or your own programmer with infite abilities if you like hacking) which is still on going, trying to support most popular MCUs, including STM32/STM8/AVR(ISP,UPDI,TPI,PDI)/PIC8/32/MSP430xx, even programming SPI EEPROMs and etc. Moreover, it's tiny just like the STLINKv2 Clone in an aluminum case, while it's also an open platform for people to diy they own firmwares and upload to it without taking the risk of losing the original firmware. Maybe to your surprise, the ultimate aim is to make  it even cheaper than a cup of Starbucks and support most open soueced PC programs like avrdude.  Guess it's going to be released in the second quarter of this year.  If you are interested, I will let you know that time.

mcuee, 2023-03-11:
@llinjupt 

Nice one with your iLINK Infinite. I am interested for sure -- I am interested in PIC18/32, AVR and STM32 myself.

BTW, for usbasp, it is actually very popular for avrdude users (for Microchip/Atmel AVR chips, not for Microchip/Atmel 8051 chips which are not officially supported) due to good support from avrdude and available good alternative FW (eg: https://github.com/nerdralph/usbasp and https://github.com/dioannidis/usbasp).

orbanp, 2023-03-12:
Hello Everyone,

Some progress.

Installed avrdude avrdude_pr1310

Updated the firmware in the usbasp dongle. The dongle version I have is a Chinese, eBay purchased one, marked "USBASP V2.0 LC Technology". It uses the ATmega 8A processor running at 12MHz.
First I tried the Chinese firmware, I did not talk to the AT89S52, neither to the ATmega8A in the other dongle.

After that I used the firmware found here on github:
[https://github.com/mamut-tme/UsbAsp-flash_89S52/blob/master/README.md](url)
This combination of avrdude and firmware talks to the AT89S52, gets the signature correctly, however the data read back from the chip is garbled, it does not match what is in the processor.
Similarly, when writing the flash the signature is correct, but the correct data transfer fails, garbled data is written out.
Enclosing the printout with the -vvvv qualifier:
```
$ sudo avrdude -p 89s52 -c usbasp -U flash:w:code.hex -vvvv

avrdude: Version 7.1-20230306 (6c8f7e1)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /usr/local/etc/avrdude.conf
         User configuration file is /root/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
avrdude: usbasp_open("usb")
avrdude: seen device from vendor >www.fischl.de<
avrdude: seen product >USBasp<
avrdude: input file code.hex auto detected as Intel Hex
         AVR Part                      : AT89S52
         Chip Erase delay              : 250000 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 4
         PollValue                     : 0x69
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash                   2    10   256    0 no       8192    1      0  4500  4500 0xff 0xff
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: usbasp_initialize()
avrdude: usbasp_transmit("USBASP_FUNC_GETCAPABILITIES", 0x00, 0x00, 0x00, 0x00)
         <= [01] [00] [01] [01] 
avrdude: usbasp_spi_set_sck_period(0)
avrdude: auto set sck period (because given equals null)
avrdude: usbasp_transmit("USBASP_FUNC_SETISPSCK", 0x00, 0x00, 0x00, 0x00)
         <= [00] 
avrdude: usbasp_transmit("USBASP_FUNC_CONNECT", 0x00, 0x00, 0x00, 0x00)
avrdude: usbasp_program_enable()
[... 62 lines cut ...]
avrdude: usbasp_spi_cmd(0x20, 0x00, 0x00, 0x00) ...
avrdude: usbasp_transmit("USBASP_FUNC_TRANSMIT", 0x20, 0x00, 0x00, 0x00)
         <= [ff] [ff] [ff] [ff] 
avrdude: usbasp_spi_cmd() => 0xff, 0xff, 0xff, 0xff
avrdude: usbasp_spi_cmd(0x20, 0x00, 0x00, 0x00) ...
avrdude: usbasp_transmit("USBASP_FUNC_TRANSMIT", 0x20, 0x00, 0x00, 0x00)
         <= [ff] [ff] [ff] [ff] 
avrdude: usbasp_spi_cmd() => 0xff, 0xff, 0xff, 0xff
avrdude: usbasp_spi_cmd(0x20, 0x00, 0x00, 0x00) ...
avrdude: usbasp_transmit("USBASP_FUNC_TRANSMIT", 0x20, 0x00, 0x00, 0x00)
         <= [ff] [ff] [ff] [ff] 
avrdude: usbasp_spi_cmd() => 0xff, 0xff, 0xff, 0xff
avrdude: usbasp_spi_cmd(0x20, 0x00, 0x00, 0x00) ...
avrdude: usbasp_transmit("USBASP_FUNC_TRANSMIT", 0x20, 0x00, 0x00, 0x00)
         <= [ff] [ff] [ff] [ff] 
avrdude: usbasp_spi_cmd() => 0xff, 0xff, 0xff, 0xff
avrdude: usbasp_spi_cmd(0x20, 0x00, 0x00, 0x00) ...
avrdude: usbasp_transmit("USBASP_FUNC_TRANSMIT", 0x20, 0x00, 0x00, 0x00)
         <= [ff] [ff] [ff] [ff] 
avrdude: usbasp_spi_cmd() => 0xff, 0xff, 0xff, 0xff
avrdude: usbasp_spi_cmd(0x20, 0x00, 0x00, 0x00) ...
avrdude: usbasp_transmit("USBASP_FUNC_TRANSMIT", 0x20, 0x00, 0x00, 0x00)
         <= [ff] [ff] [ff] [ff]
```
Thanks, Peter

stefanrueger, 2023-03-15:
So it seems with PR #1310 avrdude supports AT89S51/2 with the avrisp programmer. So, the suggested avrdude.conf entries work in principle, ie, the SPI commands do what they should

Unclear to me why usbasp deosn't work. Has that ever worked in previous versions of avrdude? If so, which .conf entry?

mcuee, 2023-03-16:
> Unclear to me why usbasp deosn't work. Has that ever worked in previous versions of avrdude? If so, which .conf entry?

@stefanrueger 
There are different reports and different FW involved. I am actually thinking that we may not be able to support that effectively. 
1) Chinese FW is not open but is supposed to work with ProgISP which supports many 8051 chips and memory chips along with AVR chips.

2) This repo claims support of AT89S51 and AT89S52 with patched version avrudude 6.3.
https://github.com/mamut-tme/UsbAsp-flash_89S52/tree/master

@orbanp 
Have you tried the patched version of avrdude 6.3 mentioned in the above repo?

stefanrueger, 2023-03-16:
When PR #1310 gets merged this issue will be closed b/c AT89S51/2 is then supported by `-c avrisp`. 

@orbanp once you know more about reasons for your usbasp programmer not working for these parts, please open a dedicated issue for that, ideally with a suggested solution that you might find in the pointers that @mcuee has found above.

We rely on community contributions, particularly for non-standard parts.

orbanp, 2023-03-18:
Hello Everyone,

Just curious, where do I find the patched version of avrdude 6.3?
The link that mentioned it could not find the page.

I will see if I can put together a simple logic analyzer to test the programming interface with an arduino uno board that I have.
I downloaded PulseView, sigrok, and an arduino library that implements logic analyzer protocols.
Also found an "ISP decoder" for PulseView to display the communication in a more readable format.
Hopefully I can make the above work.

Another question. 
Whenever I sign into github it always asks and sends a verification code. It is a bit tiresome.
Is this normal, or just my browser is configured to "strictly"?
If that is the case, specifically, what do I need to allow in my browser to get in without verification?

Thanks, Peter

stefanrueger, 2023-03-18:
> browser is configured too "strictly"

Try allowing github to store cookies so that it can remember you.

mcuee, 2023-03-25:
I see, Sorry I an not help you much here as I do not have the AT89S51/52 target myself. You may want to ask in other places.

mcuee, 2023-03-26:
@orbanp 

But I have ordered AT89S52 chip and it should arive roughly in a month . I have USBASP/USBISP so I can test this once it is here.

llinjupt, 2023-03-30:
@orbanp @mcuee 
After glancing over those souce codes of USBasp firmware which claimed to support AT89S5x, I realize that none of them could support avrdude natively.

Spending some time doing hacking, I found it's easy to make the official USBasp version to support avrdude:

```
$ ./avrdude.exe -P usb -c usbasp -p AT89S52 -U flash:w:AT89S52.bin:r

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e5206 (probably 89S52)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file AT89S52.bin for flash
         with 8192 bytes in 1 section within [0, 0x1fff]
avrdude: writing 8192 bytes flash ...

Writing | ################################################## | 100% 9.76s

avrdude: 8192 bytes of flash written
avrdude: verifying flash memory against AT89S52.bin

Reading | ################################################## | 100% 3.19s

avrdude: 8192 bytes of flash verified

avrdude done.  Thank you.
```

Keep in mind that you only need to do 2 things:
1. Rst polarity of AT89S5x is opposite of AVR.
2. probe AT89S5x, the fourth byte will get 0x69

Since I didn't hack it on a AVR MCU, so I paste alike hacked code below for you to refer:

```
uchar ispEnterProgrammingMode() {
	uchar check;
	uchar count = 32;

       ISP_OUT |= (1 << ISP_RST); 
	while (count--) {
		ispTransmit(0xAC);
		ispTransmit(0x53);
		ispTransmit(0);
		check = ispTransmit(0);

		if (check == 0x69) {
			return 0;
		}

		spiHWdisable();

		/* pulse RST */
		ispDelay();
		ISP_OUT &= ~(1 << ISP_RST); /* RST low */
		ispDelay();
		ISP_OUT |= (1 << ISP_RST); /* RST high */
                ispDelay();

		if (ispTransmit == ispTransmit_hw) {
			spiHWenable();
		}
	}

	return 1; /* error: device dosn't answer */
}
```

And don't forget to update all other functions which do rst.

dioannidis, 2023-03-30:
Hi,

> Just wonder if you are willing to add the support in your usbasp firmware. You need to update the following code as per the suggestion from @llinjupt
> https://github.com/dioannidis/usbasp/blob/master/firmware/isp.c

OK I'll order those parts and after testing I'll add it

mcuee, 2023-03-31:
> Keep in mind that you only need to do 2 things:
>
>Rst polarity of AT89S5x is opposite of AVR.
> probe AT89S5x, the fourth byte will get 0x69

@llinjupt and @stefanrueger 

In this case, it is possible to just change avrdude (to change usbasp.c and avrdude.conf.in, create a variant of usbasp, say, usbasp_at89s) without changing the usbasp open source firmware?

llinjupt, 2023-03-31:
@mcuee 
IMHO, if the programmer doesn't support setting reset polarity, that's impossible to do so. And there are more other strange chips like S2051/AT89S8252/AT89S8253 that behave pretty differently, see below probe function which comes from one variant of USBasp.

```
uchar ispEnterProgrammingMode() {
	uchar check, check89 = 0;				//.kbv
	uchar count = 4;
	int i;
	// this is crap.   We need to do a proper RESET pulse before the start sequence.
	// we never want to insert false SCK pulses.
	for (i = 0; i < 256; i++) AT89buf[i] = 0xFF;
	while (count--) {
		reset_active_high = 0;			// all AVRs are active-low
		SS_inp();
		AT89mode = 0;
		spimode_1 = 0;
		ispReset();				// .kbv put into ISP mode for another go
		ispTransmit(0xAC);
		ispTransmit(0x53);
		check = ispTransmit(0xAA);
		check89 = ispTransmit(0x33);		// NOP on a S8252
		if (check == 0x53 && check89 == 0xAA) {	// AVR
			return 0;
		}
		reset_active_high = 1;			// AT89 are active-high
		SS_out();
		if (((PINC & (1<<PC2)) == 0) || (eeprom_read_byte(&AT89reset_low) == 0x55))
			reset_active_high = 0;		// SLOW jumper overrules
		AT89mode = 0;
		spimode_1 = 0;
		ispReset();				// .kbv put into ISP mode for another go
		SS_low();
		ispTransmit(0xAC);
		ispTransmit(0x53);
		check = ispTransmit(0xAA);
		check89 = ispTransmit(0x33);
		if (check == 0x53 && check89 == 0xAA) {			// AVR or S2051
			if (reset_active_high == 0) {
				SS_inp();
				return 0;		// AVR
			}
			AT89mode = 1;			// S2051
			AT89page_size = 32;
			return 0;
		}
		if (check89 == 0x69) {
			AT89mode = 4;			// AT89S52
			AT89page_size = 256;
			return 0;
		}
		check = ispTransmit(0x55);
		check89 = ispTransmit(0);
		if (check == 0x33 && check89 == 0x55) {	// AT89S8252
			AT89mode = 2;
			AT89page_size = 0;
			return 0;
		}

		spimode_1 = 1;				// only 8253 uses SPI mode #1
		ispReset();				// .kbv put into ISP mode for another go
		SS_low();
		ispTransmit(0xAC);
		ispTransmit(0x53);
		check = ispTransmit(0xAA);
		check89 = ispTransmit(0);
		if (check == 0x53 && check89 == 0xAA) {	// AT89S8253
			AT89mode = 5;
			AT89page_size = 64;
			return 0;
		}

		spimode_1 = 0;
		ispReset();				// .kbv put into ISP mode for another go
		SS_low();
		ispTransmit(0xAA);
		ispTransmit(0xAC);
		ispTransmit(0x53);
		check = ispTransmit(0);
		check89 = ispTransmit(0);
		SS_high();
		if (check == 0x53) {
			AT89mode = 3;			// AT89LP4052
			AT89page_size = 32;
			return 0;
		}

		spimode_1 = 0;
		ispReset();				// .kbv put into ISP mode for another go
		SS_low();
		ispTransmit(0xAA);
		ispTransmit(0x55);
		ispTransmit(0xAC);
		ispTransmit(0x53);
		check = ispTransmit(0);
		check89 = ispTransmit(0);
		SS_high();
		if (check == 0x53) {
			AT89mode = 6;			// AT89LP51RD2, AT89LP6440 etc
			AT89page_size = 64;
			return 0;
		}
	}

	SS_high();
	SS_inp();
	reset_active_high = 0; // all AVRs are active-low
	return 1; /* error: device dosn't answer */
}
```

stefanrueger, 2023-04-02:
I agree with @llinjupt: this is probably nothing AVRDUDE can do about it if the programmer cannot cope with different reset polarity. I suggest closing the issue again.

mcuee, 2023-04-02:
Okay, close for now.

closed 2023-04-02 by mcuee

dioannidis, 2023-04-21:
@mcuee, @llinjupt  

The AT89S51 and AT89S52 just arrived :) . Do I need a patched avrdude and conf file or the version from here https://github.com/mcuee/avrdude/releases/tag/v7.1 supports them ?

I'll try to find time next week to test the llinjupt's and if all is ok then I'll release the 1.11 version .

mcuee, 2023-04-21:
@dioannidis 

I think the idea is not to patch avrdude, but rather patch usbasp FW.

You can use the latest github action mingw-64 build of avrdude main.
https://github.com/avrdudes/avrdude/suites/12376320894/artifacts/657514449

mcuee, 2023-04-21:
@dioannidis 

You can also use my mingw64 build if you prefer.
https://github.com/mcuee/avrdude/releases/tag/snapshot_21Apr2023

dioannidis, 2023-04-21:
@mcuee 

I'll try both ;) ... Thx !

dioannidis, 2023-04-25:
@mcuee 

Both avrdude mingw64 builds ( yours and avrdude main) are working fine with the composite firmware ! Thx !

mcuee, 2023-04-25:
> @mcuee
> 
> Both avrdude mingw64 builds ( yours and avrdude main ) are working fine with the composite firmware ! Thx !

Wondeful.

I will try your modified FW once I get the AT89S52 chip.

dioannidis, 2023-04-25:
@llinjupt 

> @mcuee IMHO, if the programmer doesn't support setting reset polarity, that's impossible to do so. And there are more other strange chips like S2051/AT89S8252/AT89S8253 that behave pretty differently,

< snip >

Fortunately for the [USBasp improved](https://github.com/dioannidis/usbasp) firmware to support the AT89S51/2 ICs only the following lines has to be added to ispEnterProgrammingMode()  :

```
            /* AT89* Probe */
            
            ISP_OUT |= (1 << ISP_RST);      /* RST high */
            clockWait(5);                   /* 1.6 ms */
            
            spiTx(0xAC);
            spiTx(0x53);
            spiTx(0);
            check = spiTx(0);

            if (check == 0x69) 
                return 0;
```

dioannidis, 2023-04-25:
> > @mcuee

  < snip >

> I will try your[ modified FW](https://nephelae.eu/Firmware_ATMega8_12MHz_TPI_HIDUART_SNWRITE_AT89.zip) once I get the AT89S52 chip.

As I don't use MCS-51 MCUs, I adapted a basic blink assembly program and used the Microchip's [c51asm assembler](https://www.microchip.com/en-us/development-tool/C51ASM) to produce a 27 bytes binary .

Please if it's possible test it with a bigger binary to see if there is a problem .

dioannidis, 2023-05-06:
@mcuee 

> I will try your[ modified FW](https://nephelae.eu/Firmware_ATMega8_12MHz_TPI_HIDUART_SNWRITE_AT89.zip) once I get the AT89S52 chip.

That file will be deleted . Please use the firmware binaries from the [1.11 pre release](https://github.com/dioannidis/usbasp/releases/tag/v1.11) from today ! 

I re upload them, because they were corrupted :(  See [Corrupted firmware binaries in 1.11 pre Release assets](https://github.com/dioannidis/usbasp/issues/21)

mcuee, 2023-05-07:
@dioannidis 

Thanks. I just got AT89S51 chip. The first test shows it works well.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p at89s51
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e5106 (probably 89S51)

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p at89s51 -U .\entest.hex
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e5106 (probably 89S51)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\entest.hex for flash
         with 512 bytes in 1 section within [0, 0x1ff]
avrdude: writing 512 bytes flash ...
Writing | ################################################## | 100% 1.32 s
avrdude: 512 bytes of flash written
avrdude: verifying flash memory against .\entest.hex
Reading | ################################################## | 100% 0.61 s
avrdude: 512 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p at89s51 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e5106 (probably 89S51)
avrdude> dump flash
Reading | ################################################## | 100% 0.45 s
0000  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
0010  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
0020  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
0030  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
0040  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|
0050  61 7a 79 20 64 6f 67 0a  54 68 65 20 71 75 69 63  |azy dog The quic|
0060  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
0070  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
0080  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
0090  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
00a0  65 72 20 74 68 65 20 6c  61 7a 79 20 64 6f 67 0a  |er the lazy dog |
00b0  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
00c0  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
00d0  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
00e0  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
00f0  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|
avrdude> dump flash
Reading | ################################################## | 100% 0.45 s
0100  61 7a 79 20 64 6f 67 0a  54 68 65 20 71 75 69 63  |azy dog The quic|
0110  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
0120  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
0130  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
0140  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
0150  65 72 20 74 68 65 20 6c  61 7a 79 20 64 6f 67 0a  |er the lazy dog |
0160  54 68 65 20 71 75 69 63  6b 20 62 72 6f 77 6e 20  |The quick brown |
0170  66 6f 78 20 6a 75 6d 70  73 20 6f 76 65 72 20 74  |fox jumps over t|
0180  68 65 20 6c 61 7a 79 20  64 6f 67 0a 54 68 65 20  |he lazy dog The |
0190  71 75 69 63 6b 20 62 72  6f 77 6e 20 66 6f 78 20  |quick brown fox |
01a0  6a 75 6d 70 73 20 6f 76  65 72 20 74 68 65 20 6c  |jumps over the l|
01b0  61 7a 79 20 64 6f 67 0a  54 68 65 20 71 75 69 63  |azy dog The quic|
01c0  6b 20 62 72 6f 77 6e 20  66 6f 78 20 6a 75 6d 70  |k brown fox jump|
01d0  73 20 6f 76 65 72 20 74  68 65 20 6c 61 7a 79 20  |s over the lazy |
01e0  64 6f 67 0a 54 68 65 20  71 75 69 63 6b 20 62 72  |dog The quick br|
01f0  6f 77 6e 20 66 6f 78 20  6a 75 6d 70 73 20 6f 76  |own fox jumps ov|
avrdude> quit

avrdude done.  Thank you.
```

mcuee, 2023-05-07:
@orbanp 

Please give the FW from @dioannidis a try. It seems to work pretty  well. I am using latest avrdude git main (built using MSYS2 mingw64).

[9 more replies]

PR #1310 (2023-03-21): Support AT89S51, AT89S52, AT90S2323 and ATtiny22
Fixes #1304

In `avrdude.conf.in` this PR
  - Adds AT89S51 and AT89S52
  - Adds AT90S2323 and ATtiny22 by inheritance from AT902343
  - Changes AT90S4433 to inherit from AT90S2333 (same datasheet)

Changes in `avrintel.[ch]` are needed to add the two new `mucid`s for AT89S51/52; as a "bycatch" a new version of avrintel.c is deployed with corrections to UART interrupts

`config_gram.y` needed changing to suppress false warnings for the odd address bit positions for AT89S51/52 SPI read signature commands.