avrdudes/avrdude #1305: Linux apt install has older version that does not support recent AVR models

eprive, 2023-02-16:
I was able to use avrdude to update an ATmega4809. It has the required boot loader loaded for use over a serial port. That version ( 6.3-20201216 ) has been working well on a Windows platform for more than a year.

When I installed avrdude this week (2023) on a machine running Linux, it turned out to be too dated and unable to handle the recent ATmega4809. The chip is not recognized, and appears to contain both an older version of avrdude and the avrdude configuration file.

dl8dtl, 2023-02-16:
But, well, what exactly would you expect *us* to do against that situation?

First, there is no "Linux" as such: there are various distributions, you don't even write which one you are using.
Second: it's up to the maintainers of these distributions to package up a recent version of the tools. We cannot do anything on that – neither do we know all Linux distributions, nor the people behind them.

dl8dtl, 2023-02-16:
Besides: Windows does not ship AVRDUDE as part of the operating system either, so you somehow did install it there yourself. You are absolutely free to do the very same on your Linux system.

closed 2023-02-16 by dl8dtl

eprive, 2023-02-16:
sorry for the incomplete explanation. I was under the impression that it had nothing to do with the Linux variant.
Because the installation of avrdude was done with the appropriate procedure with sudo apt install. And then the application is retrieved from a central server. I had no idea that this also depends on the Linux variant ( I have Xubuntu 18 ).

mcuee, 2023-02-17:
Ubuntu/Debian usually ship with quite old version of avrdude. Same for many Linux distro. You can always build your own.

mcuee, 2023-02-17:
If you do not want to build from the source codes, you can also download the version distributed by the Arduino avrdude-packing project.
https://github.com/arduino/avrdude-packing/releases/tag/7.1-arduino.1

dl8dtl, 2023-02-17:
You could also try a nightly build if you want, see:

https://github.com/avrdudes/avrdude/wiki/Getting-Nightly-Builds-for-AVRDUDE