avrdudes/avrdude #1307: pin limit too low for LePotato RPi clone

djacobow, 2023-02-27:
I use avrdude to program attiny's directly from the GPIO on Raspberry Pi and similar devices.

One such device is the LePotato which uses sysfs GPIO numbers in the 400+ range.

I was able to make avrdude work, but it required this one-line patch:
```diff
diff --git a/src/libavrdude.h b/src/libavrdude.h
index b5ba923..ced6e6b 100644
--- a/src/libavrdude.h
+++ b/src/libavrdude.h
@@ -428,7 +428,7 @@ enum {
 #ifdef HAVE_LINUXGPIO
 /* Embedded systems might have a lot more gpio than only 0-31 */
 #undef PIN_MAX
-#define PIN_MAX     400 /* largest allowed pin number */
+#define PIN_MAX     1000 /* largest allowed pin number */
 #endif
 
 /** Number of pins in each element of the bitfield */
```

and a config like this:
```
programmer
  id    = "potato_pidog_gpio";
  desc  = "Program a PiDog from the LePotato GPIO lines";
  type  = "linuxgpio";
  reset = ~495;
  sck   = 491;
  mosi  = 488;
  miso  = 489;
;
```

I don't know how to make a PR but perhaps you guys can just bump up the value in your header?

mcuee, 2023-03-01:
@djacobow 

Try this guide to create a pull request.
https://docs.github.com/en/desktop/contributing-and-collaborating-using-github-desktop/working-with-your-remote-repository-on-github-or-github-enterprise/creating-an-issue-or-pull-request

stefanrueger, 2023-03-03:
@djacobow Increasing the max number of pins is uncontroversial. For the programmer entry in avrdude.conf it would be cool if there were one or two URLs to LePotato and/or PiDog to accompany that entry. Also, once you have tested the entry in your avrdude.conf, please use the output of `avrdude -cpotato_pidog_gpio/s` for the PR (this is the *canonical* representation).

djacobow, 2023-03-03:
It wasn't my intention to add an entry to the repo avrdude.conf because I think this is a pretty specialized application. Well, I should say that LePotato is a pretty popular Raspberry Pi alternative, and people will like to do AVR programming from it. But the PiDog is just one particular target with an AVR on it. The combination together force the pin assignment to the specifics of what I showed above, though _all_ the pins on the LePotato are > 400.

https://libre.computer/products/aml-s905x-cc/

https://github.com/djacobow/pidog2

If I make a PR, it would probably just be the one-line PIN_MAX change unless you think i should do otherwise.

djacobow, 2023-03-03:
Proposed PR: https://github.com/avrdudes/avrdude/pull/1308

closed 2023-03-04 by stefanrueger