avrdudes/avrdude #1311: Using serialupdi gives mismached error message

llinjupt, 2023-03-07:
@mcuee @stefanrueger 

When I was fiddling with serialupdi and tried to speed up with high baudrate, I got this error message.
It has nothing to do with ISP clock when using serialupdi which only supports baudrate setting.
Then I tried -B and it wouldn't take any effect on the uart baudrate.  I am not sure if it's a bug but at least it should show -b entry as well.

```
$ ./avrdude.exe -P COM245 -c serialupdi -p avr128da48 -U flash:r:test.bin:r -b 460800

avrdude error: UPDI link initialization failed
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude done.  Thank you.
```

mcuee, 2023-03-07:
Which USB to TTL adapter are you using?

There are some good info from here. For example, make sure to change the latency timer of FT232R to 1ms in Windows Device Manager.
https://github.com/SpenceKonde/AVR-Guidance/blob/master/UPDI/jtag2updi.md

mcuee, 2023-03-07:
For good HW design, I think you can use @MCUdude's design can be a reference.
https://d3s5r33r268y59.cloudfront.net/datasheets/27214/2022-06-28-07-47-24/SerialUPDI_R1_Schematic.pdf

mcuee, 2023-03-07:
@llinjupt 
On the other hand, I tend to agree with you that the error message for serialupdi can be improved to mention to reduce the baud rate.

llinjupt, 2023-03-07:
> @llinjupt On the other hand, I tend to agree with you that the error message for serialupdi can be improved to mention to reduce the baud rate.

Yes, my point is the error message instead of the circuit. When I set -b 115200/230400, the programmer worked well.

MCUdude, 2023-03-15:
How about something like this? This will throw relevant errors for `serialupdi` and bitbanged serial programmers like `ponyser`. @stefanrueger I can submit a PR if this is an acceptable solution

```diff
index 68b55d4..1a9c8f9 100644
--- a/src/main.c
+++ b/src/main.c
@@ -1244,11 +1244,16 @@ int main(int argc, char * argv [])
   init_ok = (rc = pgm->initialize(pgm, p)) >= 0;
   if (!init_ok) {
     pmsg_error("initialization failed, rc=%d\n", rc);
     if(rc == -2)
       imsg_error("the programmer ISP clock is too fast for the target\n");
     else
       imsg_error("- double check the connections and try again\n");
-    imsg_error("- use -B to set lower ISP clock frequency, e.g. -B 125kHz\n");
+
+    if (strcmp(pgm->type, "serialupdi") == 0 || strcmp(pgm->type, "SERBB") == 0)
+      imsg_error("- use -b to set lower baud rate, e.g. -b %d\n", baudrate? baudrate/2: 57600);
+    else
+      imsg_error("- use -B to set lower the bit clock frequency, e.g. -B 125kHz\n");
+
     if (!ovsigck) {
       imsg_error("- use -F to override this check\n\n");
       exitrc = 1;

```

mcuee, 2023-03-15:
@MCUdude 

Your changes look good to me.

stefanrueger, 2023-03-15:
@MCUdude Perfect!

MCUdude, 2023-03-15:
Note that I replaced "ISP clock" with "bit clock" since Avrdude doesn't have a good way to determine exactly which interface the programmer is using when communicating with the chip.

Something like `pgm->prog_mode` that could hold `PM_ISP`, `PM_JTAG`, `PM_TPI`, etc. would be useful!
IMO something like this doesn't belong in avrdude.conf, but could instead be set in every programmer's `initpgm` function.
Is there interest for such a PR?

Then the error message implementation could look like this:

```c
    if (strcmp(pgm->type, "serialupdi") == 0 || strcmp(pgm->type, "SERBB") == 0)
      imsg_error("- use -b to set lower baud rate, e.g. -b %d\n", baudrate? baudrate/2: 57600);
    else {
      imsg_error("- use -B to set lower the %s clock speed, e.g. -B 125kHz\n", 
        pgm->prog_mode == PM_ISP?  "ISP":
        pgm->prog_mode == PM_TPI?  "TPI":
        pgm->prog_mode == PM_PDI?  "PDI":
        pgm->prog_mode == PM_UPDI? "UPDI":
        pgm->prog_mode &  PM_JTAG | PM_JTAGmkI | PM_XMEGAJTAG | PM_AVR32JTAG? "JTAG":
        "bit");
    }
```