avrdudes/avrdude #1313: Does avrispv2 support TPI and PDI?

llinjupt, 2023-03-08:
Hi all.

In the programmer entry, it shows that avrispv2 can support TDI and PDI.

```
#------------------------------------------------------------
# avrispv2
#------------------------------------------------------------

programmer
    id                     = "avrispv2";
    desc                   = "Atmel AVR ISP v2";
    type                   = "stk500v2";
    prog_modes             = PM_TPI | PM_ISP | PM_PDI;
    connection_type        = serial;
;
```
But after trying it. it seems that's not true. From the code,  in stk500v2_initialize, only STK600/AVRISP_MKII/JTAGICE_MKII can support PDI and TPI with STK500v2 protocol. Any one could clarify it? Thanks in advance.

```
static int stk500v2_initialize(const PROGRAMMER *pgm, const AVRPART *p) {
  LNODEID ln;
  AVRMEM * m;

  if ((PDATA(pgm)->pgmtype == PGMTYPE_STK600 ||
       PDATA(pgm)->pgmtype == PGMTYPE_AVRISP_MKII ||
       PDATA(pgm)->pgmtype == PGMTYPE_JTAGICE_MKII) != 0
      && (p->prog_modes & (PM_PDI | PM_TPI)) != 0) {
```

mcuee, 2023-03-08:
AVRISP2 is AVRISP MKII, not STK500 v2.
https://www.microchip.com/en-us/development-tool/ATAVRISP2

mcuee, 2023-03-08:
BTW, you may want to use new version of avrdude.conf.

https://github.com/avrdudes/avrdude/blob/main/src/avrdude.conf.in#L1176-L1195
```
#------------------------------------------------------------
# avrispmkII
#------------------------------------------------------------

programmer
    id                     = "avrispmkII";
    desc                   = "Atmel AVR ISP mkII";
    type                   = "stk500v2";
    prog_modes             = PM_TPI | PM_ISP | PM_PDI;
    extra_features         = HAS_VTARG_READ;
    connection_type        = usb;
;

#------------------------------------------------------------
# avrisp2
#------------------------------------------------------------

programmer parent "avrispmkII"
    id                     = "avrisp2";
;
```

llinjupt, 2023-03-08:
> AVRISP2 is AVRISP MKII, not STK500 v2. https://www.microchip.com/en-us/development-tool/ATAVRISP2

@mcuee，thanks  for the quick response. But I meant avrispv2 instead of avrisp2 and it uses serial rather than usb.

mcuee, 2023-03-08:
I believe what you mentioned AVRISP V2 is just a FW upgrade for AVRISP V1, which is similar to STK500V2, minus some features. In the context of avrdude, it is basically the the same as `stk500v2`.

llinjupt, 2023-03-08:
Search the keyword 'avrispv2' in avrdude.conf.in, you will find the programmer entry:

```
#------------------------------------------------------------
# avrispv2
#------------------------------------------------------------

programmer
    id                     = "avrispv2";
    desc                   = "Atmel AVR ISP v2";
    type                   = "stk500v2";
    prog_modes             = PM_TPI | PM_ISP | PM_PDI;
    connection_type        = serial;
;
```
I wonder what it is and if it supports the prog modes as it shows in prog_modes.

mcuee, 2023-03-08:
Looks like a mistake to me. We may have to remove the PDI support for this one.

mcuee, 2023-03-08:
Care to send a PR to fix this? Thanks.

llinjupt, 2023-03-08:
Actually, I am interested that if it possible to implement the avrispmkII through serial instead of usb.
While I was hacking I found it would be blocked if it couldn't get expecting length of bytes during receiving through serial.  
All stk500v2 functions which call stk500v2_command always give  the last parameter sizeof(buf) which is longer than the expecting length of response.

For e.g. in stk500v2_getparm2 should call stk500v2_command like `stk500v2_command(pgm, buf, 2, 4)` rather than 
`(pgm, buf, 2, sizeof(buf))`.

When using usb, it just receives one usb frame and ignore the last parameter. I am not sure if it is a bug? 

```
static int stk500v2_getparm2(const PROGRAMMER *pgm, unsigned char parm, unsigned int *value) {
  unsigned char buf[32];

  buf[0] = CMD_GET_PARAMETER;
  buf[1] = parm;

  if (stk500v2_command(pgm, buf, 2, sizeof(buf)) < 0) {
    pmsg_error("unable to get parameter 0x%02x\n", parm);
    return -1;
  }

  *value = ((unsigned)buf[2] << 8) | buf[3];

  return 0;
}
```

mcuee, 2023-03-08:
> Actually, I am interested that if it possible to implement the avrispmkII through serial instead of usb

Which programmer will your serial AVRISP mkii clone base on?

Take note there are no STK500v2 FW upgrade for a long time, unlike AVRISP mkII FW.

llinjupt, 2023-03-08:
> Are you developing your own FW? What is the motivation? 
Today virtual com port (aka. CDC) is pretty popular.  It didn't require special driver and didn't bind to VID/PID.
If  these stk500v2 functions  can/will support serial, I am glad to port AVRISP mkII using CDC. If not, I will stop.

mcuee, 2023-03-08:
> > Which programmer will your serial AVRISP mkii clone base on?
> 
> Base on Lufa, http://www.fourwalledcubicle.com/AVRISP.php. Any other advice?
> 

LUFA is for USB AVRs and it has already the AVRISP mkII FW implementation for USB. 
https://github.com/abcminiuser/lufa/tree/master/Projects/AVRISP-MKII

Is this a hobby project or a commercial project? If it is a hobby project, go ahead. If it is for a commercial project, then I do not see the points since there are so many USB based AVRISP mkII clones out there. This is especially true since you are going to use USB AVR chips (because of LUFA), then there is even less points to use Virtual COM port.

> Today virtual com port (aka. CDC) is pretty popular. It didn't require special driver and didn't bind to VID/PID.
> If these stk500v2 functions can/will support serial, I am glad to port AVRISP mkII using CDC. If not, I will stop.

I believe you can achieve what you want with some modifications to `stk500v2.c` if you develop the right FW.

mcuee, 2023-03-08:
>  If it is a hobby project, go ahead.

Even if that is the case, I think it will be more fun if you go with something like an Arduino Uno clone (which has ATmega328P and typical CH340x based USB to Serial chip). 

Please also refer to the discussion here.
* https://github.com/avrdudes/avrdude/discussions/1088

llinjupt, 2023-03-08:
> Care to send a PR to fix this? Thanks.

See #1314 . And thanks for all other helpful information.

PR #1314 (2023-03-21): Remove PM_PDI flag for avrispv2 programmer
Fixes #1313 

In `avrdude.conf.in`, this PR removes PM_PDI flags for avrispv2 which does not support PDI protocol.