avrdudes/avrdude #1321: Usbtiny initialization never resets the part

petersenm, 2023-03-18:
Line 447 of usbtiny.c has this statement:

if (pgm->program_enable(pgm, p) >= 0)
   break;  <i.e. return as if everything is good>
// no response, RESET and try again

If the program_enable call fails the first time (return code = 0), the break will still be executed and the initialization will be considered complete even though the part never got the reset it needed.

That line should be:
if (pgm->program_enable(pgm, p) > 0)

Found while programming some ATtiny13A parts with a usbtiny-type programmer.  Some would work (e.g. that first call to pgm->program_enable actually worked and no reset was needed) but most would fail (they needed the reset but never got it).  After making the above change, it worked flawlessly on all parts.

stefanrueger, 2023-03-18:
Generally, libavrdude functions return a *negative* number on failure and *zero* on success (tough there are exceptions): https://github.com/avrdudes/avrdude/blob/4019edf60d34e2dcfdafe6d00c33f58db8615781/src/libavrdude.h#L30-L39

Checking against `> 0` then means neither failure nor success... 

There must be something else that's going on here.

petersenm, 2023-03-18:
> BTW, I think you mean to say line 477 and not 447. https://github.com/avrdudes/avrdude/blob/main/src/usbtiny.c#L476-L490

Oops, you're right.  I fumbled that line number.  It was probably too late to be posting bugs last night.  :)

> Generally, libavrdude functions return a _negative_ number on failure and _zero_ on success (tough there are exceptions):

Agreed.  But I think this is one of the exceptions:
program_enable (line 477) calls usbtiny_avr_op (line 769)
usbtiny_avr_op (line 279) returns the result of pgm->cmd (line 292)
pgm->cmd (aka usbtiny_cmd, line 515) returns either:

-1 on usb_in failure (line 526)
0 if (nbytes!=4) or (res[2] != cmd[1]) (line 532)
1 if (nbytes==4) and (res[2] == cmd[1]) (line 532)

That last one is the problem.  On SOME parts (not all), res[2] != cmd[1], causing the function to return 0, which ripples back up to the line in question on line 477.  I've run it with some carefully placed printf statements that confirm the above behavior and return values,

The problem crept up in v6.4 when this part of the code was rewritten. Looking at line 357 in v6.3:
if (! usbtiny_avr_op(pgm, p, AVR_OP_PGM_ENABLE, res)) {
That performs the same basic function as line 477 in v6.4+, namely if the call to usbtiny_avr_op fails, do a reset and try again (only one retry in the case of v6.3, though).  The logical test in v6.3 is "if the function returns zero, then reset".  That's different than in v6.4+ where the test is "if the function returns negative, then reset".  In my case, I occasionally get the return value of 0 and if the part doesn't get the subsequent reset, the operation doesn't work.

petersenm, 2023-03-18:
Failed run
```
avrdude -c iseavrprog -P usb -p attiny13 -B4 -v

avrdude: Version 7.1-20230304 (4019edf)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /usr/local/etc/avrdude.conf
         User configuration file is /home/petersen/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : iseavrprog
         Setting bit clk period        : 4.0
avrdude: usbdev_open(): found USBtinyISP, bus:device: 003:046
         AVR Part                      : ATtiny13
         Chip Erase delay              : 4000 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65     5     4    0 no         64    4      0  4000  4000 0x00 0x00
           flash                  65    10    32    0 yes      1024   32     32  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny-based programmer, https://iascaled.com
avrdude pgm_default() error: programmer operation not supported

avrdude: setting SCK period to 4 usec
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000
avrdude main() error: Yikes!  Invalid device signature.
avrdude main() error: expected signature for ATtiny13 is 1E 90 07
        Double check connections and try again, or use -F to override
        this check.

avrdude done.  Thank you.
```

Passing run (same code, same programmer, different ATtiny13 part):
```
avrdude: Version 7.1-20230304 (4019edf)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /usr/local/etc/avrdude.conf
         User configuration file is /home/petersen/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : iseavrprog
         Setting bit clk period        : 4.0
avrdude: usbdev_open(): found USBtinyISP, bus:device: 003:062
         AVR Part                      : ATtiny13
         Chip Erase delay              : 4000 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65     5     4    0 no         64    4      0  4000  4000 0x00 0x00
           flash                  65    10    32    0 yes      1024   32     32  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny-based programmer, https://iascaled.com
avrdude pgm_default() error: programmer operation not supported

avrdude: setting SCK period to 4 usec
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13)

avrdude done.  Thank you.
```

petersenm, 2023-03-18:
> I've been using the USBtiny programmer for years, but have never encountered this issue. @petersenm can you provide some restrictions on what I can do to reproduce the issue?
> 
> Thanks!

Same here.  Been using avrdude for 12+ years.  Up until last night, when I upgraded to v7.1, I had been using v6.3 without issues programming these boards (well, without this issue - there were some other issues in v6.3 regarding memory and chunk sizes, but that came into play only after initializing the AVR).

I wish I could boil it down to an easily reproducible error.  Since the stock avrdude works on some parts and not others, I suspect a marginal timing issue causing the zero return value above.  Anyway, here is my setup, if it helps:

Ubuntu 22.04
Compiled avrdude from source from GitHub (v6.3, v6.4, and v7.1 have all been tried)
ckt-avrprogrammer (iseavrprog programmer type, which is just a usbtiny in a different form factor)
ATtiny13A board (one of these: https://github.com/IowaScaledEngineering/ckt-irsense)

I'd say about 10% work the first time.  Once programmed using v6.3 (no failures), they all appear to work 100% even with stock (unmodified) v7.1.  That's why I suspect something marginal in the timing of the AVR that gets centered once the fuses are successfully blown.

stefanrueger, 2023-03-19:
Interesting observations @petersenm and thanks for reporting. I can confirm that

https://github.com/avrdudes/avrdude/blob/b7f48a08dbaec03790aa09e83cd04ca995187b57/src/usbtiny.c#L532-L533 defo doesn't conform to libavrdude conventions. It suspect it should be 
```
   return (nbytes == 4) && res[2] == cmd[1]? LIBAVRUDE_SUCCESS: LIBAVRDUDE_GENERAL_FAILURE;
```

> 10% work the first time with v6.4 or higher

Hmm, do the factory parts have the CKDIV8 fuse set? Prob yes. Might be that AVRDUDE is a bit fast for these? Perhaps downgrade the `-B` option. See whether AVRDUDE makes a suggestion to that effect once you change above return line.

petersenm, 2023-03-19:
> Perhaps downgrade the `-B` option.

I tried it with -B 32 (our standard for fresh, never before programmed parts) all the way up to -B 250 (or something in that range) with the same results.  Many AVR parts do need a slower clock before first programming, but that didn't appear to have any effect in this case.  I'll look at the suggested change above later this evening.  Thanks!

petersenm, 2023-03-19:
Confirmed that the proposed change to the return statement on line 532 does fix it (using the >= comparison on line 477) , both by inspection and empirically with actual hardware.  My version is slightly modified:

`return ((nbytes == 4) && (res[2] == cmd[1])) ? LIBAVRDUDE_SUCCESS : LIBAVRDUDE_GENERAL_FAILURE;`

Fixed a typo and a few extra parentheses to make it explicitly clear, but otherwise functionally the same.

petersenm, 2023-03-19:
I take back that previous comment slightly.  It fixes the specific problem with reading the device signature, but there are other calls to usbtiny_avr_op that expect a zero return code, it appears.  In hardware, programming fails silently at the erasing chip stage.  Presumably due to line 603:
https://github.com/avrdudes/avrdude/blob/b7f48a08dbaec03790aa09e83cd04ca995187b57/src/usbtiny.c#L603-L605
There may be other cases, too.  I didn't exhaustively check.  So while the change to the return statement might be more correct (no argument here) it will result in some additional changes needed in the code.  My original proposed change to the if statement is the least impact, though possibly less formally correct.

mcuee, 2023-03-19:
@petersenm 

What if you remove the `-B4` in your command line?

mcuee, 2023-03-19:
Here is what I get with and without `-B4` using an ATtiny13A (4.8MHz internal clock). I do not see issues with 9.6MHz.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbtiny -p t13a

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13a)

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbtiny -p t13a -B4

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000
avrdude error: Yikes!  Invalid device signature.
avrdude error: expected signature for ATtiny13A is 1E 90 07
        Double check connections and try again, or use -F to override
        this check.

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbtiny -p t13a -qqt
avrdude> dump lfuse
0000  29                                                |)               |

avrdude> dump hfuse
0000  eb                                                |.               |

avrdude> quit
```

mcuee, 2023-03-19:
@stefanrueger 

I think your patch is still good. Currently the error message is not good. The error message is much better with your patch.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbtiny -p t13a -B4

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000
avrdude error: Yikes!  Invalid device signature.
avrdude error: expected signature for ATtiny13A is 1E 90 07
        Double check connections and try again, or use -F to override
        this check.

avrdude done.  Thank you.
```

With your patch, it is much better now.
```
$ git diff
warning: in the working copy of 'CMakeSettings.json', CRLF will be replaced by LF the next time Git touches it
diff --git a/src/usbtiny.c b/src/usbtiny.c
index 5090f53..19a2e90 100644
--- a/src/usbtiny.c
+++ b/src/usbtiny.c
@@ -529,8 +529,9 @@ static int usbtiny_cmd(const PROGRAMMER *pgm, const unsigned char *cmd, unsigned
   msg_notice2("CMD: [%02x %02x %02x %02x] [%02x %02x %02x %02x]\n",
            cmd[0], cmd[1], cmd[2], cmd[3],
            res[0], res[1], res[2], res[3] );
-  return ((nbytes == 4) &&      // should have read 4 bytes
-         res[2] == cmd[1]);              // AVR's do a delayed-echo thing
+  //return ((nbytes == 4) &&      // should have read 4 bytes
+//       res[2] == cmd[1]);              // AVR's do a delayed-echo thing
+       return nbytes == 4 && res[2] == cmd[1]? LIBAVRDUDE_SUCCESS : LIBAVRDUDE_GENERAL_FAILURE;
 }

 int usbtiny_cmd_tpi(const PROGRAMMER *pgm, const unsigned char *cmd,

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1321 -c usbtiny -p t13a -B4

avrdude_issue1321 error: initialization failed, rc=-1
                  - double check the connections and try again
                  - use -B to set lower ISP clock frequency, e.g. -B 125kHz
                  - use -F to override this check

avrdude_issue1321 done.  Thank you.

```

mcuee, 2023-03-19:
@stefanrueger 

Another thing, `avrdude pgm_default() error: programmer operation not supported` is an eye sore. I am not so sure how to fix it. It will be good to get rid of it.

<details>
<summary> click for -vvvv log </summary>

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1321 -c usbtiny -p t13a -vvvv

avrdude_issue1321: Version 7.1-20230304 (4019edf)
                   Copyright the AVRDUDE authors;
                   see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                   System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

                   Using Port                    : usb
                   Using Programmer              : usbtiny
avrdude_issue1321: usbdev_open(): found USBtinyISP, bus:device: 001:012
                   AVR Part                      : ATtiny13A
                   Chip Erase delay              : 4000 us
                   RESET disposition             : dedicated
                   RETRY pulse                   : SCK
                   Serial program mode           : yes
                   Parallel program mode         : yes
                   Timeout                       : 200
                   StabDelay                     : 100
                   CmdexeDelay                   : 25
                   SyncLoops                     : 32
                   PollIndex                     : 3
                   PollValue                     : 0x53
                   Memory Detail                 :

                                                     Block Poll               Page                       Polled
                     Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                     ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                     eeprom                 65     5     4    0 no         64    4      0  4000  4000 0x00 0x00
                                                     Block Poll               Page                       Polled
                     Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                     ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                     flash                  65    10    32    0 yes      1024   32     32  4500  4500 0x00 0x00
                                                     Block Poll               Page                       Polled
                     Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                     ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                     lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                                     Block Poll               Page                       Polled
                     Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                     ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                     hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                                     Block Poll               Page                       Polled
                     Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                     ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                     lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                                     Block Poll               Page                       Polled
                     Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                     ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                     signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                                                     Block Poll               Page                       Polled
                     Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                     ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                     calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00

                   Programmer Type : USBtiny
                   Description     : USBtiny simple USB programmer, https://learn.adafruit.com/usbtinyisp
avrdude_issue1321 pgm_default() [pgm.c:226] error: programmer operation not supported

avrdude_issue1321: using SCK period of 10 usec
CMD: [ac 53 00 00] [00 00 53 00]
avrdude_issue1321: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00 s CMD: [30 00 00 00] [00 30 00 1e]
CMD: [30 00 01 00] [00 30 00 90]
Reading | #################                                  | 33% 0.01 s CMD: [30 00 02 00] [00 30 00 07]
Reading | ################################################## | 100% 0.02 s

avrdude_issue1321: device signature = 0x1e9007 (probably t13a)

avrdude_issue1321 done.  Thank you.
```

</details>

mcuee, 2023-03-19:
> So while the change to the return statement might be more correct (no argument here) it will result in some additional changes needed in the code. My original proposed change to the if statement is the least impact, though possibly less formally correct.

@petersenm 
I agree that the current code in git main is not correct. 
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbtiny -p t13a -B5

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000
avrdude error: Yikes!  Invalid device signature.
avrdude error: expected signature for ATtiny13A is 1E 90 07
        Double check connections and try again, or use -F to override
        this check.

avrdude done.  Thank you.
```

But if I compare your fix and the fix from @stefanrueger, it seems to me that his fix is better in terms of error message generated for my test case (failed with `-B5`).

Your proposed fix.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1321_org -c usbtiny -p t13a -B5

avrdude_issue1321_org: AVR device initialized and ready to accept instructions
avrdude_issue1321_org: device signature = 0x009007
avrdude_issue1321_org error: expected signature for ATtiny13A is 1E 90 07
                      double check chip or use -F to override this check

avrdude_issue1321_org done.  Thank you.
```

The fix from @stefanrueger 
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1321 -c usbtiny -p t13a -B5

avrdude_issue1321: AVR device initialized and ready to accept instructions
avrdude_issue1321: device signature = 0x00ff07
avrdude_issue1321 error: expected signature for ATtiny13A is 1E 90 07
                  double check chip or use -F to override this check

avrdude_issue1321 done.  Thank you.
```

mcuee, 2023-03-19:
Even though I can not reproduce the issue from @petersenm, we can see there is a bug in the git main. So I change the label to `bug`.

usbasp performs much better. It only fails with `-B1` which is expected.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p t13a -B4

avrdude: set SCK frequency to 187500 Hz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13a)

avrdude done.  Thank you.
```

Interesting, without @stefanrueger's fix, it is actually proceeding with signature read if I use `-c usbtiny` even when I am connecting usbasp, which is quite wrong.

```

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbtiny -p t13a -B4

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000
avrdude error: Yikes!  Invalid device signature.
avrdude error: expected signature for ATtiny13A is 1E 90 07
        Double check connections and try again, or use -F to override
        this check.

avrdude done.  Thank you.
```

With the fix from @stefanrueger, it is much better.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1321 -c usbtiny -p t13a -B4

avrdude_issue1321 error: initialization failed, rc=-1
                  - double check the connections and try again
                  - use -B to set lower ISP clock frequency, e.g. -B 125kHz
                  - use -F to override this check

avrdude_issue1321 done.  Thank you.

```

stefanrueger, 2023-03-19:
@petersenm @mcuee Pls try the proposed PR. usbtiny.c had clear-cut errors contravening the libavrdude return conventions. Thanks for reporting and testing.

mcuee, 2023-03-19:
> @petersenm @mcuee Pls try the proposed PR. usbtiny.c had clear-cut errors contravening the libavrdude return conventions. Thanks for reporting and testing.

Yes PR #1323 is good for me.

petersenm, 2023-03-19:
> @petersenm @mcuee Pls try the proposed PR. usbtiny.c had clear-cut errors contravening the libavrdude return conventions. Thanks for reporting and testing.

Looks to work well for me.  Thanks!

mcuee, 2023-03-19:
> > @petersenm @mcuee Pls try the proposed PR. usbtiny.c had clear-cut errors contravening the libavrdude return conventions. Thanks for reporting and testing.
> 
> Looks to work well for me.  Thanks!

Hmm, does it sort out your original issue?

petersenm, 2023-03-19:
> Hmm, does it sort out your original issue?

Seems to, yes.  A board that failed signature check with the original v7.1 code, now works without error (both signature check and programming) using PR #1323.