avrdudes/avrdude #1322: Specify optional start address when writing to memory using -U

MCUdude, 2023-03-18:
This could be a useful feature if Avrdude is used to write "raw data" or a binary file to a specific memory.
Currently, the -U flag always starts writing data from address 0 unless a hex file where addresses are pre-defined is used.

Let's say you want to write a 4-byte serial number to a fixed address in the flash memory, and this serial number is stored in a file.
Currently, you'll either have to somehow pipe this data into terminal mode, or that file needs to be a hex file that contains the fixed address.

How about another field(?) to the -U flag so that an optional start address could be specified as well?
Perhaps it can be used to flash hex files with a user-defined offset as well?

Then this would be possible:

```
# Write user program, serial_number.bin starting from flash address 0x10000
avrdude -cusbasp -patmega1284p -Uflash:w:userprog.hex:i -Uflash:w:0x10000:serial_number.bin:r

# Write stdin starting from flash address 0x11000
echo "0x01,0x02,0x03,0x04"| avrdude -cusbasp -patmega1284p -Uflash:w:userprog.hex:i  -Uflash:w:0x11000:-:r

# Write bootloader and OSCCAL byte to flash
 avrdude -cusbasp -patmega1284p -Uflash:w:bootloader.hex:i  -Uflash:w:0x1FE00:osccal_byte.hex:i
```

It can also be used to "burn in" an OSCCAL value into a specific flash memory address where the bootloader expects it to be, for instance with Urboot.

Is this something others also might find useful? Thoughts? Other ideas?

mcuee, 2023-03-18:
Will this implicit means `-D` (no erase)?

mcuee, 2023-03-18:
I think this is somewhat related to the following issue for Terminal Mode. 
* https://github.com/avrdudes/avrdude/issues/1180

MCUdude, 2023-03-18:
> Will this implicit mean -D (no erase)?

It would be very annoying if Avrdude erased the entire flash memory because you wanted to write a calibration byte to flash.
Or that you'd have to erase the entire flash memory if you wanted to turn a few 0's into 1's. Maybe @stefanrueger's caching and page erase implementation could be used to only erase and write to the bytes specified? If so, this would be very different from what the current -U flag does. I'm not sure what would be the best here...

stefanrueger, 2023-03-18:
Neat idea!

When I last looked into this (two years ago) I thought it was very difficult to implement. Having said this, I have had a lot more codebase expose since then, and now I believe it's merely difficult (some tricky offsets in the memory structure to consider), not very difficult. I even had sussed out a syntax `-U memory[.start[.len]]:rwv:file:type` as in  `-U flash.0x1234.12:r:file.hex:i` This syntax has the benefit that it is backwards compatible.

> flash hex files with a user-defined offset as well

I am at sizes and sevens with this. Code is normally not relocatable, flash data are. Hex files have an explicit syntax for offsets. The latter makes me think we should not allow the `[.start[.len]]` syntax for .elf and .hex files *unless they only have only one section that starts at 0*.

Fun fact: Those who follow the `-c urclock` development will know that it provides metadata that describe the "unused" area in flash between sketch and metadata. I have plans to propose creating a memory `store` on the fly after a sketch upload. Store basically is `flash.<addr_after_sketch>.<size_of_unused_flash>`. This would enable the user to upload application from a separate file together with the sketch as in:

``` sh
avrdude -c urclock -U myapp.hex -U store:w:mydata.txt:h
```

Also, users could read out the storage area of an existing avr with
``` sh
avrdude -c urclock -U store:r:mydata.txt:h
```

Or could move the current application data when uploading a new sketch
```
avrdude -c urclock -U store.0.1024:r:/tmp/mydata.txt:h -U newsketch.hex -store:w:/tmp/mydata.txt:h
```

The first `store` refers to the storage area of the flash *before* upload, the second to the relocated one *after* upload.

stefanrueger, 2023-03-18:
> Will this implicit mean -D (no erase)?

No! This should be for the user to decide. Current use of flash regions (eg, through .hex files) does not imply `-D` either.

> annoying if Avrdude erased the entire flash memory because you wanted to write a calibration byte to flash

Well, if the new calibration byte wants to set a bit in flash then with classic parts (that don't have page erase) one would have to delete the entire flash and write again, as it is a NOR memory (unless you use a bootloader, where the SPM command can do a page erase). This is what the terminal mode does and this is what the whole cacheing business is all about.

MCUdude, 2023-03-18:
>> flash hex files with a user-defined offset as well

> I am at sixes and sevens with this. Code is normally not relocatable, flash data are. 

The idea was not to specify a write offset for a compiled program; this would obviously break the application. The idea was to use this feature with hex files that only contained "data". 

Another great thing that specifying a start (and end address?) provides is a way to only dump a section of the memory, for instance only the hardware bootloader section of a chip, and save it as a hex file. The same would obviously work for any memory that could be read/written to.

stefanrueger, 2023-03-18:
It's a difficult design decision.  On second thoughts, an ordinary sketch (code) might well be one section starting from 0. And it would be wrong to offset that. Looks like there are *two different* meanings for the memory interval
  - offset and restrict me: (example raw file with flash data)
  - Intersect me: (examples: read bootloader; write only the sketch, not the bootloader from a combined input .hex file)

The more I think about it, the more I think the rule should be simple. How about *offset and restrict me* for file formats that know nothing about sections and nothing about start addresses (example raw data, decimal, octal or hexadecimal sequences of numbers) and *intersect me* for .hex and .elf files that know about sections and offsets. Soooooo

`-U flash.0x3000.200:w:file.txt:h` would offset the data in file.txt to 0x3000 and restrict writing to max 200 bytes (even if the file has more data) but `-U flash.-384.-2:w:bootloader.hex:i` would take a bootloader.hex file that might be one section [0x7d80, 0x7fff]  of 384 bytes and *intersect* that with the `flash:-384:-3` interval being [0x7d80, 0x7ff**e**] assuming 32k, so copy over the bootloader *except* the version number in the last byte.

stefanrueger, 2023-03-28:
> What about the following simple idea? We have too many `:` right now already.
> `[startaddr:endaddr]` or `[startaddr:length]`

`[abc]` is a shell glob like `*` and `?`. Might need quoting in quite a few shells. I still like` -U flash.0x3000.200:w:file.txt:h` with the *same* syntax as "read/write addr len` in the terminal (negative numbers etc). And note  it's `.` not `:` that separates the addr and len bits in the memory part.

stefanrueger, 2023-05-27:
I thought about changing `-U` syntax to provide better granularity for reading or writing memories, but it occurred to me that, really, the best place to do so is the versatile terminal `write` command, which now can also write the contents of files to a particular memory address. In not so distant future there will be a terminal `save` command that would allow writing parts of a memory to a file.

So, rather than inventing a new syntax for fine-granular memory r/w of `-U`, going to the pain of implementing that functionality for `-U` again (writing to flash requires caching, potentially a read-chip-erase-chip-write-chip cycle all that kind of stuff, I suggest implementing `-T "terminal command"` instead. See [here](https://github.com/avrdudes/avrdude/issues/1327#issuecomment-1565394252) for more details of that idea.

mcuee, 2023-05-27:
This new feature (to be able to mix -U and -t) sounds like a great enhancement!

PR #1373 (2023-06-04): Implement -T cmd to mix terminal commands with -U commands
A new option `-T cmd` slots a terminal command line `cmd` into an intermixed list of `-T` terminal commands and `-U` memory operations.  The terminal command line is executed when it is its turn in the mixed list. This is done by externalising the new terminal function `terminal_line()`. As this could be *very* useful for simple high-level `libavrdude` applications, this PR moves `term.c` into `libavrdude`. As the list of to be processed operations can become long, this PR experiments with putting a paragraph before processing every `- [... cut]
closed 2023-06-04 by stefanrueger