avrdudes/avrdude #1327: Command line option improvement ideas

mcuee, 2023-03-24:
From here.
https://github.com/avrdudes/avrdude/issues/1325#issuecomment-1481869070

@dl8dtl mentions:
> The abbreviated forms (up to just -U filename) have been mainly added to the parser to make it easier to manually type these command lines.

Actually I like the abbreviated forms like `-U`. In fact, I would like to have another one for reading of flash memory.

A few potential improvement for the command line options.
1) We are not using `-R` yet, so we can use `-R` as the short form for flash reading. 

2) Getting a bit further, we can use `-RE` for reading of EEPROM. It is probably okay to use `-UE` for writing of EEPROM if we go along this line.

3) adding `-h` as the alias for `-?`.

4) mention that `-vv` is the same as `-v -v` and `-qq` is the same as `-q -q` in the help text.

5) mention that `-B` can use either period (us) or frequency (kHz).

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -h
C:\work\avr\avrdude_test\avrdude_bin\avrdude.exe: unknown option -- h
Usage: avrdude [options]
Options:
  -p <partno>                Specify AVR device
  -p <wildcard>/<flags>      Run developer options for matched AVR devices
  -b <baudrate>              Override RS-232 baud rate
  -B <bitclock>              Specify bit clock period (us)
  -C <config-file>           Specify location of configuration file
  -c <programmer>            Specify programmer type
  -c <wildcard>/<flags>      Run developer options for matched programmers
  -A                         Disable trailing-0xff removal from file and AVR read
  -D                         Disable auto erase for flash memory; implies -A
  -i <delay>                 ISP Clock Delay [in microseconds]
  -P <port>                  Specify connection port
  -F                         Override invalid signature or initialisation check
  -e                         Perform a chip erase
  -O                         Perform RC oscillator calibration (see AVR053)
  -U <memtype>:r|w|v:<filename>[:format]
                             Memory operation specification
                             Multiple -U options are allowed, each request
                             is performed in the order specified
  -n                         Do not write anything to the device
  -V                         Do not verify
  -t                         Enter terminal mode
  -E <exitspec>[,<exitspec>] List programmer exit specifications
  -x <extended_param>        Pass <extended_param> to programmer
  -v                         Verbose output; -v -v for more
  -q                         Quell progress output; -q -q for less
  -l logfile                 Use logfile rather than stderr for diagnostics
  -?                         Display this usage
```

stefanrueger, 2023-05-27:
I miss a way to run single terminal command lines interslotted with `-U ...` commands. Hence, thought of implementing `-T "<terminal command>"` and have any intermixed `-U` and `-T`  processed strictly in the order in which they appear.  Right now we can only pipe terminal commands into a CLI call, and these are all carried out *before* any of the `-U commands`. With this, one could, for example, load an application use a terminal line to write config-data to flash and then store both as new hex file all on the command line but using the power of the terminal:
```
$ avrdude -c dryrun -p m328p \
     -U flash:w:app.hex \
     -T 'write flash 0x7000 0xc0cac01a 0xcafe "secret Coca Cola recipe"' \
     -U flash:r:app+data.hex:I 
```
Or one could burn a bootloader and configure the device using mnemonics in one go:
```
$ avrdude  -c dryrun -p m168p \
  -U flash:w:urboot.hex \
  -T "config` blb1=spm_disabled_in_boot; config bootrst=boot; config bootsz=bs_128w"
```
Or show how the device is configured w/o entering terminal mode per se:
```
$ avrdude -cusbasp -pm328p -Tc
```

mcuee, 2023-05-27:
This sound like a great enhancement!

stefanrueger, 2023-06-01:
> This new feature (to be able to mix -U and -t) sounds like a great enhancement!

See PR #1373. 

The option parsing follows a traditional UN*X style, so some of the suggested changes are not advisable/necessary

 - Virtually all CLI programs allow merging of no-argument options into one string: `-abc` is expected to be a shorthand for `-a -b -c`, which is not unique to AVRDUDE, so no need to explain.
 - There is a school of though that says a program should only introduce new options for new functionality, not for abbreviations of other options,  as there are only some 62 possible characters, so giving out options `R` and `E` might restrict us later on
 - Options are parsed as stand-alone items, so combos such as `-UE` cannot be parsed with the library function getopt()

closed 2023-06-04 by stefanrueger