avrdudes/avrdude #1328: WCH CH340 serial port issue under Windows: can't set com-state for "\\.\COMx

mcuee, 2023-03-25:
I think this is not an avrdude issue but an WCH CH340 driver issue. 

Still it is good to put some info here so that people can reference to it, since there are quite some similar issues posted online.

I was playing with CH341A yesterday and install the latest CH341SER driver for CH340/CH341. Then I was encountering the above issue. Reverting to an old version sorted out the issue for me -- I am using version 3.5 (3.5.2019.1) and it is working fine.

Problematic driver version:
http://www.wch-ic.com/downloads/CH341SER_EXE.html (version 3.8, uploaded on 2023-03-16, size 642KB)

```
; CH341SER.INF
; Driver for CH340/CH341 (USB=>SERIAL chip) V3.8
; WDM&VXD for Windows 98/Me/2000/XP/Vista/7/8/8.1/10/11/SERVER 2003/2008/2012/2016/2019/2022
; Copyright (C) WCH 2001-2023
;

[Version]
Signature   = "$Chicago$"
Class       = Ports
ClassGuid   = {4D36E978-E325-11CE-BFC1-08002BE10318}
Provider    = %WinChipHead%
DriverVer   = 02/11/2023, 3.8.2023.02
CatalogFile =CH341SER.CAT
```

mcuee, 2023-03-25:
Similar report:
https://arduino.stackexchange.com/questions/55301/avrdude-ser-open-cant-set-com-state

mcuee, 2023-03-25:
The following is a different issue but with the same symptom. It is not using CH340 but the original ATmega16U2 USB to Serial FW.
https://forum.arduino.cc/t/ser_open-can-t-set-com-state-for-com4-changes-uno-not-detected-solved/659323/8

closed 2023-03-25 by mcuee

mcuee, 2023-03-25:
I found a solution for using the latest version 3.8 driver (CH341SER 3.8 version). I need to click the check box in the advanced settings --> `Enabling the Serial Port Enumerator (SerEnum)`.

[image]

mcuee, 2023-03-27:
Somehow this is dependent on the PC.
I do not need to enable serenum to get the thing working on my other laptop running Windows 10.

mcuee, 2023-07-18:
I have to use older version of the driver to work reliably in the end. The new version just does not work properly.

MCUdude, 2023-07-19:
In case someone else are having the same issue, what driver version are you using (that works), and where did you download it from?

per1234, 2023-07-19:
Since I was not able to find a way to source an older version of the driver directly from the WCH website, I am recommending sourcing it from a SparkFun tutorial since I consider SparkFun to be a trustworthy source. The instructions are here:

https://forum.arduino.cc/t/arduino-nano-upload-error/1146296/14

mcuee, 2023-07-19:
Yes, SparkFun version works fine.

mcuee, 2023-07-19:
I will carry out a bit more tests to see which version does not work and which versions work. I believe all the older version (3.7 and earlier) works but I need to double check.

mcuee, 2023-07-19:
Another reliable place to download is from Microsoft. They have version 3.8 (problem version), version 3.5 and version 3.4 (both are working).
https://www.catalog.update.microsoft.com/Search.aspx?q=CH340

mcuee, 2023-10-29:
Interestingly I do not seem to have issues using serialupdi with latest driver.