avrdudes/avrdude #1329: Query a programmer to figure out its current clock speed

MCUdude, 2023-03-26:
At the moment, Avrdude can change the clock speed using the -B flag, but there's no way to query the programmer to reveal what the current ISP clock is set to.

The default programmer clock speed varies significantly between different hardware. The USBasp for instance, has a default clock of ~750kHz and is too fast for a stock AVR running at 1 MHz. 

Several genuine Atmel/Microchip programmers store the last used clock frequency in its non-volatile memory, meaning the user has to use the -B flag to restore it to its default speed. None of this is evident to the users really.

It would be great if Avrdude could be aware of the current ISP clock speed, regardless if the user is using the -B flag or not.
Then, if the clock speed is too fast for the target, Avrdude could print the current clock speed.

Perhaps the current clock speed could be viewed in terminal mode as well using the `sck` command without passing a second argument?

MCUdude, 2023-03-26:
I've played around a little and implemented `pgm->get_sck_period(pgm)` for the USBasp and USBtiny programmer, and it seems like there's no way to read the actual clock speed from the programmer. This isn't really an issue on the USBtiny, because the firmware versions that exist out there standardize on a 10us clock. The USBasp however, operates at least two or more default speeds. Some run at 750kHz, others run at 375kHz, and AFAIK there's no way for Avrdude to tell.

I'll throw this one on the back burner for now.

mcuee, 2023-03-27:
> I've played around a little and implemented `pgm->get_sck_period(pgm)` for the USBasp and USBtiny programmer, and it seems like there's no way to read the actual clock speed from the programmer. [...]

1) USBasp/USBISP (newer version firmware, for ATmega8/8A/48/88) is supposed to have adaptive SCK.
https://github.com/nerdralph/usbasp/releases
https://github.com/dioannidis/usbasp

From version 1.07 (or maybe 1.06 version), it is using a default SCK clock of 1.5Mhz and automatic SCK slowing if target does not respond. PORTD is left as input, so this firmware also works with USBISP modules. ( nerdralph )

From version 1.08 the firmware is WCID compliant, meaning it should work on Windows without any driver or .inf install.

2) USBasp original FW (2011-05-28 version) from https://www.fischl.de/usbasp/
It is said to have SCK option to support targets with low clock speed (< 1,5MHz).

3) USBasp/USBISP China version FW (for ATmega8/8A) which may or may not claim AT89S51/52 support
avrdude will have a warning about SCK, which may or may not be the right warning.

4) Take note that USBASP/USBISP China FW which claims 8051 support (either for ATmega8/8A or ATmeg48/88/88P) is usually using USB HID and not compatible with avrdude. The USB VID/PID is often 0x03EB/0xc8b4 or similar, and not 0x16c0/0x05dc.

mcuee, 2023-06-20:
@MCUdude and @stefanrueger 

I am thinking one way to sort out this issue is to add a parameter for the programmer to advertise that it has the capability to report the bit clock and then implement the enhancement only for those programmers.

mcuee, 2023-11-02:
@MCUdude and @stefanrueger

Just wondering if you have some comments about this issue. Thanks.

Or maybe we can just close this issue as not planned for now.

Ho-Ro, 2023-11-05:
> At the moment, Avrdude can change the clock speed using the -B flag, but there's no way to query the programmer to reveal what the current ISP clock is set to.

This is not supported for stk500 and stk500v2, even if both protocols have the parameter `Parm_STK_SCK_DURATION` (0x89) for v1 or ` PARAM_SCK_DURATION` (0x98) for v2.

```sh
$ avrdude-git -c stk500v1 -pt85 -vvvv -B1  -l  B1
$ avrdude-git -c stk500v1 -pt85 -vvvv -B10 -l  B10
$ diff -u B1 B10
--- B1  2023-11-05 17:36:34.973659722 +0100
+++ B10 2023-11-05 17:36:43.433718475 +0100
@@ -8,7 +8,7 @@
 
              Using Port                    : /dev/ttyUSB0
              Using Programmer              : stk500v1
-             Setting bit clk period        : 1.0
+             Setting bit clk period        : 10.0
 avrdude-git: ser_send: 0 [30]   [20]
 avrdude-git: ser_send: 0 [30]   [20]
 avrdude-git: ser_send: 0 [30]   [20]
```

Ho-Ro, 2023-11-13:
stk500v2 already sets the `-B` value, for stk500v1 I added the same behaviour with https://github.com/Ho-Ro/avrdude/commit/373e2e0546e8966c713a177caef02a3dafbf52bf of my #1566.

closed 2023-12-18 by stefanrueger