avrdudes/avrdude #1334: Build for Bob3

drmpjz, 2023-04-02:
I hope I am stepping not on anybodies toes here....
My wife works in STEM education and uses the Bob3 "robot" for basic programming introduction.
They have a Linux version of their SW at https://www.bob3.org/de/linux .  This uses the library part of avrdude to do the actual transfer of data to the MCU. Not sure if this was/is intended by the project to be used this way. 
Anyhow building avrdude for this purpose I found two issues and propose the following changes.

a) Move the avrintel functions into the library (as opposed to directly linking it into the executable
--- src/CMakeLists.txt.orig     2023-03-03 20:05:29.274700258 +0100
+++ src/CMakeLists.txt  2023-03-05 12:44:47.583333162 +0100
@@ -145,6 +145,8 @@
     avrftdi_private.h
     avrftdi_tpi.c
     avrftdi_tpi.h
+    avrintel.c
+    avrintel.h
     avrpart.c
     bitbang.c
     bitbang.h
@@ -278,8 +280,6 @@
     main.c
     term.c
     term.h
-    avrintel.c
-    avrintel.h
     developer_opts.c
     developer_opts.h
     developer_opts_private.h

b) Would you be willing to accept a patch to build.sh that makes build_flags a parameter that can be passed to the skript? I am generating an RPM for avrdude and want to install to /usr (resp. /etc for the config file).

stefanrueger, 2023-04-02:
Great to hear from users that actually deploy libavrdude! We are making (modest) strides to convert that into a proper library, which (small advance warning) *might* entail some incompatibility in future, but yes, we are keen to improve libavrdude.

Your Change a) is pushing at open doors; only recently it occurred to me that avrintel should be in the library as one would get unresolved references otherwise when using some functionality in libavrdude and it would give an instant knowledge base about core properties for parts without having to manage a relatively complex `avrdude.conf`. So, I for one, am all for that. Probably also needs a corresponding change in `Makefile.am`. As to b) it is not clear to me what the ramifications of that are, and others in the team (@mariusgreuel?) may have a better view of this. But yes, feel free to submit PRs and the team will look at this.

PS: BobDude sounds like a cool project (bob = break-out board? or from the phrase "... and Bob's your uncle"?)

stefanrueger, 2023-04-09:
I have just created a PR #1341 that addresses a)

> b) Would you be willing to accept a patch to build.sh that makes build_flags a parameter that can be passed to the skript? I am generating an RPM for avrdude and want to install to /usr (resp. /etc for the config file).

We haven't seen any objection from the other dudes here, so yes please submit a PR for the team to consider.

drmpjz, 2023-04-16:
I've created an initial pull request for this just now. 
Notes: 

- This addresses my use case. If desired I could easily extend it to other options currently hard coded in the script.
- From what I've read long options might not work on a plain vanilla Mac, but short options still should. I can't test this, since I don't have a Mac around.  Worst case one could move the code into the Linux specific section of the script.

PR #1341 (2023-04-17): Move avrintel into libavrdude
Fixes #1334

Avrintel should be in the library as one would get unresolved references otherwise when using some functionality in libavrdude. It also provides an instant knowledge base about parts (and programmers) for any project linking to libavrdude without having to manage a relatively complex `avrdude.conf`.
closed 2023-04-17 by stefanrueger