avrdudes/avrdude #1335: Signed releases?

freswa, 2023-04-04:
As I understand this project moved over from GNU's Savannah. Over there every release provided a signature for the release tarball. Are there any plans to provide them here as well?

dl8dtl, 2023-04-04:
Well, Savannah required signed releases before they would even consider distributing them.
But then, the entire release procedure there was completely different: I built everything on my machine at home, signed the files, and uploaded them. At Github, the releases are being produced by the CI process. I don't know too much about the options here, but if it required to upload the signing key onto some Github infrastructure, I don't think this would give you any improved security.
I (or someone else) could probably just upload a detached signature post-release, manually. I just did that for the 7.1 source release packages (.zip and .tar.gz). Is this what you want?

freswa, 2023-04-04:
This is exactly what we need. Thank you!
Do you plan to continue providing those signatures in the future as well?

dl8dtl, 2023-04-04:
I (or someone else) just have to remember to do it. ;-)

mcuee, 2023-04-05:
> Guess it's time to write up a release checklist somehow somewhere …

You can put it in the wiki.

dl8dtl, 2023-04-05:
Well, I need one first. :-)
The closest I have so far is the list written in the avr-libc documentation.