avrdudes/avrdude #1336: CH341A futher improvement -- to support lower clock AVR chips

mcuee, 2023-04-07:
#1215 has been merged which addressed major part of Issue #714.

* #714 
* #1215

However, #1215 only supports fast SPI mode, which means that lower clock AVR chips (including many AVR chips out of factory which will run on internal RC oscillator at around 1MHz to 1.6Mhz). It is mentioned in #714 that bitbang SPI can be used to support AVR chips with lower clock.

mcuee, 2023-04-07:
From the following comments:
* https://github.com/avrdudes/avrdude/issues/714#issuecomment-1168047250

Reference bitbang SPI code using libusb.
https://github.com/maxxproff/avrdude-ch341/blob/main/ch341a_bitbang.c

mcuee, 2023-04-07:
From the comment here.
* https://github.com/avrdudes/avrdude/pull/1215#issuecomment-1493069143

Hans mentioned in the above comment.
> It is also not possible to change the ISP clock speed. We know that it can't go lower than 1.2 MHz, but I think it should be possible to increase this. AVRs running at 20 MHz support ISP clocks up to 5 MHz.

mcuee, 2023-04-07:
Yet another potential improvement is to add TPI support.

mcuee, 2023-04-07:
@mariusgreuel 

When you got some time, please check if you can add MSVC build to this one. Thanks.

stefanrueger, 2023-04-08:
Apart from the libusb/MSVC build issue this is pretty much a new and different driver, sth like `ch341_bb`. Needs to be contributed by people with an insight into the chip. There are simular bitbang driver, eg, for ftdi. So maybe the community can contribute? @Kaell?

MCUdude, 2023-04-08:
@maxxproff hosts an Avrdude with support for ch341a bit-banging:
https://github.com/maxxproff/avrdude-ch341

stefanrueger, 2023-04-08:
> @maxxproff hosts an Avrdude with support for ch341a bit-banging: https://github.com/maxxproff/avrdude-ch341

Great find. Making them a great candidate to submit a PR :)

mcuee, 2023-04-09:
I have split the enhancement into two issues. This issue will mainly deal with the SPI bitbang enhancement. The MSVC build issue will be tracked in #1339.

* #1339

mcuee, 2023-04-09:
> @maxxproff hosts an Avrdude with support for ch341a bit-banging: https://github.com/maxxproff/avrdude-ch341

He mentioned that the bit-bang driver to be this one, which was mentioned earlier.
https://github.com/maxxproff/avrdude-ch341/blob/main/ch341a_bitbang.c

stefanrueger, 2023-06-01:
I still think this has to be driven by the community, who use the CH341A for programming, not us as maintainers. Hence, I would argue to close this and engage with a PR if and when it comes along.

mcuee, 2023-06-02:
@stefanrueger 

The issues in github do not need to be driven by the maintainers -- the community can help to provide the PR.

Some potential contributors will first look at github to see if there are anything they can contribute. If we close issues like this, they may not know this issue at all.

mcuee, 2023-06-02:
@stefanrueger 

I will add a `help wanted` label to show that we expect the community to help for this issue.