avrdudes/avrdude #1337: AVRDUDE flashing eeprom with error on ARM64 on ATMEGA128 uC // Corrupt data

Topper69, 2023-04-07:
Hi,

When i try to burn a complete eeprom dump with AVRDUDE 7.1 (ARM64) on my rasp4 64b os, on the verification it show some errors (corrupt data) which is confirmed when i dump back the eeprom. 
I tried the debian version 

When i do the same on WIN10, it's complete without any error. 100% working. (same board, same programmer, same dump).

uC : ATMEGA128
programmer : arduino uno with ISP sketch loaded.

AVRDUDE on ARM64 (with error) log file:
<code>
pi@Rasp4:~/Documents/avrdude $ ./avrdude -p m128 -P /dev/ttyACM0 -c avrisp -b 19200 -U eeprom:w:/home/pi/Documents/avrdude/firmwarem128ok.eep:i -v

avrdude: Version 7.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /home/pi/Documents/avrdude/avrdude.conf
         User configuration file is /home/pi/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM0
         Using Programmer              : avrisp
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega128
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                  4    20    64    0 no       4096    8      0  9000  9000 0xff 0xff
           flash                  33    10   128    0 yes    131072  256    512  4500  4500 0xff 0x00
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel AVR ISP
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9702 (probably m128)
avrdude: reading input file /home/pi/Documents/avrdude/firmwarem128ok.eep for eeprom
         with 4096 bytes in 1 section within [0, 0xfff]
         using 512 pages and 0 pad bytes
avrdude: writing 4096 bytes eeprom ...

Writing | ################################################## | 100% 192.94 s 

avrdude: 4096 bytes of eeprom written
avrdude: verifying eeprom memory against /home/pi/Documents/avrdude/firmwarem128ok.eep

Reading | ################################################## | 100% 6.49 s 

avrdude avr_verify() warning: verification mismatch
        device 0x00 != input 0xff at addr 0x0339 (error)
        device 0x72 != input 0xff at addr 0x033a (error)
        device 0x73 != input 0xff at addr 0x033b (error)
        device 0x74 != input 0xff at addr 0x033c (error)
        device 0x75 != input 0xff at addr 0x033d (error)
        device 0x76 != input 0xff at addr 0x033e (error)
        device 0x77 != input 0xff at addr 0x033f (error)
        device 0x80 != input 0xff at addr 0x0340 (error)
        device 0x81 != input 0xff at addr 0x0341 (error)
        device 0x82 != input 0xff at addr 0x0342 (error)
        suppressing further verification errors
avrdude do_op() error: verification mismatch

avrdude done.  Thank you.
</code>

On WIN10: (working)
<code>
./avrdude -p m128 -P COM21 -c arduino -b 19200 -U eeprom:w:firmwarem128ok.eep:i -v

avrdude: Version 7.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is avrdude.conf

         Using Port                    : COM21
         Using Programmer              : arduino
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega128
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                  4    12    64    0 no       4096    8      0  9000  9000 0xff 0xff
           flash                  33     6   128    0 yes    131072  256    512  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino for bootloader using STK500 v1 protocol
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9702 (probably m128)
avrdude: reading input file firmwarem128ok.eep for eeprom
         with 4096 bytes in 1 section within [0, 0xfff]
         using 512 pages and 0 pad bytes
avrdude: writing 4096 bytes eeprom ...

Writing | ################################################## | 100% 193.02 s

avrdude: 4096 bytes of eeprom written
avrdude: verifying eeprom memory against firmwarem128ok.eep

Reading | ################################################## | 100% 7.77 s

avrdude: 4096 bytes of eeprom verified

avrdude done.  Thank you.
</code>

I have only issue with EEPROM, with FLASH it's fine (no corrupt data).

Best regards

mcuee, 2023-04-07:
@Topper69 

What if you try the latest git version? You can either build from source, or use [github action version](https://github.com/avrdudes/avrdude/suites/12081530638/artifacts/635956064).

mcuee, 2023-04-07:
BTW, if you use `Arduino as ISP` as your main programmer, I will highly recommend buying an AVR ISP Shield from online stores (pretty cheap from places like AliExpress). 

Before I got the shield I use ribbon cables and  I often has connection issues. After I got the shield, it becomes much more stable. 

One example of the AVR ISP Shield. There are other versions.
[image]

BTW, you can also boost the ISP speed to 57600 bps by optimizing the "Arduino as ISP" sketch. You can also use the ScratchMonkey to use `-c stk500v2`.

Ref: https://github.com/avrdudes/avrdude/discussions/1088

Topper69, 2023-04-07:
> @Topper69
> 
> What if you try the latest git version? You can either build from source, or use [github action version](https://github.com/avrdudes/avrdude/suites/12081530638/artifacts/635956064).

I use the nightly build from : https://github.com/avrdudes/avrdude/wiki/Getting-Nightly-Builds-for-AVRDUDE 
(the aarch64 version)

About the bad connection, it's unlikly because i works fine in windows (i keep the same board / link / wires). In my case i have soldered some wire to the uC.

mcuee, 2023-04-07:
I see. No problem. I can check on myside as well to see if I can reproduce the issue or not.

stefanrueger, 2023-04-08:
Do all .eep hex files trigger this error?  Does the error only appear on sections with 0xff? If it's only specific files, please share a problematic one. Please share a log with higher verbosity (`-vv`, `-vvv`, ...). This so that one of us (@mcuee?) can reproduce the problem.

Topper69, 2023-04-12:
Hi,

Yes all files (not empty) triggers the issue.
Find attached logs from my raspberry pi 4 (linux) & from win with the same eep file.
The dump is a fully random file that fill all eeprom memory of the Atmega128.

[randval.zip](https://github.com/avrdudes/avrdude/files/11202441/randval.zip)
[log_linux.txt](https://github.com/avrdudes/avrdude/files/11202454/log_linux.txt)
[log_windows.txt](https://github.com/avrdudes/avrdude/files/11207712/log_windows.txt)

mcuee, 2023-04-12:
I should be able to test this over the weekend on my Raspberry Pi 400.

mcuee, 2023-04-24:
Sorry I can not find my ATmega128A breakout board so this will be delayed.

Topper69, 2023-04-24:
I have switch to my pickit4 now without issues (drop to use my ardunio uno as programmer).

Tim666, 2023-05-27:
I have identical problem. I use Arduino Nano as Arduino ISP programmer for ATmega328P. It correctly writes flash but has problem with eeprom. It loos like it can write eeprom only from address 0x013C. 
[image]
Read/write log comparation
[image]

Tim666, 2023-06-02:
I use Intel Windows 10 PC. Command for writing eeprom:
`avrdude -c stk500v1 -p m328p -P com7 -b 19200 -v -U eeprom:w:mega328.eep:a`
I do not understand how is possible "not stable" successfully write flash and incorrectly write half eeprom. 
And this situation reproduces every time identically. 
It look like the writing protection the part of memory in ArduinoISP or chip.

stefanrueger, 2023-06-02:
@Tim666 Try `-c arduino` instead of `-c stk500v1`

Tim666, 2023-06-03:
@stefanrueger Thanks a lot!!!!
I tried  `-c arduino` before but without success, then possible after different experiments I fixed hardware issue.
Command `avrdude -c arduino -p m328p -P com7 -b 57600 -v -U eeprom:w:mega328.eep:a` correctly upload flash.
I found out my problem was the `stk500v1` incorrectly read data from address below  0x013C. It writes correctly but verification after writing does not match. So the command `avrdude -c stk500v1 -p m328p -P com7 -b 19200 -v -U eeprom:w:mega328.eep:a` do the same thing but impossible to check if eeprom uploaded correctly.

stefanrueger, 2023-06-03:
@Tim666 If `-c arduino` works then `-c urclock` should too, but `-c stk500v1` no longer works for *Arduino as ISP* b/c it has been changed to work for STK500 v1 programmers writing EEPROM. If you use an *ArduinoISP* programmer (note the difference to *Arduino as ISP*) then `-c arduinoisp` is your friend.

Tim666, 2023-06-03:
@stefanrueger  As I understand... now... the command `-c arduino` writes firmware to connected Arduino Nano. But I want to use _Arduino Nano_ as ArduinoISP programmer for uploading firmware to another ATmega328P chip. And command `-c arduino` rewrites my _Arduino Nano_ only, but not target ATmega328P :-(
I remembered why I used `-c stk500v1` because Arduino IDE uses it for "Arduino as ISP" programmer:
`avrdude -CC:[PATH]avrdude.conf -v -patmega328p -cstk500v1 -PCOM7 -b19200 -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xDA:m -Ulfuse:w:0xFF:m`

mcuee, 2023-06-04:
@Tim666 

Just wondering which "Arduino as ISP" sketch do you use? 

Last time there is a report of using the Adfruit version which does not work with EEPROM.
* https://github.com/avrdudes/avrdude/issues/990

mcuee, 2023-06-04:
No issues here under Windows 11.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c stk500v1 -P COM11 -b 19200 -p m328p -U eeprom:w:entest.hex:i
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: reading input file entest.hex for eeprom
         with 512 bytes in 1 section within [0, 0x1ff]
         using 128 pages and 0 pad bytes
avrdude: writing 512 bytes eeprom ...
Writing | ################################################## | 100% 25.80 s
avrdude: 512 bytes of eeprom written
avrdude: verifying eeprom memory against entest.hex
Reading | ################################################## | 100% 2.38 s
avrdude: 512 bytes of eeprom verified

avrdude done.  Thank you.
```

As mentioned before, the "Arduino as ISP" sketech is a bit fragile and you need good connections. 

Here is my setup.

mcuee, 2023-06-04:
All in all "Arduino as ISP" should not be used as a regular programmer, rather you can use it occassionally. I will recommend getting a more proper programmer.

For hobbist, usbasp is a good one to use. usbtinyisp can be used as well but it is not as good as usbasp.

If you are interested in latest AVR chip as well, then Microchip SNAP or PICKit 4 is a good investment.

mcuee, 2023-06-04:
I have found my ATmgea128A breakout board.

1) Writing 512Bytes hex file to the EEPROM seems to be stable
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c stk500v1 -P COM11 -b 19200 -p m128a -U eeprom:w:entest.hex:i
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9702 (probably m128a)
avrdude: reading input file entest.hex for eeprom
         with 512 bytes in 1 section within [0, 0x1ff]
         using 64 pages and 0 pad bytes
avrdude: writing 512 bytes eeprom ...
Writing | ################################################## | 100% 24.73 s
avrdude: 512 bytes of eeprom written
avrdude: verifying eeprom memory against entest.hex
Reading | ################################################## | 100% 1.39 s
avrdude: 512 bytes of eeprom verified

avrdude done.  Thank you.
```

2) Writing the big hex file (4096Bytes) as mentioned in the original report is not successful.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c stk500v1 -P COM11 -b 19200 -p m128a -U eeprom:w:randval.eep:i
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9702 (probably m128a)
avrdude warning: no end of file record found for Intel Hex file randval.eep
avrdude: reading input file randval.eep for eeprom
         with 4096 bytes in 1 section within [0, 0xfff]
         using 512 pages and 0 pad bytes
avrdude: writing 4096 bytes eeprom ...
Writing | ################################################## | 100% 194.83 s
avrdude: 4096 bytes of eeprom written
avrdude: verifying eeprom memory against randval.eep
Reading | ################################################## | 100% 9.66 s
avrdude warning: verification mismatch
        device 0x1b != input 0x41 at addr 0x0000 (error)
avrdude error: verification mismatch

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c stk500v1 -P COM11 -b 19200 -p m128a -U eeprom:w:entest_4096B.eep:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9702 (probably m128a)
avrdude: reading input file entest_4096B.eep for eeprom
         with 4096 bytes in 1 section within [0, 0xfff]
         using 512 pages and 0 pad bytes
avrdude: writing 4096 bytes eeprom ...
Writing | ################################################## | 100% 194.76 s
avrdude: 4096 bytes of eeprom written
avrdude: verifying eeprom memory against entest_4096B.eep
Reading | ################################################## | 100% 9.65 s
avrdude warning: verification mismatch
        device 0x73 != input 0x54 at addr 0x0000 (error)
avrdude error: verification mismatch

avrdude done.  Thank you.
```

mcuee, 2023-06-04:
1024B and 2048B are okay.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c stk500v1 -P COM11 -b 19200 -p m128a -U eeprom:w:entest_1024B.eep:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9702 (probably m128a)
avrdude: reading input file entest_1024B.eep for eeprom
         with 1024 bytes in 1 section within [0, 0x3ff]
         using 128 pages and 0 pad bytes
avrdude: writing 1024 bytes eeprom ...
Writing | ################################################## | 100% 49.24 s
avrdude: 1024 bytes of eeprom written
avrdude: verifying eeprom memory against entest_1024B.eep
Reading | ################################################## | 100% 2.79 s
avrdude: 1024 bytes of eeprom verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c stk500v1 -P COM11 -b 19200 -p m128a -U eeprom:w:entest_2048B.eep:i

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9702 (probably m128a)
avrdude: reading input file entest_2048B.eep for eeprom
         with 2048 bytes in 1 section within [0, 0x7ff]
         using 256 pages and 0 pad bytes
avrdude: writing 2048 bytes eeprom ...
Writing | ################################################## | 100% 97.73 s
avrdude: 2048 bytes of eeprom written
avrdude: verifying eeprom memory against entest_2048B.eep
Reading | ################################################## | 100% 5.11 s
avrdude: 2048 bytes of eeprom verified

avrdude done.  Thank you.
```

mcuee, 2023-06-04:
All in all, I do not consider this to be an avrdude problem. Rather this `Arduino as ISP` sketech is fragile.

mcuee, 2023-06-04:
I will close this issue now.

@Tim666 
You can continue to post your findings here but I recommend to switch to better programmers like usbasp.

mcuee, 2023-06-04:
BTW, I can reproduce the issue under Linux as well.
```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c stk500v1 -P /dev/ttyUSB0 -b 19200 -p m128a -U eeprom:w:randval.eep:i
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9702 (probably m128a)
avrdude warning: no end of file record found for Intel Hex file randval.eep
avrdude: reading input file randval.eep for eeprom
         with 4096 bytes in 1 section within [0, 0xfff]
         using 512 pages and 0 pad bytes
avrdude: writing 4096 bytes eeprom ...
Writing | ################################################## | 100% 194.05 s 
avrdude: 4096 bytes of eeprom written
avrdude: verifying eeprom memory against randval.eep
Reading | ################################################## | 100% 8.90 s 
avrdude warning: verification mismatch
        device 0x1b != input 0x41 at addr 0x0000 (error)
avrdude error: verification mismatch

avrdude done.  Thank you.

```

mcuee, 2023-06-04:
Please take note Arduino does not really care about EEPROM uploading at all from their AVR bootloaders.
1) The AVR code optiboot is of very old 4.x version and does not support EEPROM at all.
* https://github.com/avrdudes/avrdude/issues/1377

2) The STK500v2 bootloader for ATmega2560 has a long standing EEPROM bug which they also do not want to fix.
* https://github.com/arduino/ArduinoCore-avr/pull/24

Therefore I will not feel surprised that this "Arduino as ISP" does not work with EEPROM very well.

stefanrueger, 2023-06-04:
> the command -c arduino writes firmware to connected Arduino Nano

@Tim666 Only because `-c arduino` resets the connected target *and* the Nano has a bootloader that will intercept the upload mechanism. You can try removing the Bootloader on the Nano (or, simpler, change the BOOTRST configuration of the Nano to jump to addr 0 on reset, not the bootloader); you can try preventing the Nano be reset by the host PC that runs avrdude: is there a jumper on the Nano board that prevents reset from the host?

Basically `-c arduino` and `-c stk500v1` are the same except `-c arduino` tries to reset the connected board and EEPROM treatment of `-c arduino` is compatible with Arduino as ISP, whilst EEPROM treatment of `-c stk500v1` is compatible with the STK500 programmer with v1 firmware.

Maybe we should give `-c arduino` and the (better) alternative `-c urclock` an option `-xnoreset`...

stefanrueger, 2023-06-05:
@Tim666  and @mcuee: Give PR #1383 a try to see whether `-c avrduinoAsISP` solves this.

mcuee, 2023-06-05:
@Topper69 

You may want to try PR #1383 as well. I tested ATmega128A using your hex file and it works.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1383 -C .\avrdude_pr1383.conf -c arduinoAsISP -P COM11 -b 19200 -p m128a -U eeprom:w:randval.eep:i
avrdude_pr1383: AVR device initialized and ready to accept instructions
avrdude_pr1383: device signature = 0x1e9702 (probably m128a)

avrdude_pr1383: processing -U eeprom:w:randval.eep:i
avrdude_pr1383: reading input file randval.eep for eeprom
                with 4096 bytes in 1 section within [0, 0xfff]
                using 512 pages and 0 pad bytes
avrdude_pr1383: writing 4096 bytes eeprom ...
Writing | ################################################## | 100% 195.00 s
avrdude_pr1383: 4096 bytes of eeprom written
avrdude_pr1383: verifying eeprom memory against randval.eep
Reading | ################################################## | 100% 9.82 s
avrdude_pr1383: 4096 bytes of eeprom verified

avrdude_pr1383 done.  Thank you.
```

mcuee, 2023-06-11:
Close this one as #1383 has been merged.

closed 2023-06-11 by mcuee

Ho-Ro, 2023-09-15:
> Basically -c arduino and -c stk500v1 are the same except -c arduino tries to reset the connected board and EEPROM treatment of -c arduino is compatible with Arduino as ISP, whilst EEPROM treatment of -c stk500v1 is compatible with the STK500 programmer with v1 firmware.

Looking at the logfiles for EEPROM reading shows the difference:

This is part of the log for `stk500v1` protocol, the sent address cmd `U` is increased in steps of four (`U [55] . [00] . [00]   [20]`, `U [55] . [04] . [00]   [20]`, ...) and four bytes are read by cmd `t`:

```
avrdude -p m328p -vvvv -c stk500v1 -U eeprom:r:/dev/null:r -l read_eeprom_stk500v1.log
```
```
avrdude: Version 7.1
...
avrdude: reading eeprom memory ...

Reading | avrdude: send: U [55] . [00] . [00]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [10] 
avrdude: send: t [74] . [00] . [04] E [45]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [00] . [00] . [00] . [00] 
avrdude: recv: . [10] 
avrdude: send: U [55] . [04] . [00]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [10] 
avrdude: send: t [74] . [00] . [04] E [45]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [00] . [00] . [00] . [00] 
avrdude: recv: . [10] 
avrdude: send: U [55] . [08] . [00]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [10] 
avrdude: send: t [74] . [00] . [04] E [45]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [00] . [00] . [00] . [00] 
avrdude: recv: . [10] 
avrdude: send: U [55] . [0c] . [00]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [10] 
avrdude: send: t [74] . [00] . [04] E [45]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [00] . [00] . [00] . [00] 
avrdude: recv: . [10]
...
```

And this is part of the log for `arduino` protocol, the sent address (cmd `U`) is increased in steps of **two** (`U [55] . [00] . [00]   [20]`, `U [55] . [02] . [00]   [20]`, ...) but also four bytes are read by cmd `t`:

```
avrdude -p m328p -vvvv -c arduino -U eeprom:r:/dev/null:r -l read_eeprom_arduino.log
```
```
avrdude: Version 7.1
...
avrdude: reading eeprom memory ...

Reading | avrdude: send: U [55] . [00] . [00]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [10] 
avrdude: send: t [74] . [00] . [04] E [45]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [00] . [00] . [00] . [00] 
avrdude: recv: . [10] 
avrdude: send: U [55] . [02] . [00]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [10] 
avrdude: send: t [74] . [00] . [04] E [45]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [00] . [00] . [00] . [00] 
avrdude: recv: . [10] 
avrdude: send: U [55] . [04] . [00]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [10] 
avrdude: send: t [74] . [00] . [04] E [45]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [00] . [00] . [00] . [00] 
avrdude: recv: . [10] 
avrdude: send: U [55] . [06] . [00]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [10] 
avrdude: send: t [74] . [00] . [04] E [45]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [00] . [00] . [00] . [00] 
avrdude: recv: . [10] 
avrdude: send: U [55] . [08] . [00]   [20] 
avrdude: recv: . [14] 
avrdude: recv: . [10] 
...
```

mcuee, 2023-09-17:
Please use avrdude 7.2 release or latest git main. Thanks.

Ho-Ro, 2023-09-17:
> Please use avrdude 7.2 release or latest git main. Thanks.

As expected nothing has changed from my reported 7.1 logging.
I get the same **correct** behaviour with latest github as well as with 7.2: 
- with `-c arduino` avrdude increments the EEPROM address be **two** and reads four bytes.
- with `-c stk500v1` it increments the EEPROM address be **four** and reads four bytes.

**This is the correct behaviour** because the `stk500v1` protocol expects **byte addresses** and the modified `arduino` protocol expects **word addresses**.

Remark:
My *Arduino Nano as ISP* hardware uses my [improved firmware](https://github.com/Ho-Ro/nanoSTK) that automatically detects the used protocol (`arduino` protocol uses one `Cmnd_STK_READ_SIGN` call to read the signature while `stk500v1` protocol uses three `Cmnd_STK_UNIVERSAL` calls for this purpose).