avrdudes/avrdude #1340: Add programmer specific settings as extended parameters

MCUdude, 2023-04-09:
If you have a device connected to an STK500 board, an STK600, a Power Debugger that's supplying power, or a Curiosity nano, one might end up in a situation where the target voltage is set too low, or the clock generator (STK500/600) is set too low.
And for the Power Debugger, one would have to manually set the target voltage before it will supply power to the target.

The only want to "fix" this is to enter terminal mode using -F, and adjust them that way.

However, it would be neat if this also could be done using extended parameters.

How about adding the following extended parameter to the STK500 and STK600:
`fosc=<arg>`

And the following extended parameter for the STK500, STK600, PowerDebugger and Curiosity Nano/pkonb_updi
`vtarg=<arg>`

This should ensure that the correct target voltage and clock speed are set before target initialization.

mcuee, 2023-04-09:
I think this is a good idea and good enhancement to avrdude.

MCUdude, 2023-04-18:
The Power Debugger is able to supply power, but this needs to be manually enabled every time the Power Debugger is connected.
With the PR I'm working on, it's only a matter of adding `-xvtarg=[voltage]`, and the Power Debugger will apply power to the target before trying to initialize.

closed 2023-04-23 by stefanrueger