avrdudes/avrdude #1344: ERROR trying to write EEPROM using STK500v1 ISP programmer, in v7.1

kr4fty, 2023-04-18:
Hello, I am updating the firmware of a TransistorTester (TC-1) and when I want to upload it to the board, it gives me an error when verifying the writing of the eeprom. I have tried different transfer speeds and the result is the same

***Details**:*
- OS: Manjaro Linux
- avrdude v7.1
- [ComponentTester v1.49m](https://github.com/kubi48/TransistorTester-source/blob/master/Markus/ComponentTester-1.49m.tgz)
- MCU: ATmega324PA / 16Mhz

Then I tried compiling a previous version of avrdude, [v6.3](http://download.savannah.gnu.org/releases/avrdude/), and it wrote and verified correctly

Command output:

```
AVR Memory Usage
----------------
Device: atmega324p

Program:   27658 bytes (84.4% Full)
(.text + .data + .bootloader)

Data:        222 bytes (10.8% Full)
(.data + .bss + .noinit)

EEPROM:      639 bytes (62.4% Full)
(.eeprom)

avrdude -c stk500v1 -P /dev/ttyUSB0 -p m324pa -b 115200 \
  -U flash:w:./ComponentTester.hex:a -U eeprom:w:./ComponentTester.eep:a

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9511 (probably m324pa)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file ./ComponentTester.hex for flash
         with 27658 bytes in 1 section within [0, 0x6c09]
         using 217 pages and 118 pad bytes
avrdude: writing 27658 bytes flash ...

Writing | ################################################## | 100% 17.50 s 

avrdude: 27658 bytes of flash written
avrdude: verifying flash memory against ./ComponentTester.hex

Reading | ################################################## | 100% 8.58 s 

avrdude: 27658 bytes of flash verified
avrdude: reading input file ./ComponentTester.eep for eeprom
         with 639 bytes in 1 section within [0, 0x27e]
         using 160 pages and 1 pad byte
avrdude: writing 639 bytes eeprom ...

Writing | ################################################## | 100% 29.89 s 

avrdude: 639 bytes of eeprom written
avrdude: verifying eeprom memory against ./ComponentTester.eep

Reading | ################################################## | 100% 1.03 s 

avrdude warning: verification mismatch
        device 0x61 != input 0x00 at addr 0x0000 (error)
avrdude error: verification mismatch

avrdude done.  Thank you.

make: *** [Makefile:231: upload] Error 1
```

stefanrueger, 2023-04-18:
Thanks for reporting. What's the `-vv` or `-vvv` output? Do you *really* use an STK500 (physical) programmer with *outdated* v1 firmware (superseded in around 2005 by v2?) or is this bootloading in which case you could try `-c arduino` or `-c urclock` as they know about bootloading and the EEPROM quirks of the bootloading protocol.

kr4fty, 2023-04-18:
Thanks for answering. I use an Arduino ISP, STK500v1 protocol, as ISP programmer

`-c arduino` seems to work

```
avrdude -c arduino -P /dev/ttyUSB0 -p m324pa -b 115200 \                                                                                                          ✔  1m 8s  
  -U flash:w:./ComponentTester.hex:a -U eeprom:w:./ComponentTester.eep:a

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9511 (probably m324pa)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file ./ComponentTester.hex for flash
         with 32278 bytes in 1 section within [0, 0x7e15]
         using 253 pages and 106 pad bytes
avrdude: writing 32278 bytes flash ...

Writing | ################################################## | 100% 20.40 s 

avrdude: 32278 bytes of flash written
avrdude: verifying flash memory against ./ComponentTester.hex

Reading | ################################################## | 100% 10.01 s 

avrdude: 32278 bytes of flash verified
avrdude: reading input file ./ComponentTester.eep for eeprom
         with 746 bytes in 1 section within [0, 0x2e9]
         using 187 pages and 2 pad bytes
avrdude: writing 746 bytes eeprom ...

Writing | ################################################## | 100% 34.95 s 

avrdude: 746 bytes of eeprom written
avrdude: verifying eeprom memory against ./ComponentTester.eep

Reading | ################################################## | 100% 1.21 s 

avrdude: 746 bytes of eeprom verified

avrdude done.  Thank you.
```
Output with` -vv` option

```
> avrdude -c stk500v1 -vv -P /dev/ttyUSB0 -p m324pa -b 115200 \                                                                                                     ✔  1m 8s  
  -U flash:w:./ComponentTester.hex:a -U eeprom:w:./ComponentTester.eep:a

avrdude: Version 7.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /etc/avrdude.conf
         User configuration file is /home/usuario/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 115200
avrdude: input file ./ComponentTester.hex auto detected as Intel Hex
avrdude: input file ./ComponentTester.eep auto detected as Intel Hex
         AVR Part                      : ATmega324PA
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
[... 59 lines cut ...]
Writing | ################################################## | 100% 34.94 s 

avrdude: 746 bytes of eeprom written
avrdude: verifying eeprom memory against ./ComponentTester.eep
avrdude: reading on-chip eeprom data ...

Reading | ################################################## | 100% 1.21 s 

avrdude: verifying ...
avrdude avr_verify() warning: verification mismatch
        device 0x72 != input 0x00 at addr 0x0000 (error)
        device 0x00 != input 0x01 at addr 0x0001 (error)
        device 0x53 != input 0x02 at addr 0x0002 (error)
        device 0x75 != input 0x01 at addr 0x0003 (error)
        device 0x6d != input 0x02 at addr 0x0004 (error)
        device 0x61 != input 0x04 at addr 0x0005 (error)
        device 0x20 != input 0x02 at addr 0x0006 (error)
        device 0x64 != input 0x08 at addr 0x0007 (error)
        device 0x65 != input 0x20 at addr 0x0008 (error)
        device 0x20 != input 0x01 at addr 0x0009 (error)
        suppressing further verification errors
avrdude do_op() error: verification mismatch

avrdude done.  Thank you.
```

kr4fty, 2023-04-18:
Thanks for explaining why it doesn't work with that STK500v1 protocol

closed 2023-04-18 by kr4fty