avrdudes/avrdude #1349: `-n` dryrun command improvement

mcuee, 2023-04-23:
Reference: 
* https://github.com/avrdudes/avrdude/pull/1347#issuecomment-1517786948

PR #1347 introduces `-c dryrun` programmer, which is very useful for testing purpose and for CI. It does not test the other programmer though.

avrdude has an existing `-n` command, but it is a bit limited now.

@stefanrueger mentioned the following comparison between `-n` and `-c dryrun`.
>  It's restricted to -U updates and doesn't extend to the terminal. I think -n was meant for debugging and therefore shows what would be written to the device when used in anger. In contrast, -c dryrun is a tool to check the logic of AVRDUDE, something particularly useful for testing terminal commands. I predict that in future -c dryrun is getting more traction.

I am wondering if we can improve `-n` to remove the restrictions mentioned above, for example, to extend it to the terminal mode. The good thing is that it will test the codes inside the programmers.

If it is too difficult to enhance `-n` in a generic way for all programmers, maybe we can try only the more important and more popular programmers like `usbasp` and `stk500v2`.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1347 -C .\avrdude_pr1347.conf -c xplainedmini
 -p m328pb -n -U .\hex3\zeros2KB.hex
avrdude_pr1347: AVR device initialized and ready to accept instructions
avrdude_pr1347: device signature = 0x1e9516 (probably m328pb)
avrdude_pr1347: Note: flash memory has been specified, an erase cycle will be performed.
                To disable this feature, specify the -D option.
avrdude_pr1347 warning: conflicting -e and -n options specified, NOT erasing chip
avrdude_pr1347: reading input file .\hex3\zeros2KB.hex for flash
                with 2048 bytes in 1 section within [0, 0x7ff]
                using 16 pages and 0 pad bytes
avrdude_pr1347: writing 2048 bytes flash ...
:200000000000000000000000000000000000000000000000000000000000000000000000E0
:200020000000000000000000000000000000000000000000000000000000000000000000C0
:200040000000000000000000000000000000000000000000000000000000000000000000A0
:20006000000000000000000000000000000000000000000000000000000000000000000080
:20008000000000000000000000000000000000000000000000000000000000000000000060
:2000A000000000000000000000000000000000000000000000000000000000000000000040
:2000C000000000000000000000000000000000000000000000000000000000000000000020
:2000E000000000000000000000000000000000000000000000000000000000000000000000
:200100000000000000000000000000000000000000000000000000000000000000000000DF
:200120000000000000000000000000000000000000000000000000000000000000000000BF
:2001400000000000000000000000000000000000000000000000000000000000000000009F
:2001600000000000000000000000000000000000000000000000000000000000000000007F
:2001800000000000000000000000000000000000000000000000000000000000000000005F
:2001A00000000000000000000000000000000000000000000000000000000000000000003F
:2001C00000000000000000000000000000000000000000000000000000000000000000001F
:2001E0000000000000000000000000000000000000000000000000000000000000000000FF
:200200000000000000000000000000000000000000000000000000000000000000000000DE
:200220000000000000000000000000000000000000000000000000000000000000000000BE
:2002400000000000000000000000000000000000000000000000000000000000000000009E
:2002600000000000000000000000000000000000000000000000000000000000000000007E
:2002800000000000000000000000000000000000000000000000000000000000000000005E
:2002A00000000000000000000000000000000000000000000000000000000000000000003E
:2002C00000000000000000000000000000000000000000000000000000000000000000001E
:2002E0000000000000000000000000000000000000000000000000000000000000000000FE
:200300000000000000000000000000000000000000000000000000000000000000000000DD
:200320000000000000000000000000000000000000000000000000000000000000000000BD
:2003400000000000000000000000000000000000000000000000000000000000000000009D
:2003600000000000000000000000000000000000000000000000000000000000000000007D
:2003800000000000000000000000000000000000000000000000000000000000000000005D
:2003A00000000000000000000000000000000000000000000000000000000000000000003D
:2003C00000000000000000000000000000000000000000000000000000000000000000001D
:2003E0000000000000000000000000000000000000000000000000000000000000000000FD
:200400000000000000000000000000000000000000000000000000000000000000000000DC
:200420000000000000000000000000000000000000000000000000000000000000000000BC
:2004400000000000000000000000000000000000000000000000000000000000000000009C
:2004600000000000000000000000000000000000000000000000000000000000000000007C
:2004800000000000000000000000000000000000000000000000000000000000000000005C
:2004A00000000000000000000000000000000000000000000000000000000000000000003C
:2004C00000000000000000000000000000000000000000000000000000000000000000001C
:2004E0000000000000000000000000000000000000000000000000000000000000000000FC
:200500000000000000000000000000000000000000000000000000000000000000000000DB
:200520000000000000000000000000000000000000000000000000000000000000000000BB
:2005400000000000000000000000000000000000000000000000000000000000000000009B
:2005600000000000000000000000000000000000000000000000000000000000000000007B
:2005800000000000000000000000000000000000000000000000000000000000000000005B
:2005A00000000000000000000000000000000000000000000000000000000000000000003B
:2005C00000000000000000000000000000000000000000000000000000000000000000001B
:2005E0000000000000000000000000000000000000000000000000000000000000000000FB
:200600000000000000000000000000000000000000000000000000000000000000000000DA
:200620000000000000000000000000000000000000000000000000000000000000000000BA
:2006400000000000000000000000000000000000000000000000000000000000000000009A
:2006600000000000000000000000000000000000000000000000000000000000000000007A
:2006800000000000000000000000000000000000000000000000000000000000000000005A
:2006A00000000000000000000000000000000000000000000000000000000000000000003A
:2006C00000000000000000000000000000000000000000000000000000000000000000001A
:2006E0000000000000000000000000000000000000000000000000000000000000000000FA
:200700000000000000000000000000000000000000000000000000000000000000000000D9
:200720000000000000000000000000000000000000000000000000000000000000000000B9
:20074000000000000000000000000000000000000000000000000000000000000000000099
:20076000000000000000000000000000000000000000000000000000000000000000000079
:20078000000000000000000000000000000000000000000000000000000000000000000059
:2007A000000000000000000000000000000000000000000000000000000000000000000039
:2007C000000000000000000000000000000000000000000000000000000000000000000019
:2007E0000000000000000000000000000000000000000000000000000000000000000000F9
:00000001FF
avrdude_pr1347: 2048 bytes of flash written
avrdude_pr1347: verifying flash memory against .\hex3\zeros2KB.hex
Reading | ################################################## | 100% 0.00 s 
avrdude_pr1347: 2048 bytes of flash verified

avrdude_pr1347 done.  Thank you.
```

stefanrueger, 2023-04-23:
>  extend [-n] to the terminal mode

I'd rather improve documentation that -n only covers -U. For the terminal `-c dryrun` is an alternative. 

> The good thing is that it will test the codes inside the programmers

It is not very clear (to me) what should happen to achieve this. Does this need every `-c` programmer to emulate a (not connected) real, physical programmer?

mcuee, 2023-04-24:
If `-n` enhancement makes no sense, we can use `-xdryrun`.

And yes indeed `-xdryrun` will need to change the programmer codes. I am not so sure if there is a generic way like `-n` if we limit the support to `-U` command and not terminal mode.

Basically an enhancement of `-n` which does not need to connect to a physcial programmer.

If there is no easy and generic way, probably we can limit it to a few popular programmers like `usbasp` and `stk500v2`.

MCUdude, 2023-04-24:
> I'd rather improve documentation that -n only covers -U. For the terminal -c dryrun is an alternative.

We can perhaps print a warning or error when -n is used in combination with -t?

stefanrueger, 2023-04-24:
> -xdryrun will need to change the programmer codes. I am not so sure if there is a generic way

Could you write and show example code for *one* programmer, say `usbasp`, to see what `-xdryrun` would do so that
 - There is no need to connect a physical programmer
 - Important checks within the `usbasp` programmer are carried out

That would give me a better understanding of what the suggested enhancement is.

MCUdude, 2023-04-24:
#1351 is an OK solution IMO. I think our limited time could be spent on more valuable things than emulating programmers.

Personally, I'm really interested in #1322. And #1180 is also very interesting, even though I suspect it's a lot of work.

mcuee, 2023-04-24:
The initial idea is to make it behave similar to `-n` but without the need to connect a programmer. 

Somthing like the following.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedmini -p m328pb -U .\hex3\zeros2KB.hex -xdryrun
avrdude: The following output is simulated with the expected behavior of the chip and programmer. No I/O will be performed.
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
                To disable this feature, specify the -D option.
avrdude_pr1347: reading input file .\hex3\zeros2KB.hex for flash
                with 2048 bytes in 1 section within [0, 0x7ff]
                using 16 pages and 0 pad bytes
avrdude_pr1347: writing 2048 bytes flash ...
avrdude_pr1347: 2048 bytes of flash written
avrdude_pr1347: verifying flash memory against .\hex3\zeros2KB.hex
Reading | ################################################## | 100% 0.00 s 
avrdude_pr1347: 2048 bytes of flash verified

avrdude_pr1347 done.  Thank you.
```

mcuee, 2023-04-24:
> Could you write and show example code for _one_ programmer, say `usbasp`, to see what `-xdryrun` would do so that
>
> * There is no need to connect a physical programmer
> * Important checks within the [...]

In the ideal case, the `-vvvv` output of `-xdryrun` will be very similar to the real programmer. But I guess it is too much work.

mcuee, 2023-04-24:
> #1351 is an OK solution IMO. I think our limited time could be spent on more valuable things than emulating programmers.
> 
> Personally, I'm really interested in #1322. And #1180 is also very interesting, even though I suspect it's a lot of work.

Fair enough. I will close this one.

With `-c dryrun` implemented by @stefanrueger, we have a good tool for CI. So in the end this enhancement is not strictly required.

mcuee, 2023-04-26:
@stefanrueger 

#1351 is good to go.

Another minor improement for `-n` command is not to print the full hex file. I see that as pointless (no matter which verbose level). Hopefully that is not difficult to implement.

mcuee, 2023-04-26:
I have re-opened this issue to talk about some minor improvement to `-n` command.

1) It is pointless to print the full hex file no matter at which verbose level. Hopefully that is not difficult to implement.
2) `-n` command will not affect reading operation but writing operation. I think it should implicitly include `-D` and then probably no need to say it conflicts with `-e`.

mcuee, 2023-04-26:
My idea of the `-n` command output.

avrdude_pr1347: AVR device initialized and ready to accept instructions
avrdude_pr1347: device signature = 0x1e9516 (probably m328pb)
avrdude_pr1347: Note: `-n` command used, no flash memory erase or write will be performed. No verification will be done.

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p m328p -n -D -U .\urboot_uno_autobaud_ee_led+b5_fr_ce_ur_vbl.hex -v

avrdude: Version 7.1-20230406 (c1e725b)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

         Using Port                    : usb
         Using Programmer              : usbasp
avrdude: input file .\urboot_uno_autobaud_ee_led+b5_fr_ce_ur_vbl.hex auto detected as Intel Hex
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0x00 0x00
           flash                  65    10   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/
avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: Note: `-n` command used, no flash memory erase or write will be performed. No verification will be done.
avrdude: reading input file .\urboot_uno_autobaud_ee_led+b5_fr_ce_ur_vbl.hex for flash
         with 374 bytes in 2 sections within [0x7e80, 0x7fff]
         using 3 pages and 10 pad bytes
avrdude: writing 374 bytes flash ...
avrdude: 374 bytes of flash written
avrdude: verifying flash memory against .\urboot_uno_autobaud_ee_led+b5_fr_ce_ur_vbl.hex
Reading | ################################################## | 100% 0.00 s avrdude: 374 bytes of flash verified

avrdude done.  Thank you.
```

mcuee, 2023-04-26:
Same for `-vvvv` (but no need to specify `-D`, it should be automatic).

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -p m328p -n -D -U .\urboot_uno_autobaud_ee_led+b5_fr_ce_ur_vbl.hex -vvvv

avrdude: Version 7.1-20230406 (c1e725b)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

         Using Port                    : usb
         Using Programmer              : usbasp
avrdude: usbasp_open("usb")
avrdude: seen device from vendor >www.fischl.de<
avrdude: seen product >USBasp<
avrdude: defaulting memtype in -U w:.\urboot_uno_autobaud_ee_led+b5_fr_ce_ur_vbl.hex option to "flash"
avrdude: input file .\urboot_uno_autobaud_ee_led+b5_fr_ce_ur_vbl.hex auto detected as Intel Hex
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash                  65    10   128    0 yes     32768  128    256  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/
avrdude: usbasp_initialize()
[... 271 lines cut ...]
avrdude: avr_read_mem(): skipping page 238: no interesting data
avrdude: avr_read_mem(): skipping page 239: no interesting data
avrdude: avr_read_mem(): skipping page 240: no interesting data
avrdude: avr_read_mem(): skipping page 241: no interesting data
avrdude: avr_read_mem(): skipping page 242: no interesting data
avrdude: avr_read_mem(): skipping page 243: no interesting data
avrdude: avr_read_mem(): skipping page 244: no interesting data
avrdude: avr_read_mem(): skipping page 245: no interesting data
[... cut ...]
avrdude: avr_read_mem(): skipping page 251: no interesting data
avrdude: avr_read_mem(): skipping page 252: no interesting data
avrdude: avr_read_mem(): skipping page 253: no interesting data
avrdude: avr_read_mem(): skipping page 254: no interesting data
avrdude: avr_read_mem(): skipping page 255: no interesting data
avrdude: verifying ...
avrdude: 374 bytes of flash verified
avrdude: usbasp_close()
avrdude: usbasp_transmit("USBASP_FUNC_DISCONNECT", 0x00, 0x00, 0x00, 0x00)

avrdude done.  Thank you.
``

mcuee, 2023-04-26:
Proof-of-concept patch (disable verification as well).
```
C:\work\avr\avrdude_test\avrdude_main [main ≡ +3 ~2 -0 !]> git diff
diff --git a/src/main.c b/src/main.c
index d337669..d86863e 100644
--- a/src/main.c
+++ b/src/main.c
@@ -735,6 +735,9 @@ int main(int argc, char * argv [])

       case 'n':
         uflags |= UF_NOWRITE;
+               uflags &= ~UF_AUTO_ERASE; /* include '-D' */
+               uflags &= ~UF_VERIFY; /* include '-V' */
+               printf("Note: -n command used, no flash memory erase or write will be performed. No real verification will be done either.\n");
         break;

       case 'O': /* perform RC oscillator calibration */
diff --git a/src/update.c b/src/update.c
index ce4c207..f384d20 100644
--- a/src/update.c
+++ b/src/update.c
@@ -549,7 +549,7 @@ int do_op(const PROGRAMMER *pgm, const AVRPART *p, UPDATE *upd, enum updateflags
       report_progress(1, 1, NULL);
     } else {
       // Test mode: write to stdout in intel hex rather than to the chip
-      rc = fileio(FIO_WRITE, "-", FMT_IHEX, p, upd->memtype, size);
+      //rc = fileio(FIO_WRITE, "-", FMT_IHEX, p, upd->memtype, size);
     }

     if (rc < 0) {

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue1349 -C .\avrdude_issue1349.conf -c usbasp -p m328p -n -U .\urboot_uno_autobaud_ee_led+b5_fr_ce_ur_vbl.hex -vvvv
Note: -n command used, no flash memory erase or write will be performed. No real verification will be done either.

avrdude_issue1349: Version 7.1-20230424 (bda26d3)
                   Copyright the AVRDUDE authors;
                   see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                   System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude_issue1349.conf

                   Using Port                    : usb
                   Using Programmer              : usbasp
avrdude_issue1349: usbasp_open("usb")
avrdude_issue1349: seen device from vendor >www.fischl.de<
avrdude_issue1349: seen product >USBasp<
avrdude_issue1349: defaulting memtype in -U w:.\urboot_uno_autobaud_ee_led+b5_fr_ce_ur_vbl.hex option to "flash"
avrdude_issue1349: input file .\urboot_uno_autobaud_ee_led+b5_fr_ce_ur_vbl.hex auto detected as Intel Hex
                   AVR Part                      : ATmega328P
                   Chip Erase delay              : 9000 us
                   PAGEL                         : PD7
                   BS2                           : PC2
                   RESET disposition             : possible i/o
                   RETRY pulse                   : SCK
                   Serial program mode           : yes
                   Parallel program mode         : yes
                   Timeout                       : 200
                   StabDelay                     : 100
                   CmdexeDelay                   : 25
                   SyncLoops                     : 32
                   PollIndex                     : 3
                   PollValue                     : 0x53
                   Memory Detail                 :

                                                     Block Poll               Page                       Polled
                     Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                     ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                     eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0x00 0x00
                                                     Block Poll               Page                       Polled
                     Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                     ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
[... 40 lines cut ...]
Reading |                                                    | 0% 0.00 s avrdude_issue1349: usbasp_spi_cmd(0x30, 0x00, 0x00, 0x00) ...
avrdude_issue1349: usbasp_transmit("USBASP_FUNC_TRANSMIT", 0x30, 0x00, 0x00, 0x00)
                   <= [00] [30] [00] [1e]
avrdude_issue1349: usbasp_spi_cmd() => 0x00, 0x30, 0x00, 0x1e
avrdude_issue1349: usbasp_spi_cmd(0x30, 0x00, 0x01, 0x00) ...
avrdude_issue1349: usbasp_transmit("USBASP_FUNC_TRANSMIT", 0x30, 0x00, 0x01, 0x00)
                   <= [00] [30] [00] [95]
avrdude_issue1349: usbasp_spi_cmd() => 0x00, 0x30, 0x00, 0x95
Reading | #################                                  | 33% 0.03 s avrdude_issue1349: usbasp_spi_cmd(0x30, 0x00, 0x02, 0x00) ...
avrdude_issue1349: usbasp_transmit("USBASP_FUNC_TRANSMIT", 0x30, 0x00, 0x02, 0x00)
                   <= [00] [30] [00] [0f]
avrdude_issue1349: usbasp_spi_cmd() => 0x00, 0x30, 0x00, 0x0f
Reading | ################################################## | 100% 0.05 s
avrdude_issue1349: device signature = 0x1e950f (probably m328p)
avrdude_issue1349: reading input file .\urboot_uno_autobaud_ee_led+b5_fr_ce_ur_vbl.hex for flash
                   with 374 bytes in 2 sections within [0x7e80, 0x7fff]
                   using 3 pages and 10 pad bytes
avrdude_issue1349: writing 374 bytes flash ...
avrdude_issue1349: 374 bytes of flash written
avrdude_issue1349: usbasp_close()
avrdude_issue1349: usbasp_transmit("USBASP_FUNC_DISCONNECT", 0x00, 0x00, 0x00, 0x00)

avrdude_issue1349 done.  Thank you.

```

stefanrueger, 2023-05-01:
> to talk about some minor improvement to -n command
>  [...] pointless to print the full hex file no matter at which verbose level

It is not the hex file that is printed to stdout on `-n` but what avrdude wants to send to the programmer. This is useful for debugging, and it is documented behaviour of `-n`. As it is stdout one can redirect to `/dev/null` if  not wanted:
```
$ avrdude -qqnU build-urclock/blink.hex >/dev/null && echo OK
avrdude warning: conflicting -e and -n options specified, NOT erasing chip
OK
```

I agree, however, that the warning should only say conflicting `-e` and `-n` if actually `-e`  had explicitly been specified. 

As `-n` is a relatively old and established part of the code base, I'd rather not change its behaviour lest users who find it useful complain about this change in avrdude.

mcuee, 2023-05-02:
@stefanrueger 

Redirect to /dev/null is probably not the right thing as you still want to look at other outputs. And I still think it does not make too much sense to print the whole hex file which can be big (say 256KB for ATmega2560). But since this is the document way, I can live with it.

Now PR #1351 is good to go and you can close this issue after merging that PR.

mcuee, 2023-05-02:
@stefanrueger 

And I also agree that `-n` command will be less useful after the implementation of `-c dryrun`. Therefore I am okay not to change much on the `-n` behavior.

stefanrueger, 2023-05-02:
> Redirect to /dev/null is probably not the right thing as you still want to look at other outputs.

But you do see the *regular stderr* output when you redirect *stdout* to `/dev/null`:

```
$ avrdude -nU blink.hex >/dev/null
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude warning: -n specified, NOT erasing chip
avrdude: reading input file build-urclock/blink.hex for flash
         with 736 bytes in 1 section within [0, 0x2df]
         using 6 pages and 32 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 736 bytes flash ...
avrdude: 736 bytes of flash written
avrdude: verifying flash memory against build-urclock/blink.hex
Reading | ################################################## | 100% 0.00savrdude: 736 bytes of flash verified

avrdude done.  Thank you.
```

mcuee, 2023-05-03:
Good point.

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_pr1351 -c usbasp -p m328p -nU .\urboot_uno_autobaud_ee_led+b5_fr_ce_ur_vbl.hex  -vvvv >dev/nul

avrdude_pr1351: Version 7.1-20230501 (532387c)
                Copyright the AVRDUDE authors;
                see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

                System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

                Using Port                    : usb
                Using Programmer              : usbasp
avrdude_pr1351: usbasp_open("usb")
avrdude_pr1351: seen device from vendor >www.fischl.de<
avrdude_pr1351: seen product >USBasp<
avrdude_pr1351: defaulting memtype in -U w:.\urboot_uno_autobaud_ee_led+b5_fr_ce_ur_vbl.hex option to "flash"
avrdude_pr1351: input file .\urboot_uno_autobaud_ee_led+b5_fr_ce_ur_vbl.hex auto detected as Intel Hex
                AVR Part                      : ATmega328P
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PC2
                RESET disposition             : possible i/o
                RETRY pulse                   : SCK
                Serial program mode           : yes
                Parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                                  Block Poll               Page                       Polled
                  Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0x00 0x00
                                                  Block Poll               Page                       Polled
                  Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  flash                  65    10   128    0 yes     32768  128    256  4500  4500 0x00 0x00
                                                  Block Poll               Page                       Polled
                  Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                                  Block Poll               Page                       Polled
                  Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                                  Block Poll               Page                       Polled
                  Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                                  Block Poll               Page                       Polled
                  Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
                                                  Block Poll               Page                       Polled
                  Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
                                                  Block Poll               Page                       Polled
                  Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

                Programmer Type : usbasp
                Description     : USBasp, http://www.fischl.de/usbasp/
avrdude_pr1351: usbasp_initialize()
[... 273 lines cut ...]
avrdude_pr1351: avr_read_mem(): skipping page 238: no interesting data
avrdude_pr1351: avr_read_mem(): skipping page 239: no interesting data
[... cut ...]
avrdude_pr1351: avr_read_mem(): skipping page 252: no interesting data
avrdude_pr1351: avr_read_mem(): skipping page 253: no interesting data
avrdude_pr1351: avr_read_mem(): skipping page 254: no interesting data
avrdude_pr1351: avr_read_mem(): skipping page 255: no interesting data
avrdude_pr1351: verifying ...
avrdude_pr1351: 374 bytes of flash verified
avrdude_pr1351: usbasp_close()
avrdude_pr1351: usbasp_transmit("USBASP_FUNC_DISCONNECT", 0x00, 0x00, 0x00, 0x00)

avrdude_pr1351 done.  Thank you.
```

mcuee, 2023-05-03:
@stefanrueger 

Now PR #1351 is good to go and you can close this issue after merging PR #1351.