avrdudes/avrdude #1350: Support a mix of linuxgpio and PM_UPDI

gcormier, 2023-04-23:
Would it be possible to use linuxgpio to bitbang the serial for `PM_UPDI`, as is the case with `PM_ISP`? This might require additional code. Not sure which mapping would relate to the actual UPDI data pin.

ISP Example
```
programmer
  id          = "avrhat_5";
  desc        = "Use the Linux sysfs interface to bitbang GPIO lines";
  type        = "linuxgpio";
  prog_modes  = PM_ISP;
  reset       = 19;
  sck         = 20;
  mosi        = 21;
  miso        = 26;
;

mcuee, 2023-04-24:
I think this should be a possible enhancement but somebody needs to contribute to write the codes.

In the mean time, you may want to use serialupdi or jtag2updi.
1) serialupdi -- low cost and yet with good performance
https://github.com/SpenceKonde/AVR-Guidance/blob/master/UPDI/jtag2updi.md

2) jtag2updi -- if you have Uno/Nano
https://github.com/ElTangas/jtag2updi

MCUdude, 2023-04-24:
You should be able to build your own SerialUPDI programmer using the RX and TX lines on your Raspberry Pi Header.
I don't see the reason why anyone would want to use a bit-banged, GPIO-based SerialUPDI programmer when they can use the UARTS pins directly and achieve really high speeds.

gcormier, 2023-04-24:
> I don't see the reason why

I can solve that one.

https://github.com/gcormier/avr5hat

This allows me to program 5 ISP targets at once. With the requested functionality, the same could be done for UPDI targets.

MCUdude, 2023-04-24:
@dbuchwald was the one that wrote the SerialUPDI implementation, so maybe he can answer if this is a doable thing or not.

dbuchwald, 2023-04-24:
@MCUdude, this is generally doable, as everything else, but the effort in AVRDUDE is difficult to justify. 

What I would recommend to @gcormier is to look if there are any software based bit-bang UART devices available (see https://www.linuxquestions.org/questions/linux-embedded-and-single-board-computer-78/software-uart-driver-4175524941/ for instance) and use something like that for the mapping of virtual UART devices to GPIO pins. Then he could have AVRDUDE operate as normal by using these UART devices.

Does that make sense, @gcormier?

gcormier, 2023-04-24:
@dbuchwald Good call. There is indeed one, at https://github.com/adrianomarto/soft_uart

That said, I'm realizing now it might not be as simple as that. As mentioned on jtag2updi, hardware modifications are needed. Or, linuxgpio would need to switch between read/write at sufficient speed which I am rather skeptical of, and would agree that would not be fun to support with avrdude.

That said, I suspect we can close this issue. It is a bit of an edge case, but someone was asking if this board could be used for both ISP and UPDI. I think we found the answer :)

mcuee, 2023-04-24:
Okay, I will close this as unplanned.

closed 2023-04-24 by mcuee