avrdudes/avrdude #1354: Request: option to write all ELF sections at once

landfillbaby, 2023-04-27:
It would be great if there was an option to write every applicable section of an ELF file to the chip at once.
Maybe the syntax would be `-U all:w:file.elf:e` or `-W file.elf` or something.
Kind of the opposite of #1202.

mcuee, 2023-04-27:
@landfillbaby 

Please refer to the comment here and this will be a lower priority for the avrdude team. If you want, you can work on this feature based on the comments given by PR #1195.
* https://github.com/avrdudes/avrdude/pull/1195#issuecomment-1500980676

landfillbaby, 2023-04-27:
so in my case: 
```sh
$ avrdude -pm32u4/ot|grep write|cut -f3|uniq
eeprom
flash
lfuse
hfuse
efuse
lock
```

```sh
arvdude-elf(){
avrdude -v -pm32u4 -cusbasp -U{eeprom,flash,{l,h,e}fuse,lock}:w:"$1":e
}
```

stefanrueger, 2023-04-27:
Yes, brilliant use of brace expansion and `avrdude -p.../o | grep write | uniq` for writeable memories

mcuee, 2023-04-29:
@landfillbaby 

Is it okay to close this issue? Thanks.

closed 2023-04-29 by mcuee