avrdudes/avrdude #1355: serialupdi => avrdude OS error: unable to read: Resource temporarily unavailable

rgroener, 2023-04-27:
Hi
I use a serialupdi programmer on my Linux machine to program an attiny1616.
Burning the hex file into the attiny works, when I connect the serialupdi-programmer. 
But at the second run, I get the following error message:

`avrdude -p attiny1616 -P /dev/ttyUSB0 -c serialupdi -b 115200    -U flash:w:main.hex 

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9421 (probably t1616)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file main.hex for flash
         with 7856 bytes in 1 section within [0, 0x1eb1]
         using 123 pages and 16 pad bytes, cutting off 2 trailing 0xff bytes
avrdude: writing 7856 bytes flash ...

Writing | ####---------------------------------------------- | 8% 0.36 s 

avrdude OS error: unable to read: Resource temporarily unavailable
avrdude error: write data words operation failed
avrdude error: commit data command failed
 ***failed;`

After taking the programmer off and plugging it in again, it works flawless again.... 
In other words, it works only once after plugging in the programmer.
Has anyone heard of this problem before? Maybe it's not an avrdude problem....
Thanks a lot for any hints in advance.

mcuee, 2023-04-28:
This is for sure not an avrdude issue, but rather the programmer. It could be OS driver issue.
What is the chipset used in your USB to TTL converter? What is the OS used? You may want to upgrade the driver if you can.

rgroener, 2023-04-28:
Thanks for your fast reply. The programmer uses a CH340 chip. Arch Linux is used as OS on a Laptop.
I will check if I can upgrade the OS driver.

mcuee, 2023-04-28:
Ah, if you use an up to date Arch Linux machine as it is a rolling release, usually the Linux kernel version is pretty new. So you may just try to update your Arch Linux machine to the latest.

The other thing, if you have other USB to TTL converter based on CH340 or other converters, you may want to give it a try as well. You can follow [this guide](https://github.com/SpenceKonde/AVR-Guidance/blob/master/UPDI/jtag2updi.md) to make a serialupdi programmer.

HW stability can be another possibility.

rgroener, 2023-04-28:
Thanks for your fast reply. The programmer uses a CH340 chip. Arch Linux is used as OS on a Laptop.

I checked the driver and installed an updated driver version as suggested => ch34x-dkms-git (r44.845e6f5-1)
The programmer seems to run now more reliable, but it is much slower than before.....

Thanks for your help, I will close the issue.

closed 2023-04-28 by rgroener

dl8dtl, 2023-04-28:
Sometimes, it might also help to try everything with an USB hub in between.
Just a couple of days ago, we've been trying to attach a ham radio transceiver to a PC running Ubuntu 22.04. The transceiver uses a few CP2102 chips behind a TI USB2064 hub chip – attachin it always ended up in "… not taking address …" during USB enumerationg. Guess what – using another external USB hub between the PC and the transceiver made everything work flawlessly.

rgroener, 2023-04-28:
Great tip to try if problems still persists with the new driver. Thanks!