avrdudes/avrdude #1362: Add more usage examples to Avrdude documentation

MCUdude, 2023-05-21:
Avrdude supports some nice CLI features like read/write to stdin/stdout (-U) and piping into terminal mode, but none of these features are documented in the avrdude.texi file.

We should perhaps mention that this is possible and provide a few examples under `2.3 Example Command Line Invocations`

I can submit a PR for this, but are there other neat tricks we should add to the texi docs?

MCUdude, 2023-05-21:
Here are the development branch I'm working on.
https://github.com/MCUdude/avrdude/tree/cli-examples

I've currently added the following examples:

**Write stdin to EEPROM**
```
% echo "The quick brown fox" | avrdude -cusbasp -pattiny13 -Ueeprom:w:-:r
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13)
avrdude: reading input file <stdin> for eeprom
         with 20 bytes in 1 section within [0, 0x13]
         using 5 pages and 0 pad bytes
avrdude: writing 20 bytes eeprom ...
Writing | ################################################## | 100% 0.21 s
avrdude: 20 bytes of eeprom written
avrdude: verifying eeprom memory against <stdin>
Reading | ################################################## | 100% 0.01 s
avrdude: 20 bytes of eeprom verified

avrdude done.  Thank you.
```

**Read EEPROM and write to stdout**
```
% avrdude -cusbasp -pattiny13 -Ueeprom:r:-:I
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9007 (probably t13)
avrdude: reading eeprom memory ...
Reading | ################################################## | 100% 0.02 s
avrdude: writing output file <stdout>
:2000000054686520717569636B2062726F776E20666F780AFFFFFFFFFFFFFFFFFFFFFFFFCF // 00000> The_quick_brown_fox.............
:20002000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE0 // 00020> ................................
:00000001FF

avrdude done.  Thank you.
```

**Pipe commands into Avrdude terminal mode**
```
% echo "dump eeprom 0 0x40" | avrdude -cusbasp -patmega328p -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
Reading | ################################################## | 100% 0.02 s
0000  48 65 6c 6c 6f 20 57 6f  72 6c 64 21 00 ff ff ff  |Hello World!....|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude done.  Thank you.
```

mcuee, 2023-05-22:
Some other tips and tricks I collected here. Please check if anything is of interests to the PR as well.
https://github.com/mcuee/avrdude/wiki

stefanrueger, 2023-05-22:
- Using `bash` with `&&` and quell-output-option `-qq` to confirm that AVRDUDE command went OK:
  ```
  $ avrdude -qqc ... p ... -U ... && echo OK
  OK
  ```
- Factory fuse setting of a device:
  ```
  $ avrdude -patmega328p/St | grep initval
  .ptmm	ATmega328P	lfuse	initval	0x62
  .ptmm	ATmega328P	hfuse	initval	0xd9
  .ptmm	ATmega328P	efuse	initval	0xff
  .ptmm	ATmega328P	lock	initval	0xff
  ```
- List of all parts known to AVRDUDE
  ```
  $ avrdude -p*/d | grep = | cut -f2 -d"'"
  ```
- A little bash shell script to create terminal scripts that will reset a part to factory settings
  ```
  mkdir /tmp/factory
  for part in $(avrdude -p*/d | grep = | cut -f2 -d"'"); do
    echo $part
    avrdude -p$part/St | grep initval | cut -f3,5 | grep -ve-1 | sed "s/.*/write &/" >/tmp/factory/$part.ini
  done
  ```
- Terminal input from scripts
  ```
  $ cat /tmp/factory/ATmega168P.ini
  write lfuse	0x62
  write hfuse	0xdf
  write efuse	0xf9
  write lock	0xff

  $ avrdude -qqc usbasp -p ATmega168P -t < /tmp/factory/ATmega168P.ini && echo OK
  OK
  ```
- Output a list of non-bootloader programmers that can be used for a part. Note that `|&` folds stderr into stdout in a `bash` shell. [@MCUDude Could you repair the spelling error Digilient while you are at it?]
  ```
  $ avrdude -c? -pavr32ea32 |& grep -v bootloader
  Valid programmers for part AVR32EA32 are:
    atmelice_updi      = Atmel-ICE (ARM/AVR) via UPDI
    digilent-hs2       = Digilient JTAG HS2 (MPSSE) via ?
    dryrun             = Emulates programming without a programmer via UPDI
    jtag2updi          = JTAGv2 to UPDI bridge via UPDI
    jtag3updi          = Atmel AVR JTAGICE3 via UPDI
    pickit4_updi       = MPLAB(R) PICkit 4 via UPDI
    pkobn_updi         = Curiosity nano (nEDBG) via UPDI
    powerdebugger_updi = Atmel PowerDebugger (ARM/AVR) via UPDI
    serialupdi         = SerialUPDI via UPDI
    snap_updi          = MPLAB(R) SNAP via UPDI
    xplainedmini_updi  = Atmel AVR XplainedMini via UPDI
    xplainedpro_updi   = Atmel AVR XplainedPro via UPDI
  ```
- [Tounge in cheek] Which programmers neglect to define `prog_modes` in `avrdude.conf`?
  ```
  $ avrdude -c*/At | grep prog_modes | grep -v PM_
  .prog	digilent-hs2	prog_modes	0
  ```
  [Actually, which prog_modes should that programmer have? Maybe also repair in the same PR?]
- The example [here](https://github.com/avrdudes/avrdude/issues/1354#issuecomment-1525791333)
- Examples using metadata of the `urclock` programmer: `-xshowall` etc (you may have some from your own practice)

stefanrueger, 2023-05-22:
> Pipe commands into Avrdude terminal mode

Note that (since yesterday) AVRDUDE can also use semicolon-space separated commands in one terminal line:
```
$ echo 'dump eeprom 0 0x40; write eeprom 0 "Bonjour le mode"' | avrdude -cusbasp -patmega328p -t
```

Also, perhaps show examples of use with binary and file input for terminal write

MCUdude, 2023-05-22:
@mcuee and @stefanrueger thanks for the input! There are some _very_ neat and clever commands that deserve to be in the official user manual. I'll continue adding examples, and submit a PR when I'm ready.

closed 2023-05-27 by stefanrueger