avrdudes/avrdude #1367: PICKit 5 support

mcuee, 2023-05-26:
Microchip has released PICKit 5 as an upgrade to PICKit 4.
https://www.microchip.com/en-us/development-tool/PG164150

MCUdude, 2023-05-26:
I wasn't aware that the PICkit5 has already been released!

@xedbg how similar is this to the PICkit4? Is there still a "PIC" and an "AVR" mode? If it "speaks" the same EDBG protocol as the SNAP/PK4, it shouldn't be too tricky to add Avrdude support, which would be really cool to have in the following, 7.2 release!

xedbg, 2023-05-26:
@MCUdude - The PICkit5 to my knowledge supports only the native "PIC" mode, and has implemented AVR (and ARM) support in that mode using a system of byte-code scripts which are drawn from XML within the tool-pack.  This means that it is only supported in MPLAB (and not Studio, or any other CMSIS-DAP-enabled third-party IDE) and even if the protocol were to be published, would unfortunately require a non-trivial amount of work to get it into avrdude.
Also FYI: from MPLABX 6.10, the PICkit4 and Snap will also use this mode for AVR devices.  But the AVR mode still exists in those tools, and Studio will switch them into that mode automatically.

MCUdude, 2023-05-26:
Thanks for the info @xedbg! Do you know if the PICkit4 will be discontinued soon?

> The PICkit5 to my knowledge supports only the native "PIC" mode, and has implemented AVR (and ARM) support in that mode using a system of byte-code scripts which are drawn from XML within the [...]

From our perspective, that's a real bummer. And the fact that it's not supported by Studio tells me that for an AVR developer, the PICkit4 is for now a better choice. And yes, implementing this (new or EDBG derived?) protocol requires a substantial amount of work even if the protocol were to be published.

However, if/when the protocol is released, I hope there will be Avrdude support for it in the future. With all the new features added to Avrdude terminal mode recently, it has not turned into a very potent CLI tool that can be used for way more than just flashing binaries.

> But the AVR mode still exists in those tools, and Studio will switch them into that mode automatically.

Excellent, I hope this feature doesn't get removed then.
I even remember discussing adding this PIC/AVR switching feature to Avrdude. That's still on the Avrdude "features wanted" list, but it required a lot of research, and other things had higher priority.

xedbg, 2023-05-26:
I don't disagree with your view @MCUdude.  I can't comment on the PICkit4 supply situation - there have been many EOLed products in recent times due to component sourcing issues though...

mcuee, 2023-05-26:
> I don't disagree with your view @MCUdude. I can't comment on the PICkit4 supply situation - there have been many EOLed products in recent times due to component sourcing issues though...

Microchip.com shows zero stock for PICKit 4, but I just checked Mouser Singapore and they show 3624 in stock. 
https://www.mouser.sg/ProductDetail/Microchip-Technology/PG164140?qs=r5DSvlrkXmLKDuYNJImLWw%3D%3D

PICKit 5 has not reached here in Singapore yet.
https://www.mouser.sg/ProductDetail/Microchip-Technology/PG164150?qs=ulEaXIWI0c8vHVWqoqZ8Uw%3D%3D

mcuee, 2023-06-01:
Merged into https://github.com/avrdudes/avrdude/issues/1367

closed 2023-06-01 by mcuee

xedbg, 2023-07-18:
FYI
https://www.microchip.com/product-change-notifications/#/19745/CFAN-12EBFB824
The catalog part number (CPN) PG164140 has moved to End of Life (EOL). This catalog part number will no longer be offered after Sept 1, 2023. The replacement part number is: PG164150

MCUdude, 2023-07-18:
The PICkit4 is going EOL? That's bad news for us... Let's hope we'll see PICkit5 support Avrdude in the future. I'll argue that Avrdude is _the_ best CLI programmer for AVR devices, and it would be a shame if the next generation of programmers wouldn't be supported.

stefanrueger, 2023-07-18:
> shame if the next generation of programmers wouldn't be supported

Is there a path for AVRDUDE to support PICKit4/5? @xedbg  mentioned a closed protocol that is likely to be non-trivial. Both of which I consider a serious drawback of the programmer.

mcuee, 2023-07-19:
It is said future version of Microchip Studio will support PICkit 5.
https://forum.microchip.com/s/topic/a5C3l000000BpyJEAS/t391312

MCUdude, 2023-07-19:
> It is said future version of Microchip Studio will support PICkit 5.

That's good to know! Then maybe, maybe there's a chance someone from the Microchip/Atmel division can help out in the future.