avrdudes/avrdude #1368: MPLAB ICE 4 support

mcuee, 2023-05-26:
MPLAB ICE 4 has been released for quite a while.
https://www.microchip.com/en-us/development-tool/DV244140

I am not so sure if there are people who have it and want to play with avrdude. It is rather expensive though.

MCUdude, 2023-05-31:
@xedbg are the ICE 4 similar to the PICkit5 in which it uses byte-code scripts to program/debug AVRs instead of using a dedicated "mode" like the PICkit4 and the SNAP?

If the ICE 4 does it similarly to the PK4/SNAP, it should be trivial to add ICE 4 support even though I obviously don't have an ICE 4 to test with.

mcuee, 2023-05-31:
@MCUdude 

I think same question can be asked for MPLAB ICD5 which is of much lower cost than MPLAB ICE 4 (but a bit more expensive than Atmel-ICE).

xedbg, 2023-05-31:
Yes - all of these tools are equivalent in the way they work with PIC/AVR/MIPS/PIC32C/SAM

xedbg, 2023-05-31:
> > Yes - all of these tools are equivalent in the way they work with PIC/AVR/MIPS/PIC32C/SAM
> 
> So basically PICKit 5, MPLAB ICD 5 and MPLAB ICE 4 work the same as you mentioned in the PICKit 5 issue, right? There is no AVR mode.
> 
Correct.
> > The PICkit5 to my knowledge supports only the native "PIC" mode, and has implemented AVR (and ARM) support in that mode using a system of byte-code scripts which are drawn from XML within the tool-pack. This means that it is only supported in MPLAB (and not Studio, or any other CMSIS-DAP-enabled third-party IDE) and even if the protocol were to be published, would unfortunately require a non-trivial amount of work to get it into avrdude.

MCUdude, 2023-05-31:
OK, so _if_ someone added support for PICkit5 in the future, it would be trivial to add the ICE 4 and ICD 5 as well?
This will absolutely be something Microchip would benefit from, so I hope they put some time and effort into this sometime in the future.

If yes, maybe @mcuee can combine #1367 and #1369 into a single issue; perhaps close this one (#1368) and #1369, and edit #1367?

stefanrueger, 2023-06-01:
> combine https://github.com/avrdudes/avrdude/issues/1367 and https://github.com/avrdudes/avrdude/issues/1369 into a single issue

I agree; it would be great to see the issues list shrink before we release v 7.2 :smile:

mcuee, 2023-06-01:
Done. Closed #1367 and #1369 and merged into this one.

mcuee, 2023-10-26:
It is interesting that Microchip called the tools Gen 4 and Gen 5.

I guess the previous devices are categorized as Gen 1/2/3, roughly like the following (for the AVR side).

**AVR Side**
Gen 1 -- AVRISP, STK500v1, JTAGICE 1
Gen 2 -- AVRISP mkII, STK500v2, JTAGICE mkII, AVR Dragon
Gen 3 - JTAGICE 3, Atmel-ICE, Power Debugger, EDBG, mEDBG, nEBDG

**PIC side**
Gen 1 -- MPLAB ICD 1, PICkit 1, ICE 2000
Gen 2 -- MPLAB ICD 2, PICKIt 2, 
Gen 3 -- PICKit 3, MPLAB ICD 3, MPLAB Real ICE

**SAM side (Atmel ARM)**
SAM-ICE

MPLABX IDE 6.15 release notes:
https://ww1.microchip.com/downloads/aemDocuments/documents/DEV/ProductDocuments/SupportingCollateral/mplabx-ide-release-notes-v6.15.zip

**Hardware Tools - Gen 4**
Release Notes for MPLAB Snap
Release Notes for MPLAB PICkit 4
Release Notes for MPLAB ICD 4
Release Notes for MPLAB ICE 4

**Hardware Tools - Gen 5**
Release Notes for MPLAB PICkit 5
Release Notes for MPLAB ICD 5

xedbg, 2023-10-27:
Not quite right, but close.
AVR generations refer primarily to the codebase architecture and not the hardware directly and uses the 'n'G notation as you see in the pack changelogs.
So:
3G: JTAGICE3, Atmel-ICE, PowerDebugger, EDBG
4G: mEDBG
5G: nEDBG, AVR image in PICkit4 and Snap

mcuee, 2023-10-27:
> Not quite right, but close. AVR generations refer primarily to the codebase architecture and not the hardware directly and uses the 'n'G notation as you see in the pack changelogs. 
> So: 
> 3G: JTAGICE3, Atmel-ICE, PowerDebugger, EDBG 
> 4G: mEDBG 
> 5G: nEDBG, AVR image in PICkit4 and Snap

Thanks for the correction. Still your classification does not match MPLABX IDE 6.15 release notes for HW tools Generation classification.

MPLABX IDE 6.15 release notes:
https://ww1.microchip.com/downloads/aemDocuments/documents/DEV/ProductDocuments/SupportingCollateral/mplabx-ide-release-notes-v6.15.zip

**Hardware Tools - Gen 4**
Release Notes for MPLAB Snap
Release Notes for MPLAB PICkit 4
Release Notes for MPLAB ICD 4
Release Notes for MPLAB ICE 4

**Hardware Tools - Gen 5**
Release Notes for MPLAB PICkit 5
Release Notes for MPLAB ICD 5

xedbg, 2023-10-27:
My classification accounts for legacy Atmel debuggers.  (There is no relationship between "GEN5" and "5G")

mcuee, 2023-10-27:
> My classification accounts for legacy Atmel debuggers. (There is no relationship between "GEN5" and "5G")

I see. Thanks for the clarification.

MX682X, 2024-03-24:
I hope it's ok to revive this old issue.
I had some vacation so I decided to try to brind UPDI Debugging to the Arduino IDE (because of the backbone of DxCore/megaTinyCore) I didn't manage to have the debugging Bugfree, but Programming seems to work like a charm.
I hope this python file will be of use for you guys to figure out how to port it to avrdude.
I'm not sure if the "scripts" are the same for all chips, I admit I only checked it with the AVR32DD28 I had on hand. However the {USER_DIR} \ .mchp_packs\Microchip \ [Debugger] \ [version] \ firmware \ scripts.xml that comes with the install of MPLAB has them all, so it should be possible to build a data extractor (I ran out of time before I could start with that)
[https://github.com/MX682X/PK5-UPDI-GDB-Server/blob/main/pk_debugger_iface.py](https://github.com/MX682X/PK5-UPDI-GDB-Server/blob/main/pk_debugger_iface.py)
The functions you'll probably need are 
"prepare_target_flash" - loads new Data to a buffer
"finalize_download" - will erase the chip, load new program, and verify it
"attach_target" & "detach_target"- establishes/stops Debugger operation

MCUdude, 2024-07-24:
Hi @MX682X! It's impressive what you've been able to accomplish with the PICkit5! It looks like a lot of work, especially because of the lack of protocol documentation. I know Microchip hasn't made it easy, but it would be really cool if we somehow could get Avrdude to support the PICkit5. I know it's a lot to ask, but is this something you'd be willing to look into? Thanks!

MX682X, 2024-07-24:
@MCUdude I was planning to continue the work on it soon, I was just busy with Uni in the last couple of months.
In fact, I was able to figure out how the instruction set of the scripts looks like (ok, not really, it's all guesswork still), so it's possible to do custom functions. (Yes, the scripts.xml is just special assembler).
However, my Problem is, that I have no idea how avrdude works - and frankly, I'd like to keep the number of projects I'm involved with low, otherwise it will be hard to keep track of everything. I can give some hints on how it works, but it's really just the functions you see in the python files.

mcuee, 2024-08-23:
PICkit 5 UPDI support has been added.
* https://github.com/avrdudes/avrdude/pull/1863

MX682X, 2024-11-15:
a small update: If anyone wants to play around with the ISP support, here is the branch: https://github.com/MX682X/avrdude/tree/8.1.0---isp 
However, it's not a release version (Will end up having merge conflicts probably anyway), but enough for test-avrdude to pass with m32u4 and m2560). Should you encounter bugs, open an issue there

MCUdude, 2024-11-15:
@MX682X Awesome! I'm quite busy with other things at the moment, but I'll give it a try when I have time. When you think you have something that works, feel free to provide a draft PR. This way you'll get more test users and overall interest I think 🙂

mcuee, 2025-03-26:
* https://github.com/avrdudes/avrdude/pull/1935 has been merged as well.

stefanrueger, 2025-03-26:
Yes, so PICkit5 support now exists. I haven't followed the discussion. Are Microchip MPLAB ICE 4 and MPLAB ICD 5 still on the card?

mcuee, 2025-03-26:
> Yes, so PICkit5 support now exists. I haven't followed the discussion. Are Microchip MPLAB ICE 4 and MPLAB ICD 5 support still on the card?

It will be good to have the support ICD 5 and ICE 4 if possible. I believe @MX682X tested one of them and that did not work.

On the other hand PICkit 5, PICkit Basic and SNAP are more cost effective for most of the users.

closed 2025-05-22 by mcuee