avrdudes/avrdude #1370: Bitmasks from SPI read/write commands for fuses/lockbits sometimes differ from .ATDF bitmasks

stefanrueger, 2023-05-27:
The datasheet suggests that the efuse of the ATtiny828 uses all bits bar Bit 3, however, the `avrdude.conf` file writes only bits 0-4:
https://github.com/avrdudes/avrdude/blob/342330944990d4d13c669d5d3f74b9de79fbce84/src/avrdude.conf.in#L9068-L9074 

[image]

This indicated an error in Attiny828's efuse write command, which needs changing to
```
  write              = "1010.1100--1010.0100--xxxx.xxxx--iiii.1iii"; 
```

The table below lists more discrepancies; it was generated by comparing the bitmasks for fuses that can be generated from .atdf files and the ones generated by the `get_fuse_bitmask()` function in `avr.c` (making use of SPI read/write commands). 

Bitmasks for fuses and lock bits generated by different methods

| Part | Memory | avrdude.conf | .atdf |
| --: | --: | --: | --: |
| AT90S2333 | fuse | 0x1f | 0x3f |
| AT90S2343 | fuse | 0x01 | 0x21 |
| AT90S2343 | lock | 0x00 | 0x06 |
| AT90S2323 | fuse | 0x01 | 0x21 |
| AT90S2323 | lock | 0x00 | 0x06 |
| ATtiny22 | fuse | 0x01 | 0x21 |
| ATtiny22 | lock | 0x00 | 0x06 |
| AT90S4433 | fuse | 0x1f | 0x3f |
| AT90S4434 | fuse | 0x1f | 0x21 |
| AT90S8535 | fuse | 0x01 | 0x21 |
| AT90S8535 | lock | 0x00 | 0x06 |
| ATmega103 | fuse | 0x0b | 0x2b |
| ATmega324PB | efuse | 0x07 | 0x0f |
| ATmega162 | efuse | 0x07 | 0x1e |
| ATmega163 | lfuse | 0xcf | 0xef |
| ATmega161 | fuse | 0x5f | 0x3f |
| ATtiny828 | efuse | 0x1f | 0xf7 |
| ATtiny828R | efuse | 0x1f | 0xf7 |

There may be innocuous reasons for discrepancies (eg, read-only bits), they could come about through an error in `avrdude.conf` (as above) or by the `.atdf` being wrong (well, different to the data sheet). This needs more exploration.

mcuee, 2023-05-28:
I am not so sure whether we care about the old AT90S parts and ATtiny22L which have been long obsoleted. But we may need to check those active parts.

stefanrueger, 2023-05-29:
PR #1372 cares about the old AT90S parts and ATtiny22L in the hope that programming them will be a better experience (even if they are no longer sold, they will still be around). @mcuee Thanks for all the docs.

PR #1372 (2023-05-31): Initialise unused bits of classic-part fuses with factory defaults
Fixes #1371
Fixes #1370
 - Fixes ATmega161's sut, spien and bootrst bit masks
 - Fixes several SPI fuse/lock read/write commands
      + AT90S2323   fuse    read
      + AT90S2323   lock    read
      + AT90S2343   fuse    read
      + AT90S2343   lock    read
      + AT90S4434   fuse    read
      + AT90S4434   fuse    write
      + AT90S4434   lock    read
      + AT90S8535   fuse    read
      + AT90S8535   lock    read
      + ATmega161   fuse    write
      + ATmega162   efuse   write
      + ATmega324PB efuse   write
     [... cut]
closed 2023-05-31 by stefanrueger