avrdudes/avrdude #1379: Wiring bootloader issue

mcuee, 2023-06-04:
Reference:
* https://github.com/avrdudes/avrdude/pull/1376#issuecomment-1574982167

This may not be an avrdude issue but a Wiring bootloader FW issue.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c wiring -P COM11 -p m2560 -t
avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e9801 (probably m2560)
avrdude> dump flash 0x400 0x10
Reading | ################################################## | 100% 0.11 s
00400  00 00 00 00 00 00 ff ff  ff ff ff ff ff ff ff ff  |................|
avrdude> write flash 0x400 0xaa 0x55 0xaa 0x55 0xaa 0x55
Caching | ################################################## | 100% 0.02 s
avrdude> flush
avrdude_git: synching cache to device ...
Reading | ################################################## | 100% 35.30 s
Erasing | -------------------------------------------------- | 0% 0.01 s
avrdude_git error: command failed
avrdude_git error: chip erase failed
avrdude> quit
avrdude_git: synching cache to device ...
Reading | ################################################## | 100% 0.01 s
Erasing | -------------------------------------------------- | 0% 0.01 s
avrdude_git error: command failed
avrdude_git error: chip erase failed

avrdude_git done.  Thank you.
```

mcuee, 2023-06-04:
Bootloader FW source codes:
https://github.com/MCUdude/MegaCore/tree/master/avr/bootloaders/stk500v2

It is a minor fix of the original Arduino version which has EEPROM bug.

mcuee, 2023-06-04:
BTW, I have some HW stability issues with this bootloader before with one paticular board. But this board seems to work normally under Arduino, so the issue may be different.

* https://github.com/MCUdude/MegaCore/issues/190
* https://github.com/avrdudes/avrdude/issues/994

mcuee, 2023-06-04:
I was worrying about timeout in terminal mode but using pioeline does not help. 

```
$ echo "write flash 0x400 0x55 0xaa 0x55 0xaa" | ./avrdude -c usbasp -p m2560 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9801 (probably m2560)
avrdude> write flash 0x400 0x55 0xaa 0x55 0xaa
Caching | ################################################## | 100% 0.06 s
avrdude> avrdude: synching cache to device ...
Reading | ################################################## | 100% 42.23 s
Erasing | ################################################## | 100% 0.16 s
Writing | ################################################## | 100% 2.59 s

avrdude done.  Thank you.

$ echo "write flash 0x400 0xaa 0x55 0xaa 0x55" | ./avrdude -c wiring -P COM11 -p m2560 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9801 (probably m2560)
avrdude> write flash 0x400 0xaa 0x55 0xaa 0x55
Caching | ################################################## | 100% 0.07 s
avrdude> avrdude: synching cache to device ...
Reading | ################################################## | 100% 34.79 s
Erasing | -------------------------------------------------- | 0% 0.01 s
avrdude error: command failed
avrdude error: chip erase failed

avrdude done.  Thank you.

$ echo "dump flash 0x400 0x10" | ./avrdude -c usbasp -p m2560 -qqt
avrdude> dump flash 0x400 0x10
00400  00 00 00 00 ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
avrdude>
```

mcuee, 2023-06-04:
I think this is a limitation of the Arduino STK500v2 bootloader limitation (no support of page erase)

No issues with urboot bootloader.
```
$ echo "dump flash 0x400 0x10" | ./avrdude -c urclock -P COM11 -p m2560 -qqt
avrdude> dump flash 0x400 0x10
00400  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
avrdude>

$ echo "write flash 0x400 0x55 0xaa 0x55 0xaa" | ./avrdude -c urclock -P COM11 -p m2560 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9801 (probably m2560)
avrdude> write flash 0x400 0x55 0xaa 0x55 0xaa
Caching | ################################################## | 100% 0.07 s
avrdude> avrdude: synching cache to device ...
Writing | ################################################## | 100% 0.06 s

avrdude done.  Thank you.

$ echo "write flash 0x400 0xaa 0x55 0xaa 0x55" | ./avrdude -c urclock -P COM11 -p m2560 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9801 (probably m2560)
avrdude> write flash 0x400 0xaa 0x55 0xaa 0x55
Caching | ################################################## | 100% 0.07 s
avrdude> avrdude: synching cache to device ... done

avrdude done.  Thank you.

$ echo "dump flash 0x400 0x10" | ./avrdude -c urclock -P COM11 -p m2560 -qqt
avrdude> dump flash 0x400 0x10
00400  aa 55 aa 55 ff ff ff ff  ff ff ff ff ff ff ff ff  |.U.U............|
avrdude>
```

mcuee, 2023-06-04:
No issues using the Optiboot bootloader from MegaCore either.

```
$ echo "dump flash 0x400 0x10" | ./avrdude -c arduino -P COM11 -p m2560 -qqt
avrdude> dump flash 0x400 0x10
00400  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
avrdude>

$ echo "write flash 0x400 0xaa 0x55 0xaa 0x55" | ./avrdude -c arduino -P COM11 -p m2560 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9801 (probably m2560)
avrdude> write flash 0x400 0xaa 0x55 0xaa 0x55
Caching | ################################################## | 100% 0.07 s
avrdude> avrdude: synching cache to device ...
Writing | ################################################## | 100% 0.06 s

avrdude done.  Thank you.

$ echo "write flash 0x400 0x55 0xaa 0x55 0xaa" | ./avrdude -c arduino -P COM11 -p m2560 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9801 (probably m2560)
avrdude> write flash 0x400 0x55 0xaa 0x55 0xaa
Caching | ################################################## | 100% 0.08 s
avrdude> avrdude: synching cache to device ... done

avrdude done.  Thank you.

$ echo "dump flash 0x400 0x10" | ./avrdude -c arduino -P COM11 -p m2560 -qqt
avrdude> dump flash 0x400 0x10
00400  55 aa 55 aa ff ff ff ff  ff ff ff ff ff ff ff ff  |U.U.............|
avrdude>
```

mcuee, 2023-06-04:
As mentioned in the comments of PR #1376, that PR does not help for this issue.

```
$ echo "dump flash 0x400 0x10" | ./avrdude -c arduino -P COM11 -p m2560 -qqt
avrdude> dump flash 0x400 0x10
00400  55 aa 55 aa ff ff ff ff  ff ff ff ff ff ff ff ff  |U.U.............|
avrdude>

$ echo "dump flash 0x400 0x10" | ./avrdude_pr1376 -c wiring -P COM11 -p m2560 -qqt
avrdude> dump flash 0x400 0x10
00400  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
avrdude>

$ echo "write flash 0x400 0x55 0xaa 0x55 0xaa" | ./avrdude_pr1376 -c wiring -P COM11 -p m2560 -t
avrdude_pr1376: AVR device initialized and ready to accept instructions
avrdude_pr1376: device signature = 0x1e9801 (probably m2560)
avrdude> write flash 0x400 0x55 0xaa 0x55 0xaa
Caching | ################################################## | 100% 0.08 s
avrdude> avrdude_pr1376: synching cache to device ...
Writing | ################################################## | 100% 0.07 s

avrdude_pr1376 done.  Thank you.

$ echo "dump flash 0x400 0x10" | ./avrdude_pr1376 -c wiring -P COM11 -p m2560 -qqt
avrdude> dump flash 0x400 0x10
00400  55 aa 55 aa ff ff ff ff  ff ff ff ff ff ff ff ff  |U.U.............|
avrdude>
xiaof@LAPTOP-913SIMUD MINGW64 /c/work/avr/avrdude_test/avrdude_bin

avrdude_pr1376: AVR device initialized and ready to accept instructions
avrdude_pr1376: device signature = 0x1e9801 (probably m2560)
avrdude> write flash 0x400 0xaa 0x55 0xaa 0x55
Caching | ################################################## | 100% 0.08 s
avrdude> avrdude_pr1376: synching cache to device ...
Reading | ################################################## | 100% 34.86 s
Erasing | -------------------------------------------------- | 0% 0.01 s
avrdude_pr1376 error: command failed
avrdude_pr1376 error: chip erase failed

avrdude_pr1376 done.  Thank you.

$ echo "dump flash 0x400 0x10" | ./avrdude_pr1376 -c wiring -P COM11 -p m2560 -qqt
avrdude> dump flash 0x400 0x10
00400  00 00 00 00 ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
avrdude>
```

mcuee, 2023-06-04:
I will conclude this to be a limitation of the Arduino STK500v2 bootloader for ATmega2560.

closed 2023-06-04 by mcuee

mcuee, 2023-06-09:
@stefanrueger 

I was wrong previously in #1337. Therefore, I would like to double check with you if my conclusion that this is an Arduino STK500 V2 bootloader limitation and not an avrdude issue. Thanks.

stefanrueger, 2023-06-09:
I think so. 

 - The chip erase command fails, so guessing this is owing to the wiring bootloader not having the code to do a chip erase (bad), telling the programmer CE failed (good) upon which avrdude stops programming (sane).
 - Optiboot does not erase the chip either (bad) but *pretends* it did so (worse) to dupe `avrdude -c arduino` into flashing anyway (somehow mitigated by the fact that optiboot erases pages before writing them): you can tell b/c when a part is flashed with optiboot the remains of a previous bigger program are still in flash. That is a *security issue* because if that still visible part of the older program contained the secret Coca Cola recipe it could still be read despite a CE.
 - Urboot bootloaders tell the uploader `-c urlcock` whether or not they can erase the chip upon which either avrdude lets the bootloader erase the chip or does a (slow) chip erase by sending empty pages (0xff bytes only). 
 - `avrdude -c urclock` normally recognises optiboot bootloaders and therefore erases the chip for optiboot

But yes, if you have the source of the wiring bootloader it will become obvious what it does when asked to do a CE.

mcuee, 2023-06-09:
@stefanrueger 

I am not sure if we can fix the bootloader or provide a work-around in the `-c wiring` programmer. Please help to check if you got a chance.

Or we can just leave this as it is if you feel it is not worth the time to fix a very old bootloader.

stefanrueger, 2023-06-09:
> provide a work-around

One work-around is to use `-D` asking AVRDUDE to not do a CE

mcuee, 2023-06-09:
> > provide a work-around
> 
> One work-around is to use `-D` asking AVRDUDE to not do a CE

Yes `-D` is used already in some of the tests.
```
$ echo "write flash 0x400 0x55 0xaa 0x55 0xaa" | ./avrdude -c wiring -P COM11 -p m2560 -D -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9801 (probably m2560)
avrdude> write flash 0x400 0x55 0xaa 0x55 0xaa
Caching | ################################################## | 100% 0.08 s
avrdude> avrdude: synching cache to device ...
Reading | ################################################## | 100% 34.77 s
Erasing | -------------------------------------------------- | 0% 0.01 s
avrdude error: command failed
avrdude error: chip erase failed

avrdude done.  Thank you.
```

mcuee, 2023-06-09:
New tests with a refresh bootloader.

```
PS C:\work\avr\avrdude_test\avrdude_bin> echo "write flash 0x400 0x55 0xaa 0x55 0xaa" | .\avrdude -c wiring -P COM11 -p m2560 -D -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9801 (probably m2560)
avrdude> write flash 0x400 0x55 0xaa 0x55 0xaa
Caching | ################################################## | 100% 0.08 s
avrdude> avrdude: synching cache to device ...
Writing | ################################################## | 100% 0.07 s

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> echo "write flash 0x400 0xaa 0x55 0xaa 0x55" | .\avrdude -c wiring -P COM11 -p m2560 -D -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9801 (probably m2560)
avrdude> write flash 0x400 0xaa 0x55 0xaa 0x55
Caching | ################################################## | 100% 0.08 s
avrdude> avrdude: synching cache to device ...
Reading | ################################################## | 100% 35.47 s
Erasing | -------------------------------------------------- | 0% 0.01 s
avrdude error: command failed
avrdude error: chip erase failed
avrdude: synching cache to device ...
Reading | ################################################## | 100% 0.01 s
Erasing | -------------------------------------------------- | 0% 0.01 s
avrdude error: command failed
avrdude error: chip erase failed

avrdude done.  Thank you.
```

mcuee, 2023-06-09:
BTW, this fix does not help.
* https://github.com/arduino/Arduino-stk500v2-bootloader/pull/2

stefanrueger, 2023-06-09:
Ah, I see.

I do wonder why the terminal cache decides it needs to do a read/CE/write cycle. That is not necessary if the bootloader erases the page before writing it. The code you pointed me to, @mcuee, has (on my reading) a few peculiarities
 - Maintains different addresses for erasing and writing the page to flash
 - Only works for flash programming when input files start from address zero and do not have holes, otherwise the bootloader erases a different page than it writes to (and programming fails)
 - Assumes the bootloader is protected by preventing erasing the page, *but* it is still written to (so not protected)

This code
https://github.com/MCUdude/MegaCore/blob/8b5f16e1da21b6c3014d9accd3bb801a07ed1938/avr/bootloaders/stk500v2/stk500boot.c#L1003-L1025

better be

```
    // Write only to main section (bootloader protection)
    if(tempAddress < APP_END) {
      boot_page_erase(tempAddress); // Erase page before writing
      boot_spm_busy_wait();

      do {
        lowByte  = *p++;
        highByte = *p++;
        data     = (highByte << 8) | lowByte;
        boot_page_fill(address, data);
        address += 2;
        size    -= 2;
      } while (size);

      boot_page_write(tempaddress);
      boot_spm_busy_wait();
      boot_rww_enable();
    }
```

With above (untested!) suggested code the `eraseAddress` can be ditched (if kept there will be a compiler warning). There are a few more traps lurking in the original code: If a malicious uploader asks to program an odd size the code  loops  endlessly; defuse this by making size an `int` and condition  on `do { ... } while(size > 0)`, but anyhow sth to the effect of above would improve the bootloader.

That are the kind of problems that are *really hard* to work-around in AVRDUDE, and I would prefer people fix their wiring bootloader.

mcuee, 2023-06-10:
> That are the kind of problems that are really hard to work-around in AVRDUDE, and I would prefer people fix their wiring bootloader.

@stefanrueger 

I agree with you. I will try your suggestions and incorporate some of the changes mentioned in other places to see if I can get it working better.

For Arduino people, they probably do not care about this issue (and they do not really care about EEPROM read/write using avrdude either). So I guess they will not fix it. But @MCUdude may be interested to fix the bootloader in MegaCore.

mcuee, 2023-06-10:
@MCUdude 

You may be interested to know about this issue and some other issues about the wiring bootloader.

* https://github.com/arduino/ArduinoCore-avr/issues/23 (already fixed in your repo)
* https://github.com/arduino/ArduinoCore-avr/issues/129 (I believe it was fixed in your repo as well)
* https://github.com/arduino/ArduinoCore-avr/issues/173 (I believe still exist in your repo)
* https://github.com/arduino/ArduinoCore-avr/issues/172 (I believe still exist in your repo)
* https://github.com/arduino/ArduinoCore-avr/issues/191 (still exist in your repo, confirmed)
* https://github.com/arduino/ArduinoCore-avr/issues/201  (I believe still exist in your repo)
* https://github.com/msproul/Arduino-stk500v2-bootloader/issues/7 ( I believe still exist in your repo)

I have not checked the details of the other issues reported here.
https://github.com/msproul/Arduino-stk500v2-bootloader/issues

mcuee, 2023-06-10:
@stefanrueger 

With your fix, it seems to work now.

I have also incorporated the changes from the following two places.
* https://github.com/arduino/ArduinoCore-avr/issues/173
* https://github.com/arduino/Arduino-stk500v2-bootloader/pull/2

```
$ diff -u stk500boot_org.c stk500boot.c
--- stk500boot_org.c    2022-10-15 16:30:43.793333800 +0800
+++ stk500boot.c        2023-06-10 15:20:23.349522400 +0800
@@ -268,12 +268,12 @@
 //#define BOOTSIZE 1024
 #if FLASHEND > 0x0F000
        #define BOOTSIZE 8192
+       #define APP_END  ((FLASHEND - BOOTSIZE) + 1)
 #else
        #define BOOTSIZE 2048
+       #define APP_END  (FLASHEND - (2*BOOTSIZE) + 1)
 #endif

-#define APP_END  (FLASHEND -(2*BOOTSIZE) + 1)
-
 /*
  * Signature bytes are not available in avr-gcc io_xxx.h
  */
@@ -532,7 +532,7 @@
 int main(void)
 {
        address_t               address                 =       0;
-       address_t               eraseAddress    =       0;
+//     address_t               eraseAddress    =       0;
        unsigned char   msgParseState;
        unsigned int    ii                              =       0;
        unsigned char   checksum                =       0;
@@ -972,7 +972,7 @@
                                        break;
        #endif
                                case CMD_CHIP_ERASE_ISP:
-                                       eraseAddress    =       0;
+//                                     eraseAddress    =       0;
                                        msgLength               =       2;
                                //      msgBuffer[1]    =       STATUS_CMD_OK;
                                        msgBuffer[1]    =       STATUS_CMD_FAILED;      //*     isue 543, return FAILED instead of OK
@@ -1000,29 +1000,24 @@

                                                if ( msgBuffer[0] == CMD_PROGRAM_FLASH_ISP )
                                                {
-                                                       // erase only main section (bootloader protection)
-                                                       if (eraseAddress < APP_END )
-                                                       {
-                                                               boot_page_erase(eraseAddress);  // Perform page erase
-                                                               boot_spm_busy_wait();           // Wait until the memory is erased.
-                                                               eraseAddress += SPM_PAGESIZE;   // point to next page to be erase
+                                                       // Write only to main section (bootloader protection)
+                                                       if(tempaddress < APP_END) {
+                                                         boot_page_erase(tempaddress); // Erase page before writing
+                                                         boot_spm_busy_wait();
+
+                                                         do {
+                                                               lowByte  = *p++;
+                                                               highByte = *p++;
+                                                               data     = (highByte << 8) | lowByte;
+                                                               boot_page_fill(address, data);
+                                                               address += 2;
+                                                               size    -= 2;
+                                                         } while (size);
+
+                                                         boot_page_write(tempaddress);
[... 29 lines cut ...]
-               theChar =       pgm_read_byte_near((uint16_t)dataPtr++);
+               theChar =       pgm_read_byte_near((uint32_t)dataPtr++);
        #endif
                if (theChar != 0)
                {

PS C:\work\avr\avrdude_test\avrdude_bin> echo "write flash 0x400 0xaa 0x55 0xaa 0x55" | .\avrdude -c wiring -P COM11 -p m2560 -D -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9801 (probably m2560)
avrdude> write flash 0x400 0xaa 0x55 0xaa 0x55
Caching | ################################################## | 100% 0.08 s
avrdude> avrdude: synching cache to device ...
Writing | ################################################## | 100% 0.07 s

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> echo "write flash 0x400 0x55 0xaa 0x55 0xaa" | .\avrdude -c wiring -P COM11 -p m2560 -D -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9801 (probably m2560)
avrdude> write flash 0x400 0x55 0xaa 0x55 0xaa
Caching | ################################################## | 100% 0.09 s
avrdude> avrdude: synching cache to device ... done

avrdude done.  Thank you.
```

mcuee, 2023-06-10:
@MCUdude 

Maybe you want to give the above mod a try. Thanks.

stefanrueger, 2023-06-10:
What's the big deal about a wiring bootloader for the m2560 et al? Urboot works, and at a *fraction* of used space, too?

@mcuee: I don't get the`2*` in the  line `#define APP_END  (FLASHEND - (2*BOOTSIZE) + 1)`. Why keep a gap of `BOOTSIZE` `between the application and the bootloader? Oh well, I suspect I don't want to know.

mcuee, 2023-06-10:
> What's the big deal about a wiring bootloader for the m2560 et al? 

@stefanrueger 

The only reason is that Arduino ships the ancient stk500v2 bootloader with their Arduino Mega (Mega2560). So all the clones will do the same as well.

@MCUdude is mainly using Optiboot with MegaCore but he added the STK500v2 bootloader last year so as to make it possible to use MegaCore with any Arduino Mega 2560 without replacing the bootloader or uploading using an ISP programmer
https://github.com/MCUdude/MegaCore/commit/2b1655dcc7b64ae3b761834842b2695fbe3dab82

Then I informed him about the EEPROM issue and he updated the stk500v2 bootloader a bit later.
https://github.com/MCUdude/MegaCore/commit/eba4f4a827aa3597b8e65144caee83f69724f280

mcuee, 2023-06-10:
> @mcuee: I don't get the`2*` in the line `#define APP_END (FLASHEND - (2*BOOTSIZE) + 1)`. Why keep a gap of `BOOTSIZE` `between the application and the bootloader? Oh well, I suspect I don't want to know.

@stefanrueger 

It is from the original code and the following fixes it for ATmega2560 but leaves that code for smaller Flash parts. I guess it may be related to the size of bootloader section in some parts. But even in that case, the proper fix is probably to define `APP_END` for each part if a single formula can not cut it.
* https://github.com/msproul/Arduino-stk500v2-bootloader/issues/7
* https://github.com/arduino/Arduino-stk500v2-bootloader/pull/2

In any case, this Arduino stk500v2 bootloader is only popular in ATmega1280/2560 and similar larger Flash part.

mcuee, 2023-06-10:
> Urboot works, and at a fraction of used space, too?

@stefanrueger and @MCUdude 

I agree. So I created an issue here.

mcuee, 2023-06-10:
@stefanrueger 

BTW, there is a fork here. No idea how good it is.
https://github.com/joede/stk500v2-bootloader

But stk500v2 is used in the XMega For Arduino project along with Caterina (USB AVR109 protocol)
https://github.com/XMegaForArduino/arduino/tree/master/bootloaders/xmega

mcuee, 2023-06-10:
> More interestingly there is a port for PIC32 MIPS32 core MCUs.
> https://github.com/chipKIT32/PIC32-avrdude-bootloader
> https://github.com/chipKIT32/pic32-Arduino-Bootloader
> https://github.com/chipKIT32/pic32-Arduino-USB-Bootloader-original

I am not so sure about rhe avrdude version they are using. One of the above repo says avrdude 5.8.

There is a website which has the 5.10 binary and corresponding avrdude.conf file.
http://www.cpustick.polydevelopment.com/bootloader.htm

The avrdude.conf file here has some entries for PIC32 MCUs.
http://www.cpustick.polydevelopment.com/downloads/avrdude.conf
```

#------------------------------------------------------------
# Pic32 changes from Mark
#------------------------------------------------------------

part
    id               = "pic32";
    desc             = "32MX795F512L";
    signature        = 0x50 0x49 0x43;
    has_jtag         = yes;
#   stk500_devcode   = 0x??; # No STK500v1 support?
#   avr910_devcode   = 0x??; # Try the ATmega16 one
    avr910_devcode   = 0x74;
    pagel            = 0xd7;
    bs2              = 0xa0;
    chip_erase_delay = 9000;
    pgm_enable       = "1 0 1 0  1 1 0 0    0 1 0 1  0 0 1 1",
                       "0 0 0 0  0 0 0 0    0 0 0 0  0 0 0 0";

    chip_erase       = "1 0 1 0  1 1 0 0    1 0 0 0  0 0 0 0",
                       "0 0 0 0  0 0 0 0    0 0 0 0  0 0 0 0";

    timeout             = 200;
    stabdelay           = 100;
    cmdexedelay         = 25;
    synchloops          = 32;
    bytedelay           = 0;
    pollindex           = 3;
    pollvalue           = 0x53;
    predelay            = 1;
    postdelay           = 1;
    pollmethod          = 1;

    pp_controlstack     =
        0x0E, 0x1E, 0x0F, 0x1F, 0x2E, 0x3E, 0x2F, 0x3F,
        0x4E, 0x5E, 0x4F, 0x5F, 0x6E, 0x7E, 0x6F, 0x7F,
        0x66, 0x76, 0x67, 0x77, 0x6A, 0x7A, 0x6B, 0x7B,
        0xBE, 0xFD, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00;
    hventerstabdelay    = 100;
    progmodedelay       = 0;
    latchcycles         = 5;
    togglevtg           = 1;
    poweroffdelay       = 15;
    resetdelayms        = 1;
    resetdelayus        = 0;
    hvleavestabdelay    = 15;
    chiperasepulsewidth = 0;
    chiperasepolltimeout = 10;
    programfusepulsewidth = 0;
    programfusepolltimeout = 5;
    programlockpulsewidth = 0;
    programlockpolltimeout = 5;

    idr                 = 0x31;
    spmcr               = 0x57;
    allowfullpagebitstream = no;

[... 683 lines cut ...]

    memory "lock"
        size            = 1;
        read            = "0 1 0 1  1 0 0 0   0 0 0 0  0 0 0 0",
                          "x x x x  x x x x   x x o o  o o o o";

        write           = "1 0 1 0  1 1 0 0   1 1 1 x  x x x x",
                          "x x x x  x x x x   1 1 i i  i i i i";
        min_write_delay = 9000;
        max_write_delay = 9000;
      ;

    memory "calibration"
        size            = 1;
        read            = "0 0 1 1  1 0 0 0    x x x x  x x x x",
                          "0 0 0 0  0 0 0 0    o o o o  o o o o";
      ;

    memory "signature"
        size            = 3;
        read            = "0  0  1  1   0  0  0  0   x  x  x  x   x  x  x  x",
                          "x  x  x  x   x  x a1 a0   o  o  o  o   o  o  o  o";
      ;
  ;
```

mcuee, 2023-06-10:
And interestingly we were discussing stk500v2 bootloader for PIC32 recently here.
* https://github.com/avrdudes/avrdude/discussions/1353