avrdudes/avrdude #1382: Option `-xnoreset`for `-c arduino` and `-c urclock`

mcuee, 2023-06-05:
From here:
* https://github.com/avrdudes/avrdude/issues/1337#issuecomment-1575603430

@stefanrueger mentioned the following.
> Basically `-c arduino` and `-c stk500v1` are the same except `-c arduino` tries to reset the connected board and EEPROM treatment of `-c arduino` is compatible with Arduino as ISP, whilst EEPROM treatment of `-c stk500v1` is compatible with the STK500 programmer with v1 firmware.
>
> Maybe we should give `-c arduino` and the (better) alternative `-c urclock` an option `-xnoreset`...

mcuee, 2023-06-05:
@stefanrueger 

Just wondering whether it make sense for a programmer (not a bootloader) to implement a protocol similar to `urboot`.

For protocols like STK500V1 and STK500V2, there are both programmer and bootloader implementations.

stefanrueger, 2023-06-05:
@mcuee Checking the code base I now think PR #1383 is possibly a better solution. Could you check?

stefanrueger, 2023-06-05:
> Just wondering whether it make sense for a programmer (not a bootloader) to implement [urprotocol].

They could. Someone could write an `urclock as ISP` sketch or so and it would be a simple case of a new programmer entry in `avrdude.conf` to serve that programmer. Having said this, urprotocol, was made for bootloaders so they can be small. This is done by shifting work to the uploader/downloader, so there will be limited benefits for a programmer that does not have the same space constraints as a bootloader.