avrdudes/avrdude #1384: Poor programmer documentation

stefanrueger, 2023-06-05:
AVRDUDE supports some 155 programmers, and growing, but in many cases there is little documentation about the supported programmer. This [list](https://github.com/avrdudes/avrdude/discussions/1019#discussioncomment-4146009) is a good start for collecting URLs and background info. The aim of this issue would be to, at the very least provide each programmer with an URL with some background (where to buy, how to build that programmer, its data sheet or product sheet or similar) as comment between title and programmer definition in `avrdude.conf`; I am sure the project would be happy to receive PRs improving this.

mcuee, 2023-06-05:
Reference discussion here:

* https://github.com/avrdudes/avrdude/discussions/1019

mcuee, 2023-06-06:
Early draft in #1019.

Draft: 15-Nov-2022
Programmer info URL list.

Name | Description | URL
-- | -- | --
2232HIO | FT2232H based generic programmer |  
4232h | FT4232H based generic programmer |  
adafruit_gemma | Arduino Gemma bootloader disguised as   USBtiny |  
arduino | Arduino skeleton STK500v1 compatible   bootloader |  
arduino_gemma | Adafruit Trinket Gemma bootloader disguised   as USBtiny |  
arduino-ft232r | Arduino: FT232R connected to ISP |  
arduinoisp | Arduino ISP Programmer |  
arduinoisporg | Arduino ISP Programmer |  
atmelice | Atmel-ICE (ARM/AVR) in JTAG mode |  
atmelice_dw | Atmel-ICE (ARM/AVR) in debugWIRE mode |  
atmelice_isp | Atmel-ICE (ARM/AVR) in ISP mode |  
atmelice_pdi | Atmel-ICE (ARM/AVR) in PDI mode |  
atmelice_updi | Atmel-ICE (ARM/AVR) in UPDI mode |  
avr109 | Atmel AppNote AVR109 Boot Loader |  
avr910 | Atmel Low Cost Serial Programmer |  
avr911 | Atmel AppNote AVR911 AVROSP |  
avrftdi | FT2232D based generic programmer |  
avrisp | Atmel AVR ISP |  
avrisp2 | Atmel AVR ISP mkII |  
avrispmkII | Atmel AVR ISP mkII |  
avrispv2 | Atmel AVR ISP V2 |  
buspirate | The Bus Pirate |  
buspirate_bb | The Bus Pirate (bitbang interface, supports   TPI) |  
butterfly | Atmel Butterfly Development Board |  
butterfly_mk | Mikrokopter.de Butterfly |  
bwmega | BitWizard ftdi_atmega builtin programmer |  
c232hm | C232HM cable from FTDI |  
c2n232i | serial port banging, reset |  
dasa | serial port banging, reset |  
dasa3 | serial port banging, reset |  
diecimila | alias for arduino-ft232r |  
dragon_dw | Atmel AVR Dragon in debugWire mode |  
dragon_hvsp | Atmel AVR Dragon in HVSP mode |  
dragon_isp | Atmel AVR Dragon in ISP mode |  
dragon_jtag | Atmel AVR Dragon in JTAG mode |  
dragon_pdi | Atmel AVR Dragon in PDI mode |  
dragon_pp | Atmel AVR Dragon in PP mode |  
ehajo-isp | avr-isp-programmer from eHaJo | http://www.eHaJo.de
flip1 | FLIP USB DFU protocol version 1 (doc7618) |  
flip2 | FLIP USB DFU protocol version 2 (AVR4023) |  
ft232h | FT232H in MPSSE mode |  
ft232r | FT232R Synchronous BitBang |  
ft245r | FT245R Synchronous BitBang |  
iseavrprog | USBtiny-based programmer |  https://iascaled.com
jtag1 | Atmel JTAG ICE (mkI) |  
jtag1slow | Atmel JTAG ICE (mkI) |  
jtag2 | Atmel JTAG ICE mkII |  
jtag2avr32 | Atmel JTAG ICE mkII im AVR32 mode |  
jtag2dw | Atmel JTAG ICE mkII in debugWire mode |  
jtag2fast | Atmel JTAG ICE mkII |  
jtag2isp | Atmel JTAG ICE mkII in ISP mode |  
jtag2pdi | Atmel JTAG ICE mkII PDI mode |  
jtag2slow | Atmel JTAG ICE mkII |  
jtag2updi | JTAGv2 to UPDI bridge |  
jtag3 | Atmel AVR JTAGICE3 in JTAG mode |  
jtag3dw | Atmel AVR JTAGICE3 in debugWIRE mode |  
jtag3isp | Atmel AVR JTAGICE3 in ISP mode |  
jtag3pdi | Atmel AVR JTAGICE3 in PDI mode |  
jtag3updi | Atmel AVR JTAGICE3 in UPDI mode |  
jtagkey | Amontec JTAGKey, JTAGKey-Tiny and JTAGKey2 |  
jtagmkI | Atmel JTAG ICE (mkI) |  
jtagmkII | Atmel JTAG ICE mkII |  
jtagmkII_avr32 | Atmel JTAG ICE mkII im AVR32 mode |  
ktlink | KT-LINK FT2232H interface with IO switching   and voltage buffers. |  
linugpio | Linux   GPIO programmer |  
linuxspi | Linux   SPI programmer |  
lm3s811 | Luminary Micro LM3S811 Eval Board (Rev. A) |  
mib510 | Crossbow MIB510 programming board |  
micronucleus | Micronucleus Bootloader |  
mkbutterfly | Mikrokopter.de Butterfly |  
nibobee | NIBObee |  
o-link | O-Link, OpenJTAG from www.100ask.net |  
openmoko | Openmoko debug board (v3) |  
pavr | Jason Kyle's pAVR Serial Programmer |  
pickit2 | MicroChip's PICkit2 Programmer |  
pickit4_isp | MPLAB(R) PICkit 4 in ISP mode |  
pickit4_pdi | MPLAB(R) PICkit 4 in PDI mode |  
pickit4_updi | MPLAB(R) PICkit 4 in UPDI mode |  
pkobn_updi | Curiosity nano (nEDBG) in UPDI mode |  
ponyser | design ponyprog serial, reset |  
powerdebugger | Atmel PowerDebugger (ARM/AVR) in JTAG mode |  
powerdebugger_dw | Atmel PowerDebugger (ARM/AVR) in debugWire   mode |  
powerdebugger_isp | Atmel PowerDebugger (ARM/AVR) in ISP mode |  
powerdebugger_pdi | Atmel PowerDebugger (ARM/AVR) in PDI mode |  
powerdebugger_updi | Atmel PowerDebugger (ARM/AVR) in UPDI mode |  
serialupdi | SerialUPDI |  
siprog | Lancos SI-Prog | http://www.lancos.com/siprogsch.html
snap_isp | MPLAB(R) SNAP in ISP mode |  
snap_pdi | MPLAB(R) SNAP in PDI mode |  
snap_updi | MPLAB(R) SNAP in UPDI mode |  
stk500 | Atmel STK500 |  
stk500hvsp | Atmel STK500 V2 in high-voltage serial   programming mode |  
stk500pp | Atmel STK500 V2 in parallel programming mode |  
stk500v1 | Atmel STK500 Version 1.x firmware |  
stk500v2 | Atmel STK500 Version 2.x firmware |  
stk600 | Atmel STK600 |  
stk600hvsp | Atmel STK600 in high-voltage serial   programming mode |  
stk600pp | Atmel STK600 in parallel programming mode |  
tc2030 | Tag-Connect TC2030 |  
teensy | Teensy Bootloader |  
ttl232r | FTDI TTL232R-5V with ICSP adapter |  
tumpa | TIAO USB Multi-Protocol Adapter |  
um232h | UM232H module from FTDI |  
uncompatino | uncompatino with all pairs of pins shorted |  
usbasp | USBasp | http://www.fischl.de/usbasp/
usbasp-clone | Any usbasp clone with correct VID/PID |  
usbtiny | USBtiny simple USB programmer, | https://learn.adafruit.com/usbtinyisp
wiring | Wiring |  
xbee | XBee Series 2 Over-The-Air (XBeeBoot) |  
xplainedmini | Atmel AVR XplainedMini in ISP mode |  
xplainedmini_dw | Atmel AVR XplainedMini in debugWIRE mode |  
xplainedmini_updi | Atmel AVR XplainedMini in UPDI mode |  
xplainedpro | Atmel AVR XplainedPro in JTAG mode |  
xplainedpro_pdi | Atmel AVR XplainedPro in PDI mode |  
xplainedpro_updi | Atmel AVR XplainedPro in UPDI mode |  
  |   |  
89isp |   |  
abcmini |   |  
alf |   |  
atisp |   |  
bascom |   |  
blaster |   |  
bsd |   |  
dapa |   |  
dt006 |   |  
ere-isp-avr |   |  
frank-stk200 |   |  
futurlec |   |  
picoweb |   |  
pony-stk200 |   |  
sp12 |   |  
stk200 |   |  
xil |   |

SpenceKonde, 2023-06-16:
Have any of you considered examining the list for programmers that are extinct in wild? This just seems an obscene number of programmer, and I have never heard of a fair number of them despite being very involved in Arduino. I think some of these are may be known only from archeological records.... you should start marking the ones that are essentially unused as deprecated, so they print a warning to the effect of "This programmer is old was never popular and, and to our knowedge is exrtinct in the field. If this is not trhe case please let us know at (github link) so that we can remove it from the list for programmers being dropped. Is the correct policy to support even non-microchip-non-atmel prrogrammers sold as commercial productrs a decade ago and not since then? 

I assume you're not able to get telemetry on which programmers get used whichis a shame

stefanrueger, 2023-06-16:
> extinct in the wild

How would that happen and if it did how would we know? Someone somewhere acquired/built a programmer, and it should still work after 1, 2, 4, 8 , 16 or 32 year, no? Why should support be abandoned? Same for parts.

This issue would like to help people realise they may have programmers that they didn't know they had: For example, some ingenious peeps figured how to turn a FTDI chip into a programmer and contributed their code to avrdude. If only there was a little more documentation about that more people could use their FTDI breakout boards/cables to program some of they parts they may still have. I for one still expect to discover that my toaster actually can program AT89Sxxx parts because there is a particular  AVRDUDE programmer that makes use of the infrared radiation... :)

> telemetry

We don't. Issues are a proxy measure for what people use and in which way. A lot of what I read there continues to surprise me. It's not all arduino stuff, ...

stefanrueger, 2023-06-23:
I am thinking of closing  this Issue (having updated around 12 programmer definitions) and re-opening one with labels documentation and help wanted to ask for PRs that add comments, URLs, etc between programmer headline and `programmer` keyword in `avrdude.conf.in`. Suggesting that _to be pinned_ so it stays top. of issue list.

mcuee, 2023-06-24:
I will just pin this issue as it is already labeled with `Documentation` and `help wanted`.

stefanrueger, 2023-06-24:
It's better to close this issue and re-open with a crisp request and an example.

closed 2023-06-24 by stefanrueger

mcuee, 2023-06-24:
> It's better to close this issue and re-open with a crisp request and an example.

Thanks for creating the new issue which has very clear instruction.