avrdudes/avrdude #1390: Avrdude segfaults when writing ihex or srec files for some parts

stefanrueger, 2023-06-10:
While reviewing `fileio.c` I could not help noticing that memory offsets of some parts are not dealt with correctly and can cause segmentation faults when writing Intel Hex or Motorola S-Record files:

```
$ avrdude -qq -p AT32UC3A0512 -c dryrun -A -U flash:r:out.hex:i
Segmentation fault (core dumped)

$ avrdude -qq -p AT32UC3A0512 -c dryrun -A -U flash:r:out.srec:s
Segmentation fault (core dumped)
```

mcuee, 2023-06-10:
BTW, avrdude does not support AVR32 in reality, since only one chip is supported.

Previously Joerg mentioned that we do not want to spend time on avr32 as the family is not popular. So I closed the old enhancement feature requests (without PR) for avr32.

But of course we should fix the issue of this particular avr32 chip supported (I can not test since I do not have board with this chip). And in case people raise PR and with good test results, we may incorporate the PR.

mcuee, 2023-06-11:
Indeed there is an issue in git main and PR #1392 fixes the issue.

```
$ ./avrdude_git -qq -p AT32UC3A0512 -c dryrun -A -U flash:r:out.hex:i
Segmentation fault

$ ./avrdude_git -qq -p AT32UC3A0512 -c dryrun -A -U flash:r:out.srec:s
Segmentation fault

$ ./avrdude_pr1392 -q -p AT32UC3A0512 -c dryrun -A -U flash:r:out.hex:i
avrdude_pr1392: AVR device initialized and ready to accept instructions

avrdude_pr1392: processing -U flash:r:out.hex:i
avrdude_pr1392: reading flash memory ...
avrdude_pr1392: writing output file out.hex

avrdude_pr1392 done.  Thank you.

$ ./avrdude_pr1392 -q -p AT32UC3A0512 -c dryrun -A -U flash:r:out.srec:s
avrdude_pr1392: AVR device initialized and ready to accept instructions

avrdude_pr1392: processing -U flash:r:out.srec:s
avrdude_pr1392: reading flash memory ...
avrdude_pr1392: writing output file out.srec

avrdude_pr1392 done.  Thank you.
```

closed 2023-06-11 by stefanrueger