avrdudes/avrdude #1398: Mute missing USB device warning when programmer is present with another USB PID

MCUdude, 2023-06-13:
Some programmers, especially older Atmel ones, tend to have changed their USB PID during their lifespan.
The JTAGICE3 for instance, started out with PID 0x2110 but was later changed to 0x2140 when the never CMCIS-DSP firmware was released.

However, the way usb_hidapi.c works, Avrdude prints a warning for every possible USB PID that's not present.
Here's an example of a JTAGICE3 that's running the latest firmware.

```
$ ./avrdude -cjtag3updi -patmega4808
avrdude warning: USB device with VID: 0x03eb and PID: 0x2110 not found
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9650 (probably m4808)

avrdude done.  Thank you.
```

Perhaps we should mute the "USB device with VID" warning completely? if no programmer is connected, we'll get a "no device found matching VID" error anyways, which is just as useful.

```
$ ./avrdude -cjtag3updi -patmega4808
avrdude warning: USB device with VID: 0x03eb and PID: 0x2110 not found
avrdude warning: USB device with VID: 0x03eb and PID: 0x2140 not found
avrdude error: no device found matching VID 0x03eb and PID list: 0x2110, 0x2140
avrdude error: unable to open programmer jtag3updi on port usb

avrdude done.  Thank you.
```

stefanrueger, 2023-06-13:
Good idea. Perhaps still print with `-vv`?

MCUdude, 2023-06-14:
> Perhaps still print with -vv?

I'm file with this! I'll create a PR this evening

closed 2023-06-15 by stefanrueger