avrdudes/avrdude #1414: Unable to build libavrdude dynamic library under Windows

mcuee, 2023-06-21:
From #952 
* #952 

As of now, there is no way to build shared libavrdude.dll under Windows, either with MinGW or MSVC.

```
[68/70] cmd.exe /C "cd . && C:\msys64\mingw64\bin\cc.exe -O2 -g -DNDEBUG  -Wl,-undefined,dynamic_lookup -shared -o src\libavrdude.dll -Wl,--out-implib,src\liblibavrdude.dll.a -Wl,--major-image-version,1,--minor-image-version,0 src/CMakeFiles/libavrdude.dir/arduino.c.obj src/CMakeFiles/libavrdude.dir/avr.c.obj src/CMakeFiles/libavrdude.dir/avr910.c.obj src/CMakeFiles/libavrdude.dir/avrcache.c.obj src/CMakeFiles/libavrdude.dir/avrftdi.c.obj src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.obj src/CMakeFiles/libavrdude.dir/avrintel.c.obj src/CMakeFiles/libavrdude.dir/avrpart.c.obj src/CMakeFiles/libavrdude.dir/bitbang.c.obj src/CMakeFiles/libavrdude.dir/buspirate.c.obj src/CMakeFiles/libavrdude.dir/butterfly.c.obj src/CMakeFiles/libavrdude.dir/ch341a.c.obj src/CMakeFiles/libavrdude.dir/config.c.obj src/CMakeFiles/libavrdude.dir/confwin.c.obj src/CMakeFiles/libavrdude.dir/crc16.c.obj src/CMakeFiles/libavrdude.dir/dfu.c.obj src/CMakeFiles/libavrdude.dir/dryrun.c.obj src/CMakeFiles/libavrdude.dir/fileio.c.obj src/CMakeFiles/libavrdude.dir/flip1.c.obj src/CMakeFiles/libavrdude.dir/flip2.c.obj src/CMakeFiles/libavrdude.dir/ft245r.c.obj src/CMakeFiles/libavrdude.dir/jtagmkI.c.obj src/CMakeFiles/libavrdude.dir/jtagmkII.c.obj src/CMakeFiles/libavrdude.dir/jtag3.c.obj src/CMakeFiles/libavrdude.dir/linuxgpio.c.obj src/CMakeFiles/libavrdude.dir/linuxspi.c.obj src/CMakeFiles/libavrdude.dir/lists.c.obj src/CMakeFiles/libavrdude.dir/micronucleus.c.obj src/CMakeFiles/libavrdude.dir/par.c.obj src/CMakeFiles/libavrdude.dir/pgm.c.obj src/CMakeFiles/libavrdude.dir/pgm_type.c.obj src/CMakeFiles/libavrdude.dir/pickit2.c.obj src/CMakeFiles/libavrdude.dir/pindefs.c.obj src/CMakeFiles/libavrdude.dir/ppi.c.obj src/CMakeFiles/libavrdude.dir/ppiwin.c.obj src/CMakeFiles/libavrdude.dir/serbb_posix.c.obj src/CMakeFiles/libavrdude.dir/serbb_win32.c.obj src/CMakeFiles/libavrdude.dir/ser_avrdoper.c.obj src/CMakeFiles/libavrdude.dir/ser_posix.c.obj src/CMakeFiles/libavrdude.dir/ser_win32.c.obj src/CMakeFiles/libavrdude.dir/serialupdi.c.obj src/CMakeFiles/libavrdude.dir/stk500.c.obj src/CMakeFiles/libavrdude.dir/stk500v2.c.obj src/CMakeFiles/libavrdude.dir/stk500generic.c.obj src/CMakeFiles/libavrdude.dir/strutil.c.obj src/CMakeFiles/libavrdude.dir/teensy.c.obj src/CMakeFiles/libavrdude.dir/term.c.obj src/CMakeFiles/libavrdude.dir/updi_link.c.obj src/CMakeFiles/libavrdude.dir/updi_nvm.c.obj src/CMakeFiles/libavrdude.dir/updi_readwrite.c.obj src/CMakeFiles/libavrdude.dir/updi_state.c.obj src/CMakeFiles/libavrdude.dir/urclock.c.obj src/CMakeFiles/libavrdude.dir/usbasp.c.obj src/CMakeFiles/libavrdude.dir/usb_hidapi.c.obj src/CMakeFiles/libavrdude.dir/usb_libusb.c.obj src/CMakeFiles/libavrdude.dir/usbtiny.c.obj src/CMakeFiles/libavrdude.dir/update.c.obj src/CMakeFiles/libavrdude.dir/wiring.c.obj src/CMakeFiles/libavrdude.dir/xbee.c.obj src/CMakeFiles/libavrdude.dir/__/lexer.c.obj src/CMakeFiles/libavrdude.dir/__/config_gram.c.obj  -lm  C:/msys64/mingw64/lib/libelf.a  C:/msys64/mingw64/lib/libusb.dll.a  C:/msys64/mingw64/lib/libusb-1.0.dll.a  C:/msys64/mingw64/lib/libhidapi.dll.a  C:/msys64/mingw64/lib/libftdi1.dll.a  C:/msys64/mingw64/lib/libreadline.dll.a  C:/msys64/mingw64/lib/libncurses.a  -lsetupapi  -lhid  -lws2_32  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."
FAILED: src/libavrdude.dll src/liblibavrdude.dll.a
cmd.exe /C "cd . && C:\msys64\mingw64\bin\cc.exe -O2 -g -DNDEBUG  -Wl,-undefined,dynamic_lookup -shared -o src\libavrdude.dll -Wl,--out-implib,src\liblibavrdude.dll.a -Wl,--major-image-version,1,--minor-image-version,0 src/CMakeFiles/libavrdude.dir/arduino.c.obj src/CMakeFiles/libavrdude.dir/avr.c.obj src/CMakeFiles/libavrdude.dir/avr910.c.obj src/CMakeFiles/libavrdude.dir/avrcache.c.obj src/CMakeFiles/libavrdude.dir/avrftdi.c.obj src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.obj src/CMakeFiles/libavrdude.dir/avrintel.c.obj src/CMakeFiles/libavrdude.dir/avrpart.c.obj src/CMakeFiles/libavrdude.dir/bitbang.c.obj src/CMakeFiles/libavrdude.dir/buspirate.c.obj src/CMakeFiles/libavrdude.dir/butterfly.c.obj src/CMakeFiles/libavrdude.dir/ch341a.c.obj src/CMakeFiles/libavrdude.dir/config.c.obj src/CMakeFiles/libavrdude.dir/confwin.c.obj src/CMakeFiles/libavrdude.dir/crc16.c.obj src/CMakeFiles/libavrdude.dir/dfu.c.obj src/CMakeFiles/libavrdude.dir/dryrun.c.obj src/CMakeFiles/libavrdude.dir/fileio.c.obj src/CMakeFiles/libavrdude.dir/flip1.c.obj src/CMakeFiles/libavrdude.dir/flip2.c.obj src/CMakeFiles/libavrdude.dir/ft245r.c.obj src/CMakeFiles/libavrdude.dir/jtagmkI.c.obj src/CMakeFiles/libavrdude.dir/jtagmkII.c.obj src/CMakeFiles/libavrdude.dir/jtag3.c.obj src/CMakeFiles/libavrdude.dir/linuxgpio.c.obj src/CMakeFiles/libavrdude.dir/linuxspi.c.obj src/CMakeFiles/libavrdude.dir/lists.c.obj src/CMakeFiles/libavrdude.dir/micronucleus.c.obj src/CMakeFiles/libavrdude.dir/par.c.obj src/CMakeFiles/libavrdude.dir/pgm.c.obj src/CMakeFiles/libavrdude.dir/pgm_type.c.obj src/CMakeFiles/libavrdude.dir/pickit2.c.obj src/CMakeFiles/libavrdude.dir/pindefs.c.obj src/CMakeFiles/libavrdude.dir/ppi.c.obj src/CMakeFiles/libavrdude.dir/ppiwin.c.obj src/CMakeFiles/libavrdude.dir/serbb_posix.c.obj src/CMakeFiles/libavrdude.dir/serbb_win32.c.obj src/CMakeFiles/libavrdude.dir/ser_avrdoper.c.obj src/CMakeFiles/libavrdude.dir/ser_posix.c.obj src/CMakeFiles/libavrdude.dir/ser_win32.c.obj src/CMakeFiles/libavrdude.dir/serialupdi.c.obj src/CMakeFiles/libavrdude.dir/stk500.c.obj src/CMakeFiles/libavrdude.dir/stk500v2.c.obj src/CMakeFiles/libavrdude.dir/stk500generic.c.obj src/CMakeFiles/libavrdude.dir/strutil.c.obj src/CMakeFiles/libavrdude.dir/teensy.c.obj src/CMakeFiles/libavrdude.dir/term.c.obj src/CMakeFiles/libavrdude.dir/updi_link.c.obj src/CMakeFiles/libavrdude.dir/updi_nvm.c.obj src/CMakeFiles/libavrdude.dir/updi_readwrite.c.obj src/CMakeFiles/libavrdude.dir/updi_state.c.obj src/CMakeFiles/libavrdude.dir/urclock.c.obj src/CMakeFiles/libavrdude.dir/usbasp.c.obj src/CMakeFiles/libavrdude.dir/usb_hidapi.c.obj src/CMakeFiles/libavrdude.dir/usb_libusb.c.obj src/CMakeFiles/libavrdude.dir/usbtiny.c.obj src/CMakeFiles/libavrdude.dir/update.c.obj src/CMakeFiles/libavrdude.dir/wiring.c.obj src/CMakeFiles/libavrdude.dir/xbee.c.obj src/CMakeFiles/libavrdude.dir/__/lexer.c.obj src/CMakeFiles/libavrdude.dir/__/config_gram.c.obj  -lm  C:/msys64/mingw64/lib/libelf.a  C:/msys64/mingw64/lib/libusb.dll.a  C:/msys64/mingw64/lib/libusb-1.0.dll.a  C:/msys64/mingw64/lib/libhidapi.dll.a  C:/msys64/mingw64/lib/libftdi1.dll.a  C:/msys64/mingw64/lib/libreadline.dll.a  C:/msys64/mingw64/lib/libncurses.a  -lsetupapi  -lhid  -lws2_32  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/arduino.c.obj: in function `arduino_read_sig_bytes':
C:/work/avr/avrdude_test/avrdude_main/src/arduino.c:67: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/work/avr/avrdude_test/avrdude_main/src/arduino.c:68: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/work/avr/avrdude_test/avrdude_main/src/arduino.c:62: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/work/avr/avrdude_test/avrdude_main/src/arduino.c:63: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/work/avr/avrdude_test/avrdude_main/src/arduino.c:47: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/arduino.c.obj:C:/work/avr/avrdude_test/avrdude_main/src/arduino.c:59: more undefined references to `avrdude_message2' follow
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/avr.c.obj:avr.c:(.rdata$.refptr.quell_progress[.refptr.quell_progress]+0x0): undefined reference to `quell_progress'
[... cut ...]
[69/70] C:\msys64\mingw64\bin\cc.exe -DCONFIG_DIR="\"C:/Program Files (x86)/avrdude/etc\"" -IC:/work/avr/avrdude_test/avrdude_main/build_shared/src -IC:/work/avr/avrdude_test/avrdude_main -IC:/work/avr/avrdude_test/avrdude_main/build_shared -IC:/msys64/mingw64/include/libusb-compat -IC:/work/avr/avrdude_test/avrdude_main/src -O2 -g -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -MD -MT src/CMakeFiles/avrdude.dir/developer_opts.c.obj -MF src\CMakeFiles\avrdude.dir\developer_opts.c.obj.d -o src/CMakeFiles/avrdude.dir/developer_opts.c.obj -c C:/work/avr/avrdude_test/avrdude_main/src/developer_opts.c
ninja: build stopped: subcommand failed.
```

mcuee, 2023-10-28:
This is the CMake config log using VS2022.

```
PS C:\work\avr\avrdude_test\avrdude> cmake -D DEBUG_CMAKE=1 -D USE_STATIC_LIBS=ON -D USE_EXTERNAL_LIBS=1 -DBUILD_SHARED_LIBS=ON -D CMAKE_VERBOSE_MAKEFILE=ON  -B build_shared_msvc
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 6.2.9200.
-- The C compiler identification is MSVC 19.37.32824.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.39.1.windows.1")
-- Found FLEX: C:/ProgramData/chocolatey/bin/win_flex.exe (found version "2.6.3")
-- Found BISON: C:/ProgramData/chocolatey/bin/win_bison.exe (found version "2.7")
-- Fetching external libraries, please wait...
-- The CXX compiler identification is MSVC 19.37.32824.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using external library 'libelf'
-- Using external library 'libusb'
-- Using external library 'libhidapi'
-- Using external library 'libftdi'
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Windows-6.2.9200
-- CMAKE_SYSTEM: Windows-6.2.9200
-- CMAKE_FIND_ROOT_PATH:
-- CMAKE_C_COMPILER: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe
-- CONFIG_DIR: C:/Program Files (x86)/avrdude/etc
-- AVRDUDE_FULL_VERSION: 7.2-20231024 (d6c61f5f)
-- USE_EXTERNAL_LIBS: 1
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: 1
-- HAVE_LIBUSB: 1
-- HAVE_LIBUSB_1_0: HAVE_LIBUSB_1_0-NOTFOUND
-- HAVE_LIBUSB_WIN32: HAVE_LIBUSB_WIN32-NOTFOUND
-- HAVE_LIBHIDAPI: 1
-- HAVE_LIBFTDI: 1
-- HAVE_LIBFTDI1: HAVE_LIBFTDI1-NOTFOUND
-- HAVE_LIBREADLINE: 1
-- HAVE_LIBSERIALPORT: HAVE_LIBSERIALPORT-NOTFOUND
-- HAVE_LIBELF_H: 1
-- HAVE_LIBELF_LIBELF_H:
-- HAVE_USB_H:
-- HAVE_LUSB0_USB_H: 1
-- HAVE_LIBUSB_H:
-- HAVE_LIBUSB_1_0_LIBUSB_H:
-- HAVE_HIDAPI_HIDAPI_H: 1
-- LIBUSB_COMPAT_DIR:
-- HAVE_LIBGPIOD:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DON'T HAVE libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi
-- DON'T HAVE libftdi1
-- DO HAVE    libreadline
-- DON'T HAVE libserialport
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done (21.2s)
-- Generating done (0.1s)
[... 183 lines cut ...]
7>xbee.c
7>lexer.c
7>config_gram.c
7>getopt.c
7>gettimeofday.c
7>Generating Code...
7>Done building project "libavrdude.vcxproj" -- FAILED.
8>------ Build started: Project: avrdude, Configuration: Release x64 ------
8>Building Custom Rule C:/work/avr/avrdude_test/avrdude/src/CMakeLists.txt
8>Microsoft (R) C/C++ Optimizing Compiler Version 19.37.32824 for x64
8>Copyright (C) Microsoft Corporation.  All rights reserved.
8>cl /c /IC:\work\avr\avrdude_test\avrdude\build_shared_msvc\src /IC:\work\avr\avrdude_test\avrdude /IC:\work\avr\avrdude_test\avrdude\build_shared_msvc /IC:\work\avr\avrdude_test\avrdude\src /IC:\work\avr\avrdude_test\avrdude\src\msvc /I"C:\work\avr\avrdude_test\avrdude\build_shared_msvc\_deps\libelf-src\include" /I"C:\work\avr\avrdude_test\avrdude\build_shared_msvc\_deps\libusb-src\include" /I"C:\work\avr\avrdude_test\avrdude\build_shared_msvc\_deps\libhidapi-src\include" /I"C:\work\avr\avrdude_test\avrdude\build_shared_msvc\_deps\libftdi-src\include" /I"C:\src\vcpkg\vcpkg\installed\x64-windows\include" /W3 /WX- /diagnostics:column /O2 /Ob2 /D _MBCS /D WIN32 /D _WINDOWS /D NDEBUG /D "CONFIG_DIR=\"C:/Program Files (x86)/avrdude/etc\"" /D _CRT_SECURE_NO_WARNINGS=1 /D _CRT_NONSTDC_NO_WARNINGS=1 /D _WINSOCK_DEPRECATED_NO_WARNINGS=1 /D "CMAKE_INTDIR=\"Release\"" /Gm- /EHsc /MD /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /GR /std:c11 /Fo"avrdude.dir\Release\\" /Fd"avrdude.dir\Release\vc143.pdb" /external:W3 /Gd /TC /wd4018 /wd4244 /wd4267 /errorReport:prompt C:\work\avr\avrdude_test\avrdude\src\main.c C:\work\avr\avrdude_test\avrdude\src\developer_opts.c C:\work\avr\avrdude_test\avrdude\src\whereami.c
8>main.c
8>developer_opts.c
8>whereami.c
8>Generating Code...
8>LINK : fatal error LNK1181: cannot open input file 'Release\libavrdude.lib'
8>Done building project "avrdude.vcxproj" -- FAILED.
9>------ Build started: Project: ALL_BUILD, Configuration: Release x64 ------
9>Building Custom Rule C:/work/avr/avrdude_test/avrdude/CMakeLists.txt
10>------ Skipped Build: Project: INSTALL, Configuration: Release x64 ------
10>Project not selected to build for this solution configuration 
========== Build: 7 succeeded, 2 failed, 0 up-to-date, 1 skipped ==========
========== Build started at 3:55 pm and took 15.598 seconds ==========
```

mcuee, 2023-10-28:
Full build log with latest git main and MSYS2 mingw64.

<details>
<summary> click for the full log </summary>

```
 MINGW64 /c/work/avr/avrdude_test/avrdude
$ cmake -D CMAKE_BUILD_TYPE=RelWithDebInfo -D DEBUG_CMAKE=1 -D USE_STATIC_LIBS=OFF -DBUILD_SHARED_LIBS=ON -D CMAKE_VERBOSE_MAKEFILE=ON  -B build_shared -DCMAKE_SHARED_LINKER_FLAGS=-Wl,-undefined,dynamic_lookup
-- Building for: Ninja
-- The C compiler identification is GNU 13.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/mingw64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/msys64/usr/bin/git.exe (found version "2.40.1")
-- Found FLEX: C:/msys64/usr/bin/flex.exe (found version "2.6.4")
-- Found BISON: C:/msys64/usr/bin/bison.exe (found version "3.8.2")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - not found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Windows-10.0.22621
-- CMAKE_SYSTEM: Windows-10.0.22621
-- CMAKE_FIND_ROOT_PATH:
-- CMAKE_C_COMPILER: C:/msys64/mingw64/bin/cc.exe
-- CONFIG_DIR: C:/Program Files (x86)/avrdude/etc
-- AVRDUDE_FULL_VERSION: 7.2-20231024 (d6c61f5f)
-- USE_EXTERNAL_LIBS: OFF
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: C:/msys64/mingw64/lib/libelf.a
-- HAVE_LIBUSB: C:/msys64/mingw64/lib/libusb.dll.a
-- HAVE_LIBUSB_1_0: C:/msys64/mingw64/lib/libusb-1.0.dll.a
-- HAVE_LIBUSB_WIN32: C:/msys64/mingw64/lib/libusb0.dll.a
-- HAVE_LIBHIDAPI: C:/msys64/mingw64/lib/libhidapi.dll.a
-- HAVE_LIBFTDI: HAVE_LIBFTDI-NOTFOUND
-- HAVE_LIBFTDI1: C:/msys64/mingw64/lib/libftdi1.dll.a
-- HAVE_LIBREADLINE: C:/msys64/mingw64/lib/libreadline.dll.a
-- HAVE_LIBSERIALPORT: 1
-- HAVE_LIBELF_H: 1
-- HAVE_LIBELF_LIBELF_H: 1
-- HAVE_USB_H: 1
-- HAVE_LUSB0_USB_H: 1
-- HAVE_LIBUSB_H:
-- HAVE_LIBUSB_1_0_LIBUSB_H: 1
-- HAVE_HIDAPI_HIDAPI_H: 1
-- LIBUSB_COMPAT_DIR: C:/msys64/mingw64/include/libusb-compat
-- HAVE_LIBGPIOD:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
[... 112 lines cut ...]
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/avrcache.c.obj:C:/work/avr/avrdude_test/avrdude/src/avrcache.c:302: more undefined references to `avrdude_message2' follow
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/dryrun.c.obj:dryrun.c:(.rdata$.refptr.partdesc[.refptr.partdesc]+0x0): undefined reference to `partdesc'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/fileio.c.obj: in function `fileio_fmt_autodetect_fp':
C:/work/avr/avrdude_test/avrdude/src/fileio.c:1412: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/fileio.c.obj: in function `fileio_fmt_autodetect':
C:/work/avr/avrdude_test/avrdude/src/fileio.c:1422: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/fileio.c.obj: in function `segment_normalise':
C:/work/avr/avrdude_test/avrdude/src/fileio.c:1469: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/work/avr/avrdude_test/avrdude/src/fileio.c:1460: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/fileio.c.obj: in function `fileio_setparms':
C:/work/avr/avrdude_test/avrdude/src/fileio.c:1309: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/fileio.c.obj:C:/work/avr/avrdude_test/avrdude/src/fileio.c:1233: more undefined references to `avrdude_message2' follow
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/ft245r.c.obj:ft245r.c:(.rdata$.refptr.progbuf[.refptr.progbuf]+0x0): undefined reference to `progbuf'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/jtagmkI.c.obj: in function `jtagmkI_resync':
C:/work/avr/avrdude_test/avrdude/src/jtagmkI.c:247: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/work/avr/avrdude_test/avrdude/src/jtagmkI.c:255: undefined reference to `avrdude_message2'
[... cut ...]
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/work/avr/avrdude_test/avrdude/src/jtagmkI.c:258: undefined reference to `avrdude_message2'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/libavrdude.dir/jtagmkI.c.obj:C:/work/avr/avrdude_test/avrdude/src/jtagmkI.c:259: more undefined references to `avrdude_message2' follow
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

</details>

mcuee, 2023-11-02:
Since this has never been working, so I change the label to `enhancement` instead of `bug`.

mcuee, 2024-03-09:
It is actually the same under macOS.
```
mcuee@mcuees-Mac-mini avrdude % cmake -S . -B build/shared -DBUILD_SHARED_LIBS=ON -D CMAKE_C_FLAGS=-I/opt/homebrew/include -D CMAKE_EXE_LINKER_FLAGS=-L/opt/homebrew/Cellar 
-- The C compiler identification is AppleClang 15.0.0.15000309
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /opt/homebrew/bin/git (found version "2.44.0") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found BISON: /usr/bin/bison (found version "2.3")
-- Looking for libelf.h
-- Looking for libelf.h - not found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DO HAVE    libserialport
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done (1.0s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/mcuee/build/avr/avrdude_test/avrdude/build/shared
mcuee@mcuees-Mac-mini avrdude % cmake --build build/shared                                                                                                                  
[  1%] Generating avrdude.conf
[  1%] Built target conf
[  2%] [FLEX][Parser] Building scanner with flex 2.6.4
[  3%] [BISON][Parser] Building parser with bison 2.3
[  5%] Building C object src/CMakeFiles/libavrdude.dir/arduino.c.o
[  6%] Building C object src/CMakeFiles/libavrdude.dir/avr.c.o
[  7%] Building C object src/CMakeFiles/libavrdude.dir/avr910.c.o
[  9%] Building C object src/CMakeFiles/libavrdude.dir/avrcache.c.o
[ 10%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi.c.o
[ 11%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.o
[ 13%] Building C object src/CMakeFiles/libavrdude.dir/avrintel.c.o
[ 14%] Building C object src/CMakeFiles/libavrdude.dir/avrpart.c.o
[ 15%] Building C object src/CMakeFiles/libavrdude.dir/bitbang.c.o
[ 17%] Building C object src/CMakeFiles/libavrdude.dir/buspirate.c.o
[ 18%] Building C object src/CMakeFiles/libavrdude.dir/butterfly.c.o
[ 19%] Building C object src/CMakeFiles/libavrdude.dir/ch341a.c.o
[ 21%] Building C object src/CMakeFiles/libavrdude.dir/config.c.o
[ 22%] Building C object src/CMakeFiles/libavrdude.dir/confwin.c.o
[ 23%] Building C object src/CMakeFiles/libavrdude.dir/crc16.c.o
[ 25%] Building C object src/CMakeFiles/libavrdude.dir/dfu.c.o
[ 26%] Building C object src/CMakeFiles/libavrdude.dir/dryrun.c.o
[ 27%] Building C object src/CMakeFiles/libavrdude.dir/fileio.c.o
[ 28%] Building C object src/CMakeFiles/libavrdude.dir/flip1.c.o
[... 100 lines cut ...]
      ...
  "_quell_progress", referenced from:
      _avr_verify in avr.c.o
      _avr_flush_cache in avrcache.c.o
      _writeCachePage in avrcache.c.o
      _loadCachePage in avrcache.c.o
      _fileio_segments_normalise in fileio.c.o
      _jtag3_initialize in jtag3.c.o
      _jtag3_initialize_tpi in jtag3.c.o
      ...
  "_verbose", referenced from:
      _avr_read_byte_silent in avr.c.o
      _avr_signature in avr.c.o
      _avr_verify in avr.c.o
      _avr_verify in avr.c.o
      _silent_page_erase in avrcache.c.o
      _avrftdi_log in avrftdi.c.o
      _avrftdi_pin_setup in avrftdi.c.o
      ...
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
gmake[2]: *** [src/CMakeFiles/libavrdude.dir/build.make:1188: src/libavrdude.1.0.0.dylib] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:128: src/CMakeFiles/libavrdude.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
```

mcuee, 2024-03-09:
On the other hand, there is a trick from Homebrew formula.
https://github.com/Homebrew/homebrew-core/blob/589f58b81e8d3a03aebcf46d89821faef84ed040/Formula/a/avrdude.rb

```
mcuee@mcuees-Mac-mini avrdude % cmake -S . -B build/shared -DBUILD_SHARED_LIBS=ON -D CMAKE_C_FLAGS=-I/opt/homebrew/include -D CMAKE_EXE_LINKER_FLAGS=-L/opt/homebrew/Cellar -DCMAKE_SHARED_LINKER_FLAGS=-Wl,-undefined,dynamic_lookup
-- The C compiler identification is AppleClang 15.0.0.15000309
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /opt/homebrew/bin/git (found version "2.44.0") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found BISON: /usr/bin/bison (found version "2.3")
-- Looking for libelf.h
-- Looking for libelf.h - not found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DO HAVE    libserialport
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done (1.0s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/mcuee/build/avr/avrdude_test/avrdude/build/shared
mcuee@mcuees-Mac-mini avrdude % cmake --build build/shared 
[  1%] Generating avrdude.conf
[  1%] Built target conf
[  2%] [FLEX][Parser] Building scanner with flex 2.6.4
[  3%] [BISON][Parser] Building parser with bison 2.3
[  5%] Building C object src/CMakeFiles/libavrdude.dir/arduino.c.o
[  6%] Building C object src/CMakeFiles/libavrdude.dir/avr.c.o
[  7%] Building C object src/CMakeFiles/libavrdude.dir/avr910.c.o
[  9%] Building C object src/CMakeFiles/libavrdude.dir/avrcache.c.o
[ 10%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi.c.o
[ 11%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.o
[ 13%] Building C object src/CMakeFiles/libavrdude.dir/avrintel.c.o
[ 14%] Building C object src/CMakeFiles/libavrdude.dir/avrpart.c.o
[ 15%] Building C object src/CMakeFiles/libavrdude.dir/bitbang.c.o
[ 17%] Building C object src/CMakeFiles/libavrdude.dir/buspirate.c.o
[ 18%] Building C object src/CMakeFiles/libavrdude.dir/butterfly.c.o
[ 19%] Building C object src/CMakeFiles/libavrdude.dir/ch341a.c.o
[ 21%] Building C object src/CMakeFiles/libavrdude.dir/config.c.o
[ 22%] Building C object src/CMakeFiles/libavrdude.dir/confwin.c.o
[ 23%] Building C object src/CMakeFiles/libavrdude.dir/crc16.c.o
[ 25%] Building C object src/CMakeFiles/libavrdude.dir/dfu.c.o
[ 26%] Building C object src/CMakeFiles/libavrdude.dir/dryrun.c.o
[... 50 lines cut ...]
/Users/mcuee/build/avr/avrdude_test/avrdude/build/shared/lexer.c:2435:38: warning: comparison of integers of different signs: 'yy_size_t' (aka 'unsigned long') and 'int' [-Wsign-compare]
        if (((yy_n_chars) + number_to_move) > YY_CURRENT_BUFFER_LVALUE->yy_buf_size) {
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 warning generated.
[ 93%] Building C object src/CMakeFiles/libavrdude.dir/__/config_gram.c.o
[ 94%] Linking C shared library libavrdude.dylib
[ 94%] Built target libavrdude
[ 96%] Building C object src/CMakeFiles/avrdude.dir/main.c.o
[ 97%] Building C object src/CMakeFiles/avrdude.dir/developer_opts.c.o
[ 98%] Building C object src/CMakeFiles/avrdude.dir/whereami.c.o
[100%] Linking C executable avrdude
[100%] Built target avrdude

mcuee@mcuees-Mac-mini avrdude % otool -L ./build/shared/src/libavrdude.dylib 
./build/shared/src/libavrdude.dylib:
	@rpath/libavrdude.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1345.100.2)
	/opt/homebrew/opt/libusb-compat/lib/libusb-0.1.4.dylib (compatibility version 9.0.0, current version 9.4.0)
	/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 5.0.0, current version 5.0.0)
	/opt/homebrew/opt/hidapi/lib/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.14.0)
	/opt/homebrew/opt/libftdi/lib/libftdi1.2.dylib (compatibility version 2.0.0, current version 2.5.0)
	/usr/lib/libedit.3.dylib (compatibility version 2.0.0, current version 3.0.0)
	/opt/homebrew/opt/libserialport/lib/libserialport.0.dylib (compatibility version 2.0.0, current version 2.0.0)
	/usr/lib/libncurses.5.4.dylib (compatibility version 5.4.0, current version 5.4.0)
```

ndim, 2024-08-11:
Regarding `_avrdude_message2` and the Windows linking problems, see also https://github.com/avrdudes/avrdude/pull/1874#issuecomment-2275666777 and some of the follow-on comments.