avrdudes/avrdude #1426: Unable to read memory using -U

MCUdude, 2023-06-21:
I'm not sure it it's me that's gooved up something, or if this is a newæy introduced bug in Avrdude. For some reason, I can't read the flash content to stdout or to a file using -U:

```
$./avrdude -cserialupdi -patmega4808 -b230400 -Uflash:r:-:I
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9650 (probably m4808)

avrdude: processing -U flash:r:-:I
avrdude: reading flash memory ...
Reading | ################################################## | 100% 6.19 s 
avrdude: writing output file <stdout>

avrdude done.  Thank you.
```

If I write to a file, the file ends up empty.

MCUdude, 2023-06-21:
The same happens when I'm using the USBasp programmer. However, EEPROM works on both the jtag2updi, serialupdi and usbasp.

```
$./avrdude -cusbasp -patmega16 -Uflash:r:-:i
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9403 (probably m16)

avrdude: processing -U flash:r:-:i
avrdude: reading flash memory ...
Reading | ################################################## | 100% 2.86 s 
avrdude: writing output file <stdout>

avrdude done.  Thank you.

src $./avrdude -cusbasp -patmega16 -Ueeprom:r:-:i
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9403 (probably m16)

avrdude: processing -U eeprom:r:-:i
avrdude: reading eeprom memory ...
Reading | ################################################## | 100% 0.16 s 
avrdude: writing output file <stdout>
:20000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF00
:20002000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE0
:20004000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC0
:20006000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFA0
:20008000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF80
:2000A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF60
:2000C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF40
:2000E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF20
:20010000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
:20012000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFDF
:20014000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFBF
:20016000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF9F
:20018000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7F
:2001A000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF5F
:2001C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3F
:2001E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF1F
:00000001FF

avrdude done.  Thank you.
```

MCUdude, 2023-06-21:
Wait, it this because the flash memory is completely empty?

MCUdude, 2023-06-21:
Ah, of couse it it! However, it could be useful with a message saying that there are no output because the memory is empty.

stefanrueger, 2023-06-21:
>  it could be useful with a message saying that there are no output because the memory is empty

It does with `-v`
```
$ avrdude -vc dryrun -U flash:r:-:h -pm328p

avrdude: Version 7.1-20230616 (1459002d)
[...]         Copyright the AVRDUDE authors;
 
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)

avrdude: processing -U flash:r:-:h
avrdude: reading flash memory ...
Reading | ################################################## | 100% 0.00s
avrdude: flash is empty, resulting file has no contents
avrdude: writing output file <stdout>

avrdude done.  Thank you.
```

mcuee, 2023-06-21:
So the question is whether we want to promote the message a bit higher to appear without `-v`.

```
https://github.com/avrdudes/avrdude/blob/9269102ba5dee7706cc4432a528635293e27f05b/src/update.c#L429-L433
```

stefanrueger, 2023-06-21:
Well, flash memory has this delete trailing-0xff optimisation that can be switched off. And it says the flash memory is empty on -v. I am not worried about the current state of affairs.

mcuee, 2023-06-21:
Fair enough. I think I will close this as invalid.

closed 2023-06-21 by mcuee