avrdudes/avrdude #1427: jtag2updi programmer times out in terminal mode

MCUdude, 2023-06-21:
It would be nice if someone could check if this only applies to my setup, or of it's an Avrdude or jtag2updi FW bug.
I can't remember it was like this before, but I could be wrong...

```
$ ./avrdude -cjtag2updi -patmega4808 -b115200 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9650 (probably m4808)

avrdude: processing -t interactive terminal
avrdude> read flash
Reading | -------------------------------------------------- | 0% 0.00 s 
avrdude error: wrong checksum
avrdude warning: timeout/error communicating with programmer (status -4)
avrdude error: wrong checksum
avrdude warning: timeout/error communicating with programmer (status -4)
avrdude error: wrong checksum
avrdude warning: timeout/error communicating with programmer (status -4)
avrdude error: wrong checksum
avrdude warning: timeout/error communicating with programmer (status -4)
avrdude warning: timeout/error communicating with programmer (status -1)
avrdude error: timeout/error communicating with programmer (status -1)
quit
avrdude warning: timeout/error communicating with programmer (status -1)

```

stefanrueger, 2023-06-22:
> I can't remember it was like this before

One thing you could do is to check a previous avrdude version.

mcuee, 2023-06-22:
No issue from my side using jtag2updi based on Arduino Uno clone.

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c jtag2updi -P COM8 -p m4808 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9650 (probably m4808)

avrdude: processing -t interactive terminal
avrdude> read flash
Reading | ################################################## | 100% 0.43 s
0000  01 c0 d8 c0 11 24 80 91  40 00 88 23 69 f0 28 2f  |.....$..@..#i.(/|
0010  30 e0 83 fd 03 c0 2c 7f  23 2b 31 f4 80 93 40 00  |0.....,.#+1...@.|
0020  8c bb 80 e0 af d0 ec c0  a8 95 a0 9a a8 9a 10 92  |................|
0030  e2 05 80 91 82 12 83 70  81 30 29 f5 8c e5 90 e0  |.......p.0).....|
0040  80 93 48 08 90 93 49 08  81 e0 80 93 4b 08 83 e0  |..H...I.....K...|
0050  80 93 47 08 10 92 45 08  80 ec 80 93 46 08 88 e0  |..G...E.....F...|
0060  91 d0 07 9a 87 e0 81 50  89 f4 a8 95 0d e9 83 e0  |.......P.... ...|
0070  d8 2e 7c d0 81 34 f1 f4  79 d0 18 2f 8c d0 12 38  |..|..4..y../...8|
0080  89 f4 80 e0 13 c0 83 e7  90 e0 da cf 17 9a 2e ec  |................|
0090  36 e5 a8 95 90 91 44 08  97 fd e8 cf 21 50 31 09  |6.....D.....!P1 |
00a0  c1 f7 e1 cf 89 e0 11 38  09 f0 83 e0 58 d0 80 e1  |.......8 ...X...|
00b0  56 d0 df cf 82 34 19 f4  84 e1 75 d0 f8 cf 85 34  |V....4....u....4|
00c0  11 f4 85 e0 fa cf 85 35  31 f4 50 d0 c8 2f 4e d0  |.......51.P../N.|
00d0  d8 2f 61 d0 ec cf 86 35  19 f4 84 e0 64 d0 d1 cf  |./a....5....d...|
00e0  84 36 b9 f4 43 d0 42 d0  18 2f 40 d0 86 34 79 f4  |.6..C.B../@..4y.|
00f0  d0 5c 3c d0 88 83 21 96  11 50 d9 f7 4c d0 04 bf  |.\<...!..P..L...|
avrdude> quit

avrdude done.  Thank you.
```

mcuee, 2023-06-22:
@MCUdude 

Are you using the Nano 4808's built-in jtag2updi programmer? Or are you using a Arduino Uno/Nano based jtag2updi?

MCUdude, 2023-06-22:
I'm using a dedicated jtag2updi programmer connected to an ATmega4808 with a SerialUPDI header.

I'll try to download the latest jtag2updi firmware and re-flash the programmer to see if the issue goes away.

MCUdude, 2023-06-22:
Well, I messed up.

It turned out that I forgot to save the jtag2updi sketch where I uncommented `#define DISABLE_HOST_TIMEOUT`.
So after a new recompile, the programmer doesn't time out anymore!

closed 2023-06-22 by MCUdude