avrdudes/avrdude #1428: Combine identical AVR programmers in avrdude.conf

MCUdude, 2023-06-23:
Looking at the avrdude.conf file, there are seveal programmers that has "parents", but are otherwise identical except for their name. Perhaps we should combine this like @stefanrueger has done in PR #1419?

Diecimila:
https://github.com/avrdudes/avrdude/blob/9269102ba5dee7706cc4432a528635293e27f05b/src/avrdude.conf.in#L1589-L1598

AVRISPmkII:
https://github.com/avrdudes/avrdude/blob/9269102ba5dee7706cc4432a528635293e27f05b/src/avrdude.conf.in#L1303-L1322

JTAG1:
https://github.com/avrdudes/avrdude/blob/9269102ba5dee7706cc4432a528635293e27f05b/src/avrdude.conf.in#L1947-L1969

JTAG2:
https://github.com/avrdudes/avrdude/blob/9269102ba5dee7706cc4432a528635293e27f05b/src/avrdude.conf.in#L1982-L2031

JTAG2AVR32:
https://github.com/avrdudes/avrdude/blob/9269102ba5dee7706cc4432a528635293e27f05b/src/avrdude.conf.in#L2064-L2088

stefanrueger, 2023-06-23:
Can do though this requires PR #1424 to be merged: AVRDUDE programmers often compare the *first* programmer id with a fixed string to do this or omit that. Putting otherwise equivalent programmers into an alias list might subtly change avrdude behaviour. Once PR #1424 is merged the comparisons are no longer with the first programmer id in the list but with the id that matched with the user -c string.

BTW, the list feature of the programmer `id` has only rarely been used so far.
```
$ avrdude -sc*/At | grep -P "\tid\t" | grep ,
.prog	xplainedmini	id	"xplainedmini", "xplainedmini_isp"
.prog	atmelice	id	"atmelice", "atmelice_jtag"
.prog	powerdebugger	id	"powerdebugger", "powerdebugger_jtag"
.prog	pickit4	id	"pickit4", "pickit4_jtag"
.prog	snap	id	"snap", "snap_jtag"
```

MCUdude, 2023-06-23:
> BTW, the list feature of the programmer id has only rarely been used so far.

Yeah, that's my work 😁

BTW I'm leaving on family vacation today, and won't be much available until a week later

stefanrueger, 2023-06-23:
> family vacation

Have a great time!

PR #1430 (2023-06-25): Use pgm->id list for different names of the same AVR programmer
Fixes #1428
closed 2023-06-25 by stefanrueger