avrdudes/avrdude #1437: Improve programmer documentation in avrdude.conf

stefanrueger, 2023-06-24:
```
$ avrdude -c programmer/s
```

shows the `avrdude.conf` definition of the _programmer_ as AVRDUDE understands it, including comments. This issue asks for for Pull Requests that add comments in `src/avrdude.conf.in` between the programmer's headline and the `programmer` keyword that illuminate where to get the programmer from, its schematics, pinout, cabling, firmware, usage or any other useful information. A good example is `usbasp`, see below. This to improve the overall programmer-specific documentation.

```
$ avrdude -cusbasp/s
#------------------------------------------------------------
# usbasp
#------------------------------------------------------------

# From http://www.fischl.de/usbasp
# USBasp is a USB in-circuit programmer for Atmel AVR controllers. It
# simply consists of an ATmega88 or an ATmega8 and a couple of passive
# components. The programmer uses a firmware-only USB driver, no
# special USB controller is needed.
#
# Firmware, schematics and drivers are available under above URL, too.
#
# Alternative firmware for the external programmer:
#
# - https://github.com/mariusgreuel/USBasp
#   Windows OS descriptor to load WinUSB driver automatically
#
# - https://github.com/nerdralph/usbasp
#   WCID compliant, meaning it should work on Windows without any
#   driver or .inf install; default SCK clock of 1.5 MHz and automatic
#   SCK slowing if target does not respond; PORTD is left as input, so
#   this firmware also works with USBISP modules, eg,
#   https://www.sciencetronics.com/greenphotons/?p=938
#
# - https://github.com/dioannidis/usbasp
#   As above plus HID UART support, setting unique serial number and
#   AT89S51/AT89S52 support

programmer
    id                     = "usbasp";
    desc                   = "USBasp ISP and TPI programmer";
    type                   = "usbasp";
    prog_modes             = PM_TPI | PM_ISP;
    connection_type        = usb;
    usbvid                 = 0x16c0; # VOTI
    usbpid                 = 0x05dc; # Obdev's free shared PID
    usbvendor              = "www.fischl.de";
    usbproduct             = "USBasp";
  # Old usbasp from fischl.de:
  # usbvid                 = 0x03EB; # ATMEL
  # usbpid                 = 0xC7B4; # (unoffical) USBasp
  # usbvendor              = "www.fischl.de";
  # usbproduct             = "USBasp";
;
```

mcuee, 2023-06-24:
I think it is okay to pin this issue.

mcuee, 2024-02-06:
Documentation update
*  https://github.com/avrdudes/avrdude/discussions/1473#discussioncomment-8330439

From @stefanrueger 
> I still think it would be wonderful if we had a short section in the .pdf/.html documentation about (current) limitations.

mcuee, 2024-02-06:
I have documented the knows limitations here and I just did some updates to list down the issues I know of.
https://github.com/avrdudes/avrdude/wiki/Known-limitations-of-avrdude

FAQ page may need some updates.
https://github.com/avrdudes/avrdude/wiki/FAQ

mcuee, 2024-02-07:
> Documentation update
> 
> * [avrdude 7.3 release discussion #1473 (comment)](https://github.com/avrdudes/avrdude/discussions/1473#discussioncomment-8330439)
> 
> From @stefanrueger
> 
> > I still think it would be wonderful if we had a short section in the .pdf/.html documentation about (current) limitations.

This will be addressed in another issue Post avrdude 7.3 release.