avrdudes/avrdude #1446: Can not build on raspbian (Rasperry PI) with libgpiod-dev installed.

merlinblack, 2023-06-27:
Pulled master as of 10 minutes ago (2023-06-27 06:00 UTC)

libgpiod details:
Package: libgpiod-dev
Version: 1.2-3+rpi1

```
-- Configuration summary:
-- ----------------------
-- DON'T HAVE libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DON'T HAVE libhidapi
-- DO HAVE    libftdi
-- DON'T HAVE libftdi1
-- DO HAVE    libreadline
-- DISABLED   doc
-- DISABLED   parport
-- ENABLED    linuxgpio
-- DO HAVE    libgpiod
-- ENABLED    linuxspi
-- ----------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/prog/avrdude/build_linux
[  1%] [BISON][Parser] Building parser with bison 3.3.2
Scanning dependencies of target libavrdude
[  2%] Building C object src/CMakeFiles/libavrdude.dir/arduino.c.o
[  4%] Building C object src/CMakeFiles/libavrdude.dir/avr.c.o
[  5%] Building C object src/CMakeFiles/libavrdude.dir/avr910.c.o
[  7%] Building C object src/CMakeFiles/libavrdude.dir/avrcache.c.o
[  8%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi.c.o
[ 10%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi_tpi.c.o
[ 11%] Building C object src/CMakeFiles/libavrdude.dir/avrintel.c.o
[ 13%] Building C object src/CMakeFiles/libavrdude.dir/avrpart.c.o
[ 14%] Building C object src/CMakeFiles/libavrdude.dir/bitbang.c.o
[ 15%] Building C object src/CMakeFiles/libavrdude.dir/buspirate.c.o
[ 17%] Building C object src/CMakeFiles/libavrdude.dir/butterfly.c.o
[ 18%] Building C object src/CMakeFiles/libavrdude.dir/ch341a.c.o
[ 20%] Building C object src/CMakeFiles/libavrdude.dir/config.c.o
[ 21%] Building C object src/CMakeFiles/libavrdude.dir/confwin.c.o
[ 23%] Building C object src/CMakeFiles/libavrdude.dir/dfu.c.o
[ 24%] Building C object src/CMakeFiles/libavrdude.dir/dryrun.c.o
[ 26%] Building C object src/CMakeFiles/libavrdude.dir/fileio.c.o
[ 27%] Building C object src/CMakeFiles/libavrdude.dir/flip1.c.o
[ 28%] Building C object src/CMakeFiles/libavrdude.dir/flip2.c.o
[ 30%] Building C object src/CMakeFiles/libavrdude.dir/ft245r.c.o
[ 31%] Building C object src/CMakeFiles/libavrdude.dir/jtagmkI.c.o
[ 33%] Building C object src/CMakeFiles/libavrdude.dir/jtagmkII.c.o
[ 34%] Building C object src/CMakeFiles/libavrdude.dir/jtag3.c.o
[ 36%] Building C object src/CMakeFiles/libavrdude.dir/linuxgpio.c.o
/home/pi/prog/avrdude/src/linuxgpio.c: In function ‘linuxgpio_libgpiod_close’:
/home/pi/prog/avrdude/src/linuxgpio.c:415:15: warning: implicit declaration of function ‘gpiod_line_set_direction_input’; did you mean ‘gpiod_line_request_input’? [-Wimplicit-function-declaration]
       int r = gpiod_line_set_direction_input(linuxgpio_libgpiod_lines[i]);
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
               gpiod_line_request_input
[ 37%] Building C object src/CMakeFiles/libavrdude.dir/linuxspi.c.o
[ 39%] Building C object src/CMakeFiles/libavrdude.dir/lists.c.o
[ 40%] Building C object src/CMakeFiles/libavrdude.dir/micronucleus.c.o
[ 42%] Building C object src/CMakeFiles/libavrdude.dir/par.c.o
[ 43%] Building C object src/CMakeFiles/libavrdude.dir/pgm.c.o
[ 44%] Building C object src/CMakeFiles/libavrdude.dir/pgm_type.c.o
[ 46%] Building C object src/CMakeFiles/libavrdude.dir/pickit2.c.o
[ 47%] Building C object src/CMakeFiles/libavrdude.dir/pindefs.c.o
[ 49%] Building C object src/CMakeFiles/libavrdude.dir/ppi.c.o
[ 50%] Building C object src/CMakeFiles/libavrdude.dir/ppiwin.c.o
[ 52%] Building C object src/CMakeFiles/libavrdude.dir/serbb_posix.c.o
[ 53%] Building C object src/CMakeFiles/libavrdude.dir/serbb_win32.c.o
[ 55%] Building C object src/CMakeFiles/libavrdude.dir/ser_avrdoper.c.o
[ 56%] Building C object src/CMakeFiles/libavrdude.dir/ser_posix.c.o
[ 57%] Building C object src/CMakeFiles/libavrdude.dir/ser_win32.c.o
[ 59%] Building C object src/CMakeFiles/libavrdude.dir/serialupdi.c.o
[ 60%] Building C object src/CMakeFiles/libavrdude.dir/stk500.c.o
[ 62%] Building C object src/CMakeFiles/libavrdude.dir/stk500v2.c.o
[ 63%] Building C object src/CMakeFiles/libavrdude.dir/stk500generic.c.o
[ 65%] Building C object src/CMakeFiles/libavrdude.dir/strutil.c.o
[ 66%] Building C object src/CMakeFiles/libavrdude.dir/teensy.c.o
[ 68%] Building C object src/CMakeFiles/libavrdude.dir/term.c.o
[ 69%] Building C object src/CMakeFiles/libavrdude.dir/updi_link.c.o
[ 71%] Building C object src/CMakeFiles/libavrdude.dir/updi_nvm.c.o
[ 72%] Building C object src/CMakeFiles/libavrdude.dir/updi_readwrite.c.o
[ 73%] Building C object src/CMakeFiles/libavrdude.dir/updi_state.c.o
[ 75%] Building C object src/CMakeFiles/libavrdude.dir/urclock.c.o
[ 76%] Building C object src/CMakeFiles/libavrdude.dir/usbasp.c.o
[ 78%] Building C object src/CMakeFiles/libavrdude.dir/usb_hidapi.c.o
[ 79%] Building C object src/CMakeFiles/libavrdude.dir/usb_libusb.c.o
[ 81%] Building C object src/CMakeFiles/libavrdude.dir/usbtiny.c.o
[ 82%] Building C object src/CMakeFiles/libavrdude.dir/update.c.o
[ 84%] Building C object src/CMakeFiles/libavrdude.dir/wiring.c.o
[ 85%] Building C object src/CMakeFiles/libavrdude.dir/xbee.c.o
[ 86%] Building C object src/CMakeFiles/libavrdude.dir/__/lexer.c.o
[ 88%] Building C object src/CMakeFiles/libavrdude.dir/__/config_gram.c.o
[ 89%] Linking C static library libavrdude.a
[ 92%] Built target libavrdude
Scanning dependencies of target avrdude
[ 94%] Building C object src/CMakeFiles/avrdude.dir/main.c.o
[ 95%] Building C object src/CMakeFiles/avrdude.dir/developer_opts.c.o
[ 97%] Linking C executable avrdude
/usr/bin/ld: libavrdude.a(linuxgpio.c.o): in function `linuxgpio_libgpiod_close':
/home/pi/prog/avrdude/src/linuxgpio.c:415: undefined reference to `gpiod_line_set_direction_input'
/usr/bin/ld: /home/pi/prog/avrdude/src/linuxgpio.c:426: undefined reference to `gpiod_line_set_direction_input'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/avrdude.dir/build.make:121: src/avrdude] Error 1
make[1]: *** [CMakeFiles/Makefile2:98: src/CMakeFiles/avrdude.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
Build failed.
```

mcuee, 2023-06-27:
What is the Raspberry Pi OS version used here?

merlinblack, 2023-06-27:
Raspbian GNU/Linux 10 (buster)

Thanks for the libelf-dev and libftdi1-dev tip :-)

mcuee, 2023-06-27:
Maybe you can still keep your Raspberry Pi OS version but you can try to build libgpiod from source codes (1.5.x branch or newer branches).
https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/

merlinblack, 2023-06-27:
Upgrading to raspbian 11 did the trick. Compiling a newer libgpiod most likely would have worked too.

closed 2023-06-27 by merlinblack