avrdudes/avrdude #1452: Warnings for MSVC build

mcuee, 2023-06-29:
From github action (also from my local VS2022 build)

```
D:\a\avrdude\avrdude\src\stk500v2.c(2728,34): warning C4146: unary minus operator applied to unsigned type, result still unsigned [D:\a\avrdude\avrdude\build\src\libavrdude.vcxproj]
D:\a\avrdude\avrdude\src\stk500v2.c(2740,35): warning C4146: unary minus operator applied to unsigned type, result still unsigned [D:\a\avrdude\avrdude\build\src\libavrdude.vcxproj]
```

stefanrueger, 2023-06-29:
The compiler is wrong, and the code is correct.

Some compilers are overzealous when the unary minus operates on an unsigned number, and issue a warning. In fact, `-u` is a perfectly valid unsigned for any unsigned `u`, and always has been from K&R C to C++20. `-u` is the same as `~u + 1`. C/C++ unsigned numbers form a group, where `-u` is the unique unsigned number that when added to `u` results in `0`. Neither `u` nor `-u` can be negative because no unsigned number is.

The overzealous compiler should note that while `-u` is *always* defined for any unsigned `u`, the same is not necessarily true for `-i` when `i` has a *signed* type: for most modern computer architectures the number `-INT_MIN` does not exist, and in fact compilers *are allowed undefined behaviour* when they come across this, mostly for, wait for it ...,  optimisation. Here a small test program, where the compiler *should* warn when the sign-change operation is applied to a signed `int`:
```
$ cat dopp.c
#include <stdio.h>
#include <limits.h>

int main() {
  int i = INT_MIN;
  if(i < 0)
    printf("i = INT_MIN is negative as expected!\n");

  int ub = -i;
  if(ub < 0)
    printf("ub = -i is also negative in this compiler implementation\n");
  if(-i >= 0)
    printf("-i, however, is *not* negative in this compiler implementation\n");

  printf("Will the self-driving car avoid oncoming traffic? Depends whether ub is tested or -i\n");
  if(ub >= 0)
    printf("Avoiding oncoming traffic\n");
  else
    printf("Sadly not in this case as the compiler is allowed undefined behaviour for -INT_MIN\n");
}

$ cc -Wall dopp.c
$ a.out
i = INT_MIN is negative as expected!
ub = -i is also negative in this compiler implementation
-i, however, is *not* negative in this compiler implementation
Will the self-driving car avoid oncoming traffic? Depends whether ub is tested or -i
Sadly not in this case as the compiler is allowed undefined behaviour for -INT_MIN
```

Optimisation leads to a case where compilers can arbitrarily drop comparisons, branches etc depending on whether they detect it's an overflow or not.

TLDR: It's the signed numbers you want to be afraid of, not the well-defined and well-behaved unsigned numbers.

mcuee, 2023-06-29:
This is the explanation from Microsoft about the warning.
https://learn.microsoft.com/en-us/cpp/error-messages/compiler-warnings/compiler-warning-level-2-c4146?view=msvc-170

mcuee, 2023-06-29:
And it seems to be common to disable the warning.

```
#ifdef _MSC_VER
/* disable: "warning C4146: unary minus operator applied to unsigned type, result still unsigned" */
#pragma warning(disable:4146)
#endif
```

stefanrueger, 2023-06-29:
> unary minus (negation) usually doesn't make sense when applied to an unsigned type

Ah, but what does Microsoft know about arithmetic? :wink:  Seriously, this reasoning is simply wrong. Unsigned arithmetic is defined as a modulo group operation. $-u$ is the uniquely defined element of the group for which $-u + u = 0$ holds.

closed 2023-07-01 by stefanrueger