avrdudes/avrdude #1453: avrdude git main test failure

mcuee, 2023-06-29:
The test script and hex files are from @stefanrueger which I think are good tests for avrdude.
[avrtest.zip](https://github.com/avrdudes/avrdude/files/11901753/avrtest.zip)

Somehow the following two tests do not pass.
```
# Test srec and hex formats: read one write the other and vice versa
./avrdude -qqc dryrun -p m328p \
  -U urboot_m328p_1s_autobaud_uart0_pr_ee_ce.srec \
  -T 'write flash 0x7000 0xc0cac01a 0xcafe "secret Coca Cola recipe"' \
  -U flash:w:cola-vending-machine.srec \
  -T "config bod=*2v7" \
  -U flash:r:app+data-m328p.hex:I
diff expected-flash-m328p.hex app+data-m328p.hex && echo srec/hex test m328p OK

./avrdude -qqc dryrun -p m328p \
  -U urboot_m328p_1s_autobaud_uart0_pr_ee_ce.hex \
  -T 'write flash 0x7000 0xc0cac01a 0xcafe "secret Coca Cola recipe"' \
  -U flash:w:cola-vending-machine.hex \
  -T "config bod=*2v7" \
  -U flash:r:app+data-m328p.srec:s
diff expected-flash-m328p.srec app+data-m328p.srec && echo hex/srec test m328p OK
```

Detailed test log

mcuee, 2023-06-29:
No issues under Linux (Ubuntu 20.04 x64 version).
```
mcuee@UbuntuSwift3:~/build/avr/avrtest$ ./avrtest 
hole/lext test m2560 OK
srec/hex test m328p OK
hex/srec test m328p OK
raw test m2560 OK
b number test m2560 OK
o number test m2560 OK
d number test m2560 OK
h number test m2560 OK
R number test m2560 OK
```

stefanrueger, 2023-06-29:
Windows creates \r\n line separators and the ground truth file has Unix \n line separators. Try `diff -w` instead of `diff`.

mcuee, 2023-06-29:
> Windows creates \r\n line separators and the ground truth file has Unix \n line separators. Try `diff -w` instead of `diff`.

That did the trick.

```
MINGW64 /c/work/avr/avrdude_test/avrdude_bin/avrtest
$ ./avrtest
hole/lext test m2560 OK
srec/hex test m328p OK
hex/srec test m328p OK
raw test m2560 OK
b number test m2560 OK
o number test m2560 OK
d number test m2560 OK
h number test m2560 OK
R number test m2560 OK
```

closed 2023-06-29 by mcuee