avrdudes/avrdude #1455: Cannot Configure/Build for: avrdude + ft232h + Windows 10

bbbH8TR, 2023-06-30:
Hello yall,

The [Adafruit FT232H](https://www.adafruit.com/product/2264) works nicely in Linux without hassle. Thank you to all the devs :-)
The avrdude version that was installed, from the repo, was 6.3-20171130.

However, when I tried to use the FT232H adapter in Windows 10 the error: "avrdude: Error: no libftdi or libusb support. Install libftdi1/libusb-1.0 or libftdi/libusb and run configure/make again." The version was avrdude-6.4-mingw32.zip (21-Feb-2022 11:24)

Setup guides regarding installation in Windows were largely outdated or depended on Visual Studio - which I cannot use.

Attempted:
1) Installed drivers with Zadig. All failed.
2) Downloaded [libusb](https://libusb.info/) and [libftdi](https://www.intra2net.com/en/developer/libftdi/download.php) drivers.  They could not be used.
3) Installed MinGW and MSYS2. Successful.
4) Downloaded and unpacked avrdude-6.4.tar.gz.
5) Attempted modified configure: (pg 43 of avrdude-doc-6.4.pdf )

    ./configure LDFLAGS="-static" --prefix=dir1 --datadir=dir1 --sysconfdir=dir1/bin --enable-versioned-doc=no --enable-libusb
    ./configure LDFLAGS="-static" --prefix=dir1 --datadir=dir1 --sysconfdir=dir1/bin --enable-versioned-doc=no --enable-libftdi
    ./configure LDFLAGS="-static" --prefix=dir1 --datadir=dir1 --sysconfdir=dir1/bin --enable-versioned-doc=no --enable-libusb_1_0
    ./configure LDFLAGS="-static" --prefix=dir1 --datadir=dir1 --sysconfdir=dir1/bin --enable-versioned-doc=no --disable-libusb_1_0

These all failed. 

Used: 
$ ./configure --help

But it did not glean anymore useful information. Could someone please better guide me on how to set the attributes for the _configure_ command to enable either libftdi or libusb ?

Thank you for reading the post.

mcuee, 2023-06-30:
Please use the latest 7.1 release and CMake. Thanks.

mcuee, 2023-07-01:
If you prefer to build from source, please follow the instructions below.
https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-Windows-using-Visual-Studio (no need to use Zadig to switch driver as FTDI vendor driver is supported)
https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-Windows-using-MSYS2 (Zadig is necessary as you need to use libftdi/libusb).

closed 2023-06-30 by mcuee

bbbH8TR, 2023-07-01:
Thank you for setting me on the right path, mcuee :-)

It was just like you said; reverted to official FTDIChip drivers and Version 7.1 worked "out of the box."