avrdudes/avrdude #1462: [Regression] -F flag not correctly honored

dl8dtl, 2023-07-10:
The `-F` flag is meant to proceed even though seemingly fatal errors are present.
That does not work anymore.
One of the most important cases where it needs to be used is an STK500 that has been misconfigured in a way that the target can no longer respond, e.g. by setting `vtarg` to 0 V.
Using `-F` is supposed to override the non-communicating target, so the parameter dialog can be reached (in terminal mode) anyway.
This used to work in version 7.1, but fails now.
```
j@rock:~/src/avrdude % ./build_freebsd/src/avrdude -c stk500v2 -B 125khz -P /dev/cuaU0 -p atmega1281 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9704 (probably m1281)

avrdude: processing -t interactive terminal
avrdude> vtarg 0
avrdude warning: reducing V[aref] from 3.0 to 0.0
avrdude> 

avrdude done.  Thank you.

j@rock:~/src/avrdude % ./build_freebsd/src/avrdude -c stk500v2 -B 125khz -P /dev/cuaU0 -p atmega1281 -t
avrdude error: command failed
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower the bit clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude done.  Thank you.

j@rock:~/src/avrdude % ./build_freebsd/src/avrdude -c stk500v2 -B 125khz -P /dev/cuaU0 -p atmega1281 -tF
avrdude error: command failed
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower the bit clock frequency, e.g. -B 125kHz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000
avrdude error: Yikes!  Invalid device signature.
avrdude warning: expected signature for ATmega1281 is 1E 97 04

avrdude done.  Thank you.

j@rock:~/src/avrdude % ./build_freebsd_7.1/src/avrdude -c stk500v2 -B 125khz -P /dev/cuaU0 -p atmega1281 -tF

avrdude error: command failed
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000
avrdude error: Yikes!  Invalid device signature.
avrdude warning: expected signature for ATmega1281 is 1E 97 04
avrdude> vtarg 3
avrdude> 

avrdude done.  Thank you.
```

mcuee, 2023-07-11:
@stefanrueger 

I am not so sure if this has something to do with the enhancement with `-T` and `-U`.

MCUdude, 2023-07-11:
Good catch!

I'm not sure when this "feature" broke, but in this particular example, you can adjust the target voltage outside terminal mode, by using `-xvtarg=[voltage]`.

dl8dtl, 2023-07-11:
Sure, now that there's also a `-x` option for that, this would work.
Anyways, we once added `-F` since we (Brian and me) both managed to have at least one damaged (early) AVR where the signature memory was dropped; one reported `0xff 0xff 0xff` (so it was erased), the other one `0x00 0x00 0x00` (reprogrammed). This was with those parallel-port bit-bang adapters of those days, but it triggered us to have `-F` to say "Hey, I want to coninue here anyway, I know what I'm doing."

stefanrueger, 2023-07-11:
Yes, this is a regression. Fun fact: `-F` never was a true carry-on-regardless option. `-U` processing stopped even if `-F` was issued. Only the `-t` terminal  was allowed by `-F` to run. Here the v7.0 code
https://github.com/avrdudes/avrdude/blob/a855692d973c2e72d38b59be545375982af3d4f7/src/main.c#L1293-L1316

When recently `-t` was changed to be processed alongside (and in order of) other `-U`, `-T` and possibly other `-t` commands `-F`  would also exit for `-t` (not only `-U`). As a stickler for documentation I would say both is wrong. We can easily implement the `-F` properly (carry on regardless of wrong signature or initialisation problems as the documentation says) by PR #1462.

closed 2023-07-14 by dl8dtl

dl8dtl, 2023-07-14:
> Good catch!
> 
> I'm not sure when this "feature" broke, but in this particular example, you can adjust the target voltage outside terminal mode, by using `-xvtarg=[voltage]`.

Just to confirm: yes, this also works.

@stefanrueger Thanks for fixing! As you have shown, it was already broken before with the introduction of `-U` (which replaced a number of separate options for each purpose), but nobody noticed that so far. `-F` is really meant to be a "I know what I'm doing, just proceed" flag.

PR #1463 (2023-07-14): Bail out on failed initialisation unless -F was given
Fixes #1462