avrdudes/avrdude #1469: JTAG3 programmers hardcoded to read target voltage

MCUdude, 2023-07-17:
I just got an Xplained Pro ATtiny817, and when looking at its capabilities, I realized that Avrdude doesn't treat
`extra_features = HAS_VTARG_READ;` correctly. In fact, it ignores this parameter and tries to read the target voltage no matter what. And there are sure programmers, like the Xplained Mini that don't have target voltage read support, and instead report back a fake value.

https://github.com/avrdudes/avrdude/blob/5e715e2851eeea5593d323bd670da6ee1ea8afb4/src/jtag3.c#L2561-L2567

I can see if I can create a PR that resolved this and adds `extra_features = HAS_VTARG_READ;` to all JTAG3 programmers that support this (Xplained Pro supports vtarg read, but is not specified in avrdude.conf).

MCUdude, 2023-07-17:
Another thing, in terminal mode, `parms` is according to the help menu supposed to `display adjustable parameters`.
However, it also prints the current target voltage, which is usually not adjustable. Not sure how we should deal with this, as reading the target voltage _is_ useful sometimes.

mcuee, 2023-07-17:
> I just got an Xplained Pro ATtiny817, and when looking at its capabilities, I realized that Avrdude doesn't treat `extra_features = HAS_VTARG_READ;` correctly. In fact, it ignores this parameter and [...]

You can see no programmers use this paremeter: only `src/config.c`, `src/libavrdude.h` and `developer_opts.c` are using it.
https://github.com/search?q=repo%3Aavrdudes%2Favrdude+HAS_VTARG_READ&type=code

BTW, one relevant PR is your previous PR #1286 (to address Issue #1235). There you use the parameter `HAS_VTARG_ADJ` in three programmers: `stk500v1.c`, stk500v2.c` and `jtag3.c`.
https://github.com/avrdudes/avrdude/issues/1235
https://github.com/avrdudes/avrdude/pull/1286

> I can see if I can create a PR that resolved this and adds `extra_features = HAS_VTARG_READ;` to all JTAG3 programmers that support this (Xplained Pro supports vtarg read, but is not specified in avrdude.conf).

I think that is good.

stefanrueger, 2023-07-17:
It is not uncommon that avrdude.conf describes parameters of parts that are (currently) ignored by AVRDUDE but are there for later use or just to describe the part better. This might have been the case when the project introduced `HAS_VTARG_READ`. Unless you have an immediate idea how to use that (and provide a PR) it's OK to leave it as is.

> However, [`parms` in the terminal] also prints the current target voltage, which is usually not adjustable.

Easy solution is to change the help message from "display adjustable parameters" to "display useful parameters".

closed 2023-08-04 by stefanrueger