avrdudes/avrdude #1471: Autotools build in doc/ broken

dl8dtl, 2023-07-20:
Right now, using autotools is still the (IMHO) only way to build the documentation.
However, something is broken there, as the Makefile tries to build more than docs in the `docs/` subdir, and fails:
```
cd src
./bootstrap
./configure --enable-doc
…
cd doc && make
…
Updating ./version.texi
make -C .. avrdude
  CC       avrdude-main.o
  CC       avrdude-whereami.o
  CC       avrdude-developer_opts.o
make[1]: don't know how to make ./libavrdude.a. Stop

make[1]: stopped in /home/joerg/src/avrdude-official/src
*** Error code 2

Stop.
make: stopped in /home/joerg/src/avrdude-official/src/doc
```
When enforcing to proceed with `make -k`, the documentation is built fine nevertheless.
For the same reason, `make distcheck` is broken as well (which is the method to roll a release tarball from autotools).

mcuee, 2023-07-20:
@dl8dtl 

CMake documentation build is working fine. It is part of the Linux github action build.
https://github.com/avrdudes/avrdude/actions/runs/5603816956/jobs/10250936455

```
[ 60%] Linking C static library libavrdude.a
[ 60%] Built target libavrdude
[ 60%] Building C object src/CMakeFiles/avrdude.dir/main.c.o
[ 61%] Building C object src/CMakeFiles/avrdude.dir/developer_opts.c.o
[ 62%] Building C object src/CMakeFiles/avrdude.dir/whereami.c.o
[ 63%] Linking C executable avrdude
[ 63%] Built target avrdude
[ 63%] Built target avrdude_binaries
[ 64%] Generating version.texi
[ 65%] Generating programmers.txt
[ 65%] Generating programmers.texi
[ 66%] Generating programmer_types.txt
[ 67%] Generating programmer_types.texi
[ 68%] Generating parts.txt
[ 69%] Generating parts.texi
[ 70%] Generating avrdude.info
[ 70%] Built target info
[ 71%] Generating avrdude.dvi
[ [77](https://github.com/avrdudes/avrdude/actions/runs/5603816956/jobs/10250936455#step:5:78)%] Built target dvi
[ [78](https://github.com/avrdudes/avrdude/actions/runs/5603816956/jobs/10250936455#step:5:79)%] Generating avrdude.pdf
[ [84](https://github.com/avrdudes/avrdude/actions/runs/5603816956/jobs/10250936455#step:5:85)%] Built target pdf
[ [85](https://github.com/avrdudes/avrdude/actions/runs/5603816956/jobs/10250936455#step:5:86)%] Generating avrdude.ps
[ [92](https://github.com/avrdudes/avrdude/actions/runs/5603816956/jobs/10250936455#step:5:93)%] Built target ps
[ [93](https://github.com/avrdudes/avrdude/actions/runs/5603816956/jobs/10250936455#step:5:94)%] Generating avrdude-html/avrdude.html
[100%] Built target html
```

mcuee, 2023-07-20:
The generated documentations (pdf, ps, html, dvi, info) file is part of the build archive.
https://github.com/avrdudes/avrdude/suites/14423519513/artifacts/813751199
Here is the zip file with just the documentation files (pdf, ps, html, dvi, info) 
[avrdude-v7.2_doc.zip](https://github.com/avrdudes/avrdude/files/12102198/avrdude-v7.2_doc.zip)

mcuee, 2023-07-20:
@dl8dtl @MCUdude @mariusgreuel and @stefanrueger 
So the question is whether we want to remove the auto-tools files or not. There are other issues with the auto-tools which we may not want to spend time to fix.

mcuee, 2023-07-20:
@dl8dtl 

If you want to try to build the documentation, you need to install a few dependencies (texinfo, texlive, texi2html) and use 
```
cmake          -D BUILD_DOC=1
```.

Reference:
https://github.com/avrdudes/avrdude/blob/main/.github/workflows/build.yml

It was done by @mariusgreuel in PR #1096. I have tested under Linux and macOS without problems. And I think it should work under FreeBSD as well (I will check it out later).
* #1096

mcuee, 2023-07-20:
> And I think it should work under FreeBSD as well (I will check it out later).

Ah, Texlive is just too big (>7GB) and my FreeBSD VM does not have enough space. So I will not test that. But it should work.

mcuee, 2023-07-20:
@dl8dtl 

And there is no issue to build the doc under Ubuntu 20.04 with auo-tools.
```
mcuee@UbuntuSwift3:~/build/avr/avrdude_main/src$ ./bootstrap 
+ rm -rf autom4te.cache
+ LIBTOOLIZE=libtoolize
+ glibtoolize --version
+ libtoolize
libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
+ aclocal
+ autoheader
+ autoconf
configure.ac:46: warning: AC_PROG_LEX without either yywrap or noyywrap is obsolete
./lib/autoconf/programs.m4:716: _AC_PROG_LEX is expanded from...
./lib/autoconf/programs.m4:709: AC_PROG_LEX is expanded from...
configure.ac:46: the top level
configure.ac:123: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:123: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:123: the top level
configure.ac:230: warning: The macro `AC_HEADER_TIME' is obsolete.
configure.ac:230: You should run autoupdate.
./lib/autoconf/headers.m4:743: AC_HEADER_TIME is expanded from...
configure.ac:230: the top level
configure.ac:252: warning: The macro `AC_TRY_LINK' is obsolete.
configure.ac:252: You should run autoupdate.
./lib/autoconf/general.m4:2920: AC_TRY_LINK is expanded from...
configure.ac:252: the top level
configure.ac:284: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:284: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:284: the top level
configure.ac:302: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:302: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:302: the top level
configure.ac:314: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:314: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:314: the top level
configure.ac:326: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:326: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:326: the top level
configure.ac:338: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:338: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:338: the top level
configure.ac:461: warning: The macro `AC_TRY_COMPILE' is obsolete.
configure.ac:461: You should run autoupdate.
./lib/autoconf/general.m4:2847: AC_TRY_COMPILE is expanded from...
configure.ac:461: the top level
configure.ac:486: warning: The macro `AC_TRY_COMPILE' is obsolete.
configure.ac:486: You should run autoupdate.
./lib/autoconf/general.m4:2847: AC_TRY_COMPILE is expanded from...
configure.ac:486: the top level
configure.ac:510: warning: The macro `AC_TRY_LINK' is obsolete.
configure.ac:510: You should run autoupdate.
./lib/autoconf/general.m4:2920: AC_TRY_LINK is expanded from...
configure.ac:510: the top level
+ automake -a -c
configure.ac:37: installing './compile'
[... 252 lines cut ...]
make[1]: Leaving directory '/home/mcuee/build/avr/avrdude_main/src'
../avrdude -C ../avrdude.conf -c \? 2>&1 \
| gawk '$2 ~ /^=$/ {printf("@item @code{%s} @tab %s\n",$1,gensub("[^=]+=[ \t]*","",1))}' \
| sed "s# *,\? *<\?\(http://[^ \t>]*\)>\?#,@*\n@url{\1}#g" \
>programmers.texi
../avrdude -C ../avrdude.conf -p \? 2>&1 \
| gawk '$2 ~ /^=$/ {printf("@item @code{%s} @tab %s\n",$1,$3)}' \
| sed -e "`sed 's:\([^ \t]*\)[ \t]*\(.*\):s/\1$/\1 \2/g:g' <parts_comments.txt`" \
>parts.texi
../avrdude -C ../avrdude.conf -c \?type 2>&1 \
| gawk '$2 ~ /^=$/ {printf("@item @code{%s} @tab %s\n",$1,gensub("[^=]+=[ \t]*","",1))}' \
| sed "s#<\?\(http://[^ \t,>]*\)>\?#@url{\1}#g" \
>programmer_types.texi
  MAKEINFO avrdude.info
texi2html --split=node --css-include=./avrdude.css ./avrdude.texi
if [ -e ./avrdude.html -o -e ./avrdude_1.html ]; then \
 mkdir -p avrdude-html ; \
 mv -f *.html avrdude-html ; \
else \
 mv -f avrdude avrdude-html; \
fi;
  TEXI2DVI avrdude.dvi
  DVIPS    avrdude.ps
  TEXI2PDF avrdude.pdf
```

mcuee, 2023-07-20:
BTW, no issue with `make distcheck` either under Linux.
```
mcuee@UbuntuSwift3:~/build/avr/avrdude_main/src$ make distcheck
make  dist-gzip am__post_remove_distdir='@:'
make[1]: Entering directory '/home/mcuee/build/avr/avrdude_main/src'
make  distdir-am
make[2]: Entering directory '/home/mcuee/build/avr/avrdude_main/src'
if test -d "avrdude-7.2-20230720"; then find "avrdude-7.2-20230720" -type d ! -perm -200 -exec chmod u+w {} ';' && rm -rf "avrdude-7.2-20230720" || { sleep 5 && rm -rf "avrdude-7.2-20230720"; }; else :; fi
test -d "avrdude-7.2-20230720" || mkdir "avrdude-7.2-20230720"
 (cd doc && make  top_distdir=../avrdude-7.2-20230720 distdir=../avrdude-7.2-20230720/doc \
     am__remove_distdir=: am__skip_length_check=: am__skip_mode_fix=: distdir)
make[3]: Entering directory '/home/mcuee/build/avr/avrdude_main/src/doc'
make  distdir-am
make[4]: Entering directory '/home/mcuee/build/avr/avrdude_main/src/doc'
make  \
  top_distdir="../avrdude-7.2-20230720" distdir="../avrdude-7.2-20230720/doc" \
  dist-info
make[5]: Entering directory '/home/mcuee/build/avr/avrdude_main/src/doc'
make[5]: Leaving directory '/home/mcuee/build/avr/avrdude_main/src/doc'
make[4]: Leaving directory '/home/mcuee/build/avr/avrdude_main/src/doc'
make[3]: Leaving directory '/home/mcuee/build/avr/avrdude_main/src/doc'
make  \
  top_distdir="avrdude-7.2-20230720" distdir="avrdude-7.2-20230720" \
  dist-hook
make[3]: Entering directory '/home/mcuee/build/avr/avrdude_main/src'
rm -f \
avrdude-7.2-20230720/lexer.c \
avrdude-7.2-20230720/config_gram.c \
avrdude-7.2-20230720/config_gram.h
make[3]: Leaving directory '/home/mcuee/build/avr/avrdude_main/src'
test -n "" \
|| find "avrdude-7.2-20230720" -type d ! -perm -755 \
	-exec chmod u+rwx,go+rx {} \; -o \
  ! -type d ! -perm -444 -links 1 -exec chmod a+r {} \; -o \
  ! -type d ! -perm -400 -exec chmod a+r {} \; -o \
  ! -type d ! -perm -444 -exec /bin/bash /home/mcuee/build/avr/avrdude_main/src/install-sh -c -m a+r {} {} \; \
|| chmod -R a+r "avrdude-7.2-20230720"
make[2]: Leaving directory '/home/mcuee/build/avr/avrdude_main/src'
tardir=avrdude-7.2-20230720 && ${TAR-tar} chof - "$tardir" | eval GZIP= gzip --best -c >avrdude-7.2-20230720.tar.gz
make[1]: Leaving directory '/home/mcuee/build/avr/avrdude_main/src'
if test -d "avrdude-7.2-20230720"; then find "avrdude-7.2-20230720" -type d ! -perm -200 -exec chmod u+w {} ';' && rm -rf "avrdude-7.2-20230720" || { sleep 5 && rm -rf "avrdude-7.2-20230720"; }; else :; fi
case 'avrdude-7.2-20230720.tar.gz' in \
*.tar.gz*) \
  eval GZIP= gzip --best -dc avrdude-7.2-20230720.tar.gz | ${TAR-tar} xf - ;;\
*.tar.bz2*) \
  bzip2 -dc avrdude-7.2-20230720.tar.bz2 | ${TAR-tar} xf - ;;\
*.tar.lz*) \
  lzip -dc avrdude-7.2-20230720.tar.lz | ${TAR-tar} xf - ;;\
*.tar.xz*) \
  xz -dc avrdude-7.2-20230720.tar.xz | ${TAR-tar} xf - ;;\
*.tar.Z*) \
  uncompress -c avrdude-7.2-20230720.tar.Z | ${TAR-tar} xf - ;;\
*.shar.gz*) \
  eval GZIP= gzip --best -dc avrdude-7.2-20230720.shar.gz | unshar ;;\
*.zip*) \
  unzip avrdude-7.2-20230720.zip ;;\
*.tar.zst*) \
  zstd -dc avrdude-7.2-20230720.tar.zst | ${TAR-tar} xf - ;;\
esac
chmod -R a-w avrdude-7.2-20230720
chmod u+w avrdude-7.2-20230720
mkdir avrdude-7.2-20230720/_build avrdude-7.2-20230720/_build/sub avrdude-7.2-20230720/_inst
chmod a-w avrdude-7.2-20230720
test -d avrdude-7.2-20230720/_build || exit 0; \
dc_install_base=`CDPATH="${ZSH_VERSION+.}:" && cd avrdude-7.2-20230720/_inst && pwd | sed -e 's,^[^:\\/]:[\\/],/,'` \
  && dc_destdir="${TMPDIR-/tmp}/am-dc-$$/" \
  && am__cwd=`pwd` \
  && CDPATH="${ZSH_VERSION+.}:" && cd avrdude-7.2-20230720/_build/sub \
  && ../../configure \
     \
[... 638 lines cut ...]
rm -f ./.deps/libavrdude_la-stk500generic.Plo
rm -f ./.deps/libavrdude_la-stk500v2.Plo
rm -f ./.deps/libavrdude_la-strutil.Plo
rm -f ./.deps/libavrdude_la-teensy.Plo
rm -f ./.deps/libavrdude_la-term.Plo
rm -f ./.deps/libavrdude_la-update.Plo
rm -f ./.deps/libavrdude_la-updi_link.Plo
rm -f ./.deps/libavrdude_la-updi_nvm.Plo
rm -f ./.deps/libavrdude_la-updi_readwrite.Plo
rm -f ./.deps/libavrdude_la-updi_state.Plo
rm -f ./.deps/libavrdude_la-urclock.Plo
rm -f ./.deps/libavrdude_la-usb_hidapi.Plo
rm -f ./.deps/libavrdude_la-usb_libusb.Plo
rm -f ./.deps/libavrdude_la-usbasp.Plo
rm -f ./.deps/libavrdude_la-usbtiny.Plo
rm -f ./.deps/libavrdude_la-wiring.Plo
rm -f ./.deps/libavrdude_la-xbee.Plo
rm -f Makefile
make[1]: Leaving directory '/home/mcuee/build/avr/avrdude_main/src/avrdude-7.2-20230720/_build/sub'
if test -d "avrdude-7.2-20230720"; then find "avrdude-7.2-20230720" -type d ! -perm -200 -exec chmod u+w {} ';' && rm -rf "avrdude-7.2-20230720" || { sleep 5 && rm -rf "avrdude-7.2-20230720"; }; else :; fi
======================================================
avrdude-7.2-20230720 archives ready for distribution: 
avrdude-7.2-20230720.tar.gz
======================================================
```

mcuee, 2023-07-20:
Result of the build.
[avrdude-7.2.tar.gz](https://github.com/avrdudes/avrdude/files/12107869/avrdude-7.2.tar.gz)
[avrdude.pdf](https://github.com/avrdudes/avrdude/files/12107896/avrdude.pdf)
[avrdude-html.zip](https://github.com/avrdudes/avrdude/files/12107916/avrdude-html.zip)

dl8dtl, 2023-07-20:
Ah well, I remember … for whatever reason, this used to be broken under FreeBSD standard make ("bmake") already before, and needs GNU make ("gmake"). No idea why … in theory, the autotools-generated files are supposed to work with both flavors.

Nevertheless, as we basically deprecate autotools anyway, not worth spending more time.

Good to know the documentation is also built as part of CMake builds. I didn't look well enough then.

closed 2023-07-20 by dl8dtl