avrdudes/avrdude #1472: Use Avrdude to read silicon revision from "modern" AVRs

MCUdude, 2023-07-24:
It seems like Pymcuprog can read a byte from (an undocumented?) memory section that can be used to determine which silicon revision the chip is. If you're struggling with a weird bug in the MCU program, it _might_ actually be a silicon bug. Being able to use Avrdude to read this revision number/letter would only be a good thing.

From [pymcuprog/serialupdi/application.py](
https://github.com/microchip-pic-avr-tools/pymcuprog/blob/593afdc8e089e39a4fed9f4fb19ae81f5f51e9a5/pymcuprog/serialupdi/application.py#L89-L145)

```py
def read_device_info(self):
        """
        Reads out device information from various sources
        """
        sib = self.readwrite.read_sib()
        sib_info = decode_sib(sib)

        # Unable to read SIB?
        if sib_info is None:
            self.logger.warning("Unable to read SIB from device; attempting double-break recovery...")
            # Send double break and try again
            self.phy.send_double_break()
            sib = self.readwrite.read_sib()
            sib_info = decode_sib(sib)
            if sib_info is None:
                self.logger.error("Double-break recovery failed.  Unable to contact device.")
                raise PymcuprogError("Failed to read device info.")

        # Select correct NVM driver:
        # P:0 = tiny0, mega0 (16-bit, page oriented)
        # P:1 = N/A
        # P:2 = AVR DA, DB, DD (24-bit, word-oriented)
        # P:3 = AVR EA (16-bit, page oriented)
        if sib_info['NVM'] == '0':
            self.logger.info("NVM type 0: 16-bit, page oriented write")
            # DL is correctly configured already
            # Create new NVM driver
            self.nvm = NvmUpdiV0(self.readwrite, self.device)
        elif sib_info['NVM'] == '2':
            # This is a Dx-family member, and needs new DL and NVM
            self.logger.info("NVM type 2: 24-bit, word oriented write")
            # Create new DL
            datalink = UpdiDatalink24bit()
            # Use the existing PHY
            datalink.set_physical(self.phy)
            # And re-init
            datalink.init_datalink()
            # Create a read write access layer using this data link
            self.readwrite = UpdiReadWrite(datalink)
            # Create new NVM driver
            self.nvm = NvmUpdiAvrV2(self.readwrite, self.device)
        elif sib_info['NVM'] == '3':
            self.logger.info("NVM type 3: 16-bit, page oriented")
            # DL is correctly configured already
            # Create new NVM driver
            self.nvm = NvmUpdiAvrV3(self.readwrite, self.device)
        else:
            self.logger.error("Unsupported NVM revision - update pymcuprog.")

        self.logger.info("PDI revision = 0x%02X", self.readwrite.read_cs(constants.UPDI_CS_STATUSA) >> 4)
        if self.in_prog_mode():
            if self.device is not None:
                devid = self.read_data(self.device.sigrow_address, 3)
                devrev = self.read_data(self.device.syscfg_address + 1, 1)
                self.logger.info("Device ID from serialupdi = '%02X%02X%02X' rev '%s'", devid[0], devid[1], devid[2],
                                 chr(ord('A') + devrev[0]))
        return sib_info
```

MCUdude, 2023-07-24:
It looks like we need to add `syscfg_base` to avrdude.conf for the `.avr8x` and `.avrdx` "parents"

When looking at every syscfg_base value for every "modern" AVR supported by pymcuprog, they all have `0x0f00` as their value.

From avr16dd20.py

```py
    # Some extra AVR specific fields
    'nvmctrl_base': 0x00001000,
    'syscfg_base': 0x00000F00,
    'ocd_base': 0x00000F80,
    'address_size': '24-bit',
    'prog_clock_khz': 1800,
    'hv_implementation': 2,
    'device_id': 0x1E9433,
```

SpenceKonde, 2023-07-29:
Do a 32-btye read of the SIB.

That will also tell you the silicon revision - Format is like:
"tinyAVR P:aD:b-3M2 (XX.YYYYY.0)
where a is the programmer version
and B is the onchip debugger. There may be a null character in the data that comes off the chip to confuse you there, but do not be fooled, There is more to it:
-3M2 (XX.YYYYY.0)
-3M2 has to do with the PDI oscillator according to what leaks out of microchip in their code. 
XX is the **silicon die rev** represented as ascii characters that represent a single hexadecimal byte. 
YYYYY appears to be yet another bloody ide identifier: 
There are two large scale  clades here: 
The AVR Dx-series parts with word write and all sorts of yummy features like that have YYYYY = "KV00x" where x seems to have increased monotonically in order of product release. 
The shitty tiny-derived parts with the crapola 16/20 MHz fuse selectable oscillator, generally paired with much worse calibration facilities than the actual tinies have (EA certainly does take the worst of both worlds) have YYYYY = "59Bxy where y is any hexidecimal digit, and x  is a 0 or 1. MegaAVR 0's have 59B2y, and EA's are 59Fyy

But the cool think is that as early as the SIB read, you have knowledge of which silicon rev you're uploading to.

That all said, considering Microchips sense of urgency (less than a three toed sloth on a monday morning) about those die revs, this may not be an urgent thing to address, I belive (aside from cradle revs, while the part is barely shipping, and hardly any of the bad ones making it out) that only 3 modern AVRs have gotten a die rev that fixed problems (1-series 16k got a rev B that fixed 3 problems. 3217/3216 for a rev. C that fixed a lot more problems, and 3208/3209 is up to Rev. D, with a fairly nice batch of fixes. and everything else is milling around holding their broken peripheral wondering when it will be repaired.

If we started getting a lot of fixes  like this, we could pass a an argument if we were assuming something about an errata, and get an error code back saying our assumption is invalid, or a core could even compile multiple copies of a sketch if they had to and then have the upload tool select the right one

closed 2023-08-05 by stefanrueger

SpenceKonde, 2023-08-05:
lol, it's not a bug it's just a feature they didn't tell us about. It's obviously intentional. That's from Microchip. They don't use behavior that they consider incorrect. They mark it as errata and fix the hardware (granted, by then the sun may have exhausted most of it's fuel, expanded to envelope and incinerate the earth, so we may have other concerns than errata). 

I imagine it's a lot more useful if you have a database you can access that tells you which code corresponds to which die. 

There's one tricky thing about the data out of the SIB. It's meant to be displayed as ASCII, but there's a nul in the middle there that trips up a lot of methods of displaying it. Within the parenthesis, the first two digits are the die revision as hexadecimal formatted per REVID. The the numbers appear to avance I have some AVR32DA's from before and am getting some from after the bug was fixed. I'm very very interested in whether there will be a change in the SIB between my old busted one and my new working ones (The die rev WAS NOT CHANGED, nor was the bug ever publically acknowledged except informally by a senior engineer posting in avrfreaks telling people to contact support for replacements). But these parts, you know the ones qualified for use in life safety critical applications? Well the first AVR32DA's that reached customer hands through extrordinarily poor QA had 2 byte vectors instead of 4 byte vectors, meaning that only the first and last two 4k regions of memory could contain an interrupt..... and that's if the compiler did the best it could. But it doesn't even know, so it generates 4 byte vectors. Reset will work. Every even vector will execute a different vector than intended (generally one that wasn't defined hence jumping to BADISR, which jumps to reset which will then (if it's on current DxCore) issue a software reset to correct this dirty reset. The odd reset vectors will depend on whether -mrelax is used. If it is, most of these will be harmless NOP's and they will run the wrong ISR as before. If instead they are JMPs or had to be JMPs even with -mrelax, then the ISR will *attempt to execute the address of the location of the ISR as if it were the instruction*, which could do quite literally half of all possible things, though only a small number aremore than incrementally worse than just restarting there. THIS was "certified for life safety critical systems"?! I'd hate to see our non-critical systems. Oh, right, I disassembled some a few weeks ago, they were pretty shit. Lately I've gotten to participate in scrapping some discarded life sciences equipment. I dunno how much that cost new, but I almost felt unworthy to take it apart (But it was left at the dump after the company was bought out. Some of them used *all english screws*, internal substructures were hogged out of aluminum or fiberglass blocks (like FR4 only >1/2 inch thick and no circuit or soldermask or that crap - it was used to mount the motor that would bring this stage up into a location with incredibly sharp needles and some samples in a tray did something involving some sort of light (we couldn't find the light source, only a very fancy mirror assembly. Previous equipment has yielded several single photon detectors (!!). All of it BEAUTIFULLY built. The one with the photon detectors also had an Atmel AT94SPLC or whatever they called them, the wimpy AVR embedded in an FPGA? That are like $10 per chip new today?, a couple of ATmega's and an ATmel CPLD. Dated to 2002. Looked in beautiful condition. That one was FULL of E. C. O. (ie, post manufacturing, pre-sales retrofits - typicaly implemented by people with razors to cut traces and hair-thin wire and UV-cure glue to hold it in place - It's what you do when paying someone to hand-repair your batch of bad. fully loaded, boards is cheaper than just trashing them (ie, when the boards cost a fucking fortune). 

During development, I've heard of ultra-rush short turn time orders coming with a power rail shorted to ground or someting. No problem, despite the expense back in those days, they dropped a few grand on a thermal camera, thendump current between the shorted rails until they could see the hot spot, drill it out, and patch up the board. These were boards the size of whole panels, 20+ layers, always being designed with greatest sense of urgency - and when you're growing 30% a year whatever, if the thermal camera and rework people having to fix up 20+ layer boards will let your engineers develop faster, you do it. I don't know which products that was used on, but I know the cheapest devices they sold were around a mil for the "poverty model" and at least twice that with all the options (semiconductor test equipment - it's what would test either dies or assembled devices coming out of manufacturing and determing if they worked. Fairly small volume, and like every other piece of equipment involved with semiconductor production until you get down to the level of things like reeling machines, crazy expensive,

PR #1474 (2023-08-05): Read chip silicon revision and add IO memory on chips with PDI and UPDI
Resolves #1472.

This PR allows the jtag2updi or any JTAG3 compatible programmer that speaks UPDI to read the chip silicon revision. pymcuprog already has this feature, and I thought it might be useful in case someone is chasing a silicon bug "ghost" and don't know the revision of their chip.

I've tested this PR with an AVR128DA48, AVR128DB48, ATtiny3217, and AVR64DD32 using the `pkobn_updi` programmer.

I've tested the AVR128DA32 and the ATmega4809 using the jtag2updi and the xplainedmini_updi programmer.

I still haven' [... cut]